simatec/ioBroker.solax #1: Request interval

Schnutexxl, 2021-08-29:
Hallo,

bei mir funktioniert der request interval nicht, obwohl auf 6 Minuten eingestellt. 
Bei einem Neustart des Adapters werden die Daten Richtig abgerufen. Nach 6 Minuten kommen keine neuen Daten. Im Log erscheint folgendes:

solax.0 | 2021-08-29 11:27:50.703 | debug | is Night: true
solax.0 | 2021-08-29 11:27:50.686 | debug | Sun Azimut: NaN°
solax.0 | 2021-08-29 11:27:50.686 | debug | Sun Altitude: NaN°
solax.0 | 2021-08-29 11:27:50.686 | debug | calculate astrodata ...
solax.0 | 2021-08-29 11:27:50.685 | debug | current local Time: 11:27
solax.0 | 2021-08-29 11:27:50.685 | debug | nauticalDawn: aN:aN
solax.0 | 2021-08-29 11:27:50.684 | debug | nauticalDusk: aN:aN
solax.0 | 2021-08-29 11:27:50.683 | debug | nightCalc started ...

Kann es daran liegen, dass hier eine Variable "Night" auf true steht, obwohl das nicht stimmt und somit keine Daten abgerufen werden? Oder wo liegt der bug.

simatec, 2021-08-29:
Richtig, wenn Night = true werden keine Daten geholt. Hast du die richtigen Astrodaten eingetragen?  
Night sollte nur Nachts erfolgen, wi eh keine Einspeisung erfolgt

Schnutexxl, 2021-08-29:
Ich hatte erst die über das System holen lassen, hier wurde aber nichts angezeigt. Dann manuel eingetragen, auch keine Astrodaten.
In welcher Form müssen lan&lon eingetragen werden?

simatec, 2021-08-29:
Eigentlich ganz normal, wie sie auch in den Systemsettings eingetragen sind.
Breitengrad: 52.XXXXXX
Längengrad: 11.XXXXXX

Schnutexxl, 2021-08-29:
Ok, bei meinen Koordinaten war es wie folgt:

Breitengrad: 52.XXXXXX,XX
Längengrad:  9.XXXXX

nun auf Breitengrad: 52.XXXXXX
             Längengrad:  9.XXXXXX

angepasst und es funktioniert.

Bislang hatte ich das alles mit NodeRed realisiert und auch noch folgende Datenpunkte abgerufen:

[image]

Ist das auch noch geplant?

Und danke!

simatec, 2021-08-29:
Kannst du mir mal den Github Link zu deinem NodeRed Paket mal posten?
Dann schaue ich mir das mal genauer an

Schnutexxl, 2021-08-29:
Habe dafür kein Github. Ich kann es morgen exportieren, NodeRed oder JSON? Dann per Mail?

simatec, 2021-08-29:
Wenn du dafür ein Paket installiert hast, sollte der Name des Pakets reichen.

closed 2022-02-03 by simatec