simatec/ioBroker.solax #80: Falsche und fehlende Daten in den Objekten

OKTP, 2022-04-10:
**Describe the bug**  
In den Objekten werden mir teilweise falsche und teilweise keine Daten angezeigt. Ich kann auch nicht über die locale IP auf den Dongle zugreifen. Das Problem tritt sowohl im Localen als auch im Cloud Modus auf. Wenn ich über den AP des Dongles zugreife wird mir die lokale IP angezeigt. Wenn ich in der Solax Cloud schaue passen die Daten.

[image]
[image]

**Versions:**  
 - Adapter version: V0.4.5
 - JS-Controller version: Node.js v14.19.0 
 - Node version: <14.19.0> 
 - Operating system: Linux (Raspberry Pi)

**Additional context**  
Add any other context about the problem here.
[image]
[image]

simatec, 2022-04-10:
Leider kann ich mit deinen Screenshots nix anfangen und dein Issue verstehe ich nicht wirklich so recht. 

Was wird wann und wo nicht angezeigt? Welche Firmware ist auf dem Wifi Stick? Mit welcher IP greifst du lokal zu?
Was steht im Debug-Log?

OKTP, 2022-04-10:
Hallo, sorry das es so unübersichtlich ist.
In den Datenpunkten im Screenshot sind falsche Daten. Zum Beispiel 5000 Grad Akku Temperatur , 2500 Hz und so weiter. Einige Datenpunkte sind auch ganz leer.
Auf dem Stick ist die Version 2.033.20 . 
Im Debug Log vom Iobroker sind keine Einträge zu dem Problem vorhanden.
Ich greife über die IP Adresse zu die mir in der Weboberfläche vom Stick unter IP Adresse angezeigt wird.

OKTP, 2022-04-10:
Hier sind noch Sreens von den Einstellungen im Stick.

[image]
[image]
[image]

simatec, 2022-04-11:
Der Adapter konvertiert keine Daten.
Er bekommt diese direkt vom Wechselrichter bzw. vom Wifi Stick.

Warum bei dir die Werte nicht passen, kann ich aktuell nicht sagen.
Wie sehen die Werte aus, wenn du die Cloud im Adapter nutzt?

OKTP, 2022-04-11:
Wenn ich im Adapter die Cloud als Quelle auswähle, stimmen die Datenpunkte die mir unter Info angezeigt werden. Also Seriennummer des WR,  Status und so weiter. Die meisten anderen Datenpunkte sind leer.
Ist es möglich direkt an die Daten zu kommen die aus dem WR kommen? Dann könnte ich mal schauen wie der String aussieht.

simatec, 2022-04-11:
In der Cloud Variante sollte es keine leeren Datenpunkte geben.
Aktualisiere mal deine Objektbaum bzw. deinen Browser (F5)

OKTP, 2022-04-11:
[image]

Habe ich beides versucht. Auf dem Bild siehst du welche Datenpunkte leer sind.

OKTP, 2022-04-11:
{"success":true,"exception":"Query success!","result":{"inverterSN":"H34T12H8022030","sn":"SWQLQRY9MV","acpower":-790.0,"yieldtoday":3.6,"yieldtotal":6.2,"feedinpower":-19.0,"feedinenergy":5.33,"consumeenergy":5.38,"feedinpowerM2":0.0,"soc":20.0,"peps1":0.0,"peps2":0.0,"peps3":0.0,"inverterType":"14","inverterStatus":"102","uploadTime":"2022-04-11 13:30:06","batPower":826.0,"powerdc1":0.0,"powerdc2":0.0,"powerdc3":null,"powerdc4":null}}

Das gibt er mir da aus. Das die powerdc 1 , 2 und 3 null sind ist richtig. Die PV-strings sind noch nicht angeschlossen. Laden tut er im Moment nur über eine andere Anlage.

simatec, 2022-04-11:
Für die lokalen Daten kannst du mal bitte die Ausgabe von folgenden Befehl posten.

`curl -d "optType=ReadRealTimeData&pwd=<Wechselrichter-Passwort>" -X POST http://<Wechselrichter-IP>`

OKTP, 2022-04-11:
curl: (52) Empty reply from server

das kommt als Ausgabe.
Es funktioniert bei mir auch nicht das ich über die zugewiesene IP Adresse auf den Stick komme. Also über den Browser meine ich.

simatec, 2022-04-12:
Hmm der Stick mit externer Antenne sollte im Prinzip auch funktionieren.
Ich würde einfach die Firmware noch einmal flashen und den Stick mal komplett neu starten oder einmal vom Strom trennen.

OKTP, 2022-04-12:
Habe ich beider nochmal probiert. Leider ohne Erfolg. Eine Verbindung zum Stick muss aber bestehen. Ich habe in den Adapter Einstellungen mal eine Falsche IP Adresse eintrage, da wechselt der Datenpunkt für den Online Status auf fals.

simatec, 2022-04-12:
Ich habe mal einen Silly Log für die API-Ausgabe eingebaut.
Installiere mal direkt von Github und stelle die Adapterinstanz in der Logstufe auf "Silly" ("Alles")
Dann schaue mal, was für Werte im Log von deinem Stick kommen

OKTP, 2022-04-12:
Die Ausgabe kommt einmal nach dem Start. Danach kommt alle 10 Sekunden die Ausgabe die ich vorhin schon gepostet habe.

solax.0
2022-04-12 14:49:19.214	debug	Local Request Interval started ...

solax.0
2022-04-12 14:49:19.213	debug	Request Interval: 10 seconds

solax.0
2022-04-12 14:49:19.156	debug	Sun Azimut: 212.3°

solax.0
2022-04-12 14:49:19.155	debug	Sun Altitude: 42.6°

solax.0
2022-04-12 14:49:19.154	debug	calculate astrodata ...

solax.0
2022-04-12 14:49:19.151	debug	current local Time: 14:49

solax.0
2022-04-12 14:49:19.149	debug	nauticalDawn: 05:06

solax.0
2022-04-12 14:49:19.148	debug	nauticalDusk: 21:25

solax.0
2022-04-12 14:49:19.144	debug	nightCalc started ...

solax.0
2022-04-12 14:49:19.142	debug	System longitude: 11.440715789794921 System latitude: 52.05966806958091

solax.0
2022-04-12 14:49:19.135	debug	Solax is started in local-mode

node-red.0
2022-04-12 14:49:17.951	warn	State value to set for "0_userdata.0.EasyMeter.L2_aktuell" has value "-590.91" less than min "0"

solax.0
2022-04-12 14:49:17.043	silly	States system redis pmessage system.adapter.solax.0.logLevel/system.adapter.solax.0.logLevel:{"val":"silly","ack":true,"ts":1649767757032,"q":0,"from":"system.adapter.solax.0","lc":1649767673696}

solax.0
2022-04-12 14:49:16.992	info	starting. Version 0.4.6 in /opt/iobroker/node_modules/iobroker.solax, node: v14.19.0, js-controller: 3.3.22

solax.0
2022-04-12 14:49:16.310	debug	Plugin sentry Initialize Plugin (enabled=true)

solax.0
2022-04-12 14:49:16.074	debug	statesDB connected

solax.0
2022-04-12 14:49:16.073	debug	States connected to redis: 127.0.0.1:9000

solax.0
2022-04-12 14:49:16.065	debug	States create User PubSub Client

solax.0
2022-04-12 14:49:16.064	debug	States create System PubSub Client

solax.0
2022-04-12 14:49:16.053	debug	Redis States: Use Redis connection: 127.0.0.1:9000

solax.0
2022-04-12 14:49:16.049	debug	objectDB connected

solax.0
2022-04-12 14:49:16.042	silly	redis psubscribe cfg.o.system.user.*

solax.0
2022-04-12 14:49:16.038	debug	Objects connected to redis: 127.0.0.1:9001

solax.0
2022-04-12 14:49:16.018	debug	Objects client initialize lua scripts

solax.0
2022-04-12 14:49:16.016	debug	Objects create User PubSub Client

solax.0
2022-04-12 14:49:16.014	debug	Objects create System PubSub Client

solax.0
2022-04-12 14:49:16.006	debug	Objects client ready ... initialize now

solax.0
2022-04-12 14:49:15.933	debug	Redis Objects: Use Redis connection: 127.0.0.1:9001

node-red.0
2022-04-12 14:49:15.951	warn	State value to set for "0_userdata.0.EasyMeter.L2_aktuell" has value "-588.53" less than min "0"
host.raspberrypi
2022-04-12 14:49:14.526	info	instance system.adapter.solax.0 started with pid 5503
host.raspberrypi
2022-04-12 14:49:14.365	info	"system.adapter.solax.0" enabled

node-red.0
2022-04-12 14:49:13.953	warn	State value to set for "0_userdata.0.EasyMeter.L2_aktuell" has value "-582.76" less than min "0"

node-red.0
2022-04-12 14:49:11.955	warn	State value to set for "0_userdata.0.EasyMeter.L2_aktuell" has value "-575.86" less than min "0"
host.raspberrypi
2022-04-12 14:49:11.209	info	instance system.adapter.solax.0 terminated with code 11 (ADAPTER_REQUESTED_TERMINATION)
host.raspberrypi
2022-04-12 14:49:10.611	info	stopInstance system.adapter.solax.0 send kill signal

solax.0
2022-04-12 14:49:10.606	info	Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason

solax.0
2022-04-12 14:49:10.605	debug	Plugin sentry destroyed

solax.0
2022-04-12 14:49:10.604	info	terminating

solax.0
2022-04-12 14:49:10.602	info	Got terminate signal TERMINATE_YOURSELF

solax.0
2022-04-12 14:49:10.601	silly	States system redis pmessage system.adapter.solax.0.sigKill/system.adapter.solax.0.sigKill:{"val":-1,"ack":false,"ts":1649767750549,"q":0,"from":"system.host.raspberrypi","lc":1649767750549}
host.raspberrypi
2022-04-12 14:49:10.530	info	stopInstance system.adapter.solax.0 (force=false, process=true)
host.raspberrypi
2022-04-12 14:49:10.518	info	"system.adapter.solax.0" disabled

closed 2022-08-03 by simatec