simatec/ioBroker.solax #102: Solax adapter not refreshing data

louisvangeldrop, 2022-06-22:
The adapter connects to the pocket Wifi and collects the data only once after a restart of the adapter.
 
After the restart of the adapter, the adapter is not refreshing/polling the solax wifi pocket.

Admin version used: 6.0.7
Solax adapter version: 0.4.6

Johan219, 2022-06-23:
Unfortunately, I am experiencing the same problem. Data from the solar panels is no longer updated. I have tried both API cloud and advanced settings through the wifi stick. 

I am using the latest version of the adapter. I have tried to reinstall the adapter a few times. After installation, the data is grabbed, but that's it. There is no updating.

simatec, 2022-06-24:
Please post the debug log of the API request or the Wifi query

louisvangeldrop, 2022-06-24:
There is no debug message in the "Logs"-tab from the Solax Adapter.

simatec, 2022-06-24:
You need to set the adapter instance to debug

louisvangeldrop, 2022-06-24:
solax.0 | 2022-06-24 13:21:01.578 | debug | Local Request Interval started ...
-- | -- | -- | --
solax.0 | 2022-06-24 13:21:01.576 | debug | Request Interval: 10 seconds
solax.0 | 2022-06-24 13:21:01.518 | debug | Sun Azimut: 170.6°
solax.0 | 2022-06-24 13:21:01.518 | debug | Sun Altitude: 61.7°
solax.0 | 2022-06-24 13:21:01.516 | debug | calculate astrodata ...
solax.0 | 2022-06-24 13:21:01.514 | debug | current local Time: 13:21
solax.0 | 2022-06-24 13:21:01.513 | debug | nauticalDawn: 03:20
solax.0 | 2022-06-24 13:21:01.512 | debug | nauticalDusk: 00:01
solax.0 | 2022-06-24 13:21:01.508 | debug | nightCalc started ...
solax.0 | 2022-06-24 13:21:01.506 | debug | System longitude: 5.606619 System latitude: 51.4393085
solax.0 | 2022-06-24 13:21:01.497 | debug | Solax is started in local-mode
solax.0 | 2022-06-24 13:20:59.134 | info | starting. Version 0.4.6 in /opt/iobroker/node_modules/iobroker.solax, node: v16.15.1, js-controller: 4.0.23

simatec, 2022-06-24:
which firmware version is on your wifi stick?  Is the cloud connection working?

louisvangeldrop, 2022-06-24:
Firmware version: 2.033.20
The cloud connection is working using my own software implementation in JS.

simatec, 2022-06-24:
Please post the Config from the Adapter and the Wifi Stick

Johan219, 2022-06-24:
If I can help, just let me know

louisvangeldrop, 2022-06-24:
You are welcome to help.

simatec, 2022-06-24:
@louisvangeldrop  If you have access to the Wifi Stick, please post the settings

Johan219, 2022-06-24:
Here is my debug log

2022-06-24 23:38:46.474  - debug: solax.0 (14966) Solax is started in local-mode
--
2022-06-24 23:38:46.477  - debug: solax.0 (14966) System longitude: 6.17887272535988 System latitude: 52.4028345
2022-06-24 23:38:46.478  - debug: solax.0 (14966) nightCalc started ...
2022-06-24 23:38:46.478  - debug: solax.0 (14966) nauticalDusk: 00:14
2022-06-24 23:38:46.479  - debug: solax.0 (14966) nauticalDawn: 03:02
2022-06-24 23:38:46.479  - debug: solax.0 (14966) current local Time: 23:38
2022-06-24 23:38:46.479  - debug: solax.0 (14966) calculate astrodata ...
2022-06-24 23:38:46.480  - debug: solax.0 (14966) Sun Altitude: -9.8°
2022-06-24 23:38:46.480  - debug: solax.0 (14966) Sun Azimut: 332.4°
2022-06-24 23:38:46.541  - debug: solax.0 (14966) Request Interval: 10 seconds
2022-06-24 23:38:46.541  - debug: solax.0 (14966) Local Request Interval started ...

Johan219, 2022-06-24:
I am also using firmware version: 2.033.20

Johan219, 2022-06-24:
[image]

Johan219, 2022-06-24:
[image]
[image]
[image]

louisvangeldrop, 2022-07-03:
Wifi-stick System page

DHCP Static IP System
Firmware Version: 2.033.20
MAC address: D0-BA-E4-39-07-6E
SN: SWJVGXVXPP
IP address: 192.168.0.86
SSID: myssid
 Find AP
Key: Unhide
save
Select language: English

Johan219, 2022-07-12:
@louisvangeldrop  @simatec The adapter is working again, data is being refreshed

simatec, 2022-07-12:
Thx for your Feedback

closed 2022-07-12 by simatec