simatec/ioBroker.solax #114: Bekomme den Pocket Wifi V3 nicht ans Laufen

gidhap, 2022-09-09:
Hallo, erst mal Vielen dank für die Arbeit mit dem Adapter !
Leider bekomme ich den Pocket Wifi V3 nicht ans Laufen.
Ich habe allerdings auch keine Seriennummer des Kommunikationsmoduls gefunden.
Der Pocket Wifi hat nur eine Registrierungsnummer.

Die Fehlermeldung ist entweder
solaxRequest: {"success":false,"exception":"Query success!","result":"no auth!"}
-- wenn ich die Seriennummer des X1 verwende
oder
solaxRequest: {"success":false,"exception":"Query success!","result":"this sn did not access!"}
-- wenn ich die Registrierungsnummer des Pocket Mini verwende

Mach ich hier was falsch, oder geht der Pocket Mini V3 nicht (noch nicht ;) ) ?

simatec, 2022-09-09:
Installiere mal direkt die Github Version. 
dort wurde Wifi v3 gerade intrigiert

gidhap, 2022-09-09:
Danke für die rasche Antwort. Ich habe jetzt die 0.4.12 installiert, bekomme dieselben Fehlermeldungen. Je nachdem welche Nummer ich eintrage.
Beim Installieren gab es ein paar warnings
`npm WARN crc@4.1.1 requires a peer of buffer@>=6.0.3 but none is installed. You must install peer dependencies yourself.

npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@2.3.2 (node_modules/fsevents):

npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.3.2: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})

npm WARN optional SKIPPING OPTIONAL DEPENDENCY: osx-temperature-sensor@1.0.7 (node_modules/osx-temperature-sensor):

npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for osx-temperature-sensor@1.0.7: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})`

simatec, 2022-09-09:
Eine 0.4.12 gibt es nicht. 
installiere bitte direkt aus dem iobroker die Github Version

gidhap, 2022-09-09:
sorry, eine Zeile verrutscht. Ich habe mit der URL vom GitHub installiert, ist 0.4.6.
[image]

gidhap, 2022-09-09:
...ausser daß er nicht connected ;) Ist aber nicht tragisch, ich warte einfach mal bis zur nächsten Version

simatec, 2022-09-09:
Ohne deine Seriennummer wirst du keine Verbindung zur Cloud bekommen

gidhap, 2022-09-09:
Die Frage ist welche Ser Nr?
Wenn ich die des Wechselrichters verwende bekomme ich

solaxRequest: {"success":false,"exception":"Query success!","result":"no auth!"}

Verwende ich die des Pocket Wifi bekomme ich
solaxRequest: {"success":false,"exception":"Query success!","result":"this sn did not access!"}

gidhap, 2022-09-09:
Ich denk eich habe mein Problem gefunden. Mir ist aufgefallen dass beim cut/paste der WR Seriennummer viel mehr Stellen eingefügt werden als die Nummer lang ist. Mehrfach versucht, immer dasselbe (Browser-?) Problem. Nachdem ich die Nummer händisch eingegeben habe, habe ich zwar nach 10 Minuten noch keine Werte, es wird aber auch keine Fehlermeldung ausgegeben. Ich warte jetzt einfach mal ab ob sich noch was tut.
`<html>
<body>

solax.0 | 2022-09-09 19:12:42.958 | debug | API Request started ...
-- | -- | -- | --
solax.0 | 2022-09-09 19:12:42.869 | debug | is Night: false
solax.0 | 2022-09-09 19:12:42.848 | debug | Sun Azimut: 272.2°
solax.0 | 2022-09-09 19:12:42.848 | debug | Sun Altitude: 5.1°
solax.0 | 2022-09-09 19:12:42.847 | debug | calculate astrodata ...
solax.0 | 2022-09-09 19:12:42.847 | debug | current local Time: 19:12
solax.0 | 2022-09-09 19:12:42.846 | debug | nauticalDawn: 05:43
solax.0 | 2022-09-09 19:12:42.846 | debug | nauticalDusk: 20:58
solax.0 | 2022-09-09 19:12:42.845 | debug | nightCalc started ...
solax.0 | 2022-09-09 19:07:42.958 | debug | API Request started ...
solax.0 | 2022-09-09 19:07:42.914 | debug | is Night: false
solax.0 | 2022-09-09 19:07:42.848 | debug | Sun Azimut: 271.3°
solax.0 | 2022-09-09 19:07:42.848 | debug | Sun Altitude: 6°
solax.0 | 2022-09-09 19:07:42.848 | debug | calculate astrodata ...
solax.0 | 2022-09-09 19:07:42.847 | debug | current local Time: 19:07
solax.0 | 2022-09-09 19:07:42.847 | debug | nauticalDawn: 05:43
solax.0 | 2022-09-09 19:07:42.847 | debug | nauticalDusk: 20:58
solax.0 | 2022-09-09 19:07:42.845 | debug | nightCalc started ...
solax.0 | 2022-09-09 19:02:42.957 | debug | Request Interval: 5 minute(s)
solax.0 | 2022-09-09 19:02:42.817 | debug | Sun Azimut: 270.4°
solax.0 | 2022-09-09 19:02:42.816 | debug | Sun Altitude: 6.8°
solax.0 | 2022-09-09 19:02:42.816 | debug | calculate astrodata ...
solax.0 | 2022-09-09 19:02:42.815 | debug | current local Time: 19:02
solax.0 | 2022-09-09 19:02:42.815 | debug | nauticalDawn: 05:43
solax.0 | 2022-09-09 19:02:42.815 | debug | nauticalDusk: 20:58
solax.0 | 2022-09-09 19:02:42.813 | debug | nightCalc started ...
solax.0 | 2022-09-09 19:02:42.812 | debug | System longitude: xxxx System latitude: xxxx
solax.0 | 2022-09-09 19:02:42.803 | debug | Solax is started in cloud-mode
solax.0 | 2022-09-09 19:02:40.452 | info | starting. Version 0.4.6 (non-npm: simatec/ioBroker.solax#fbb6dfa485d602b36c55bee9f9ba9cbf458f6c48) in /opt/iobroker/node_modules/iobroker.solax, node: v14.19.1, js-controller: 4.0.23

</body>
</html>`

simatec, 2022-09-09:
Naja aktuell sendet dein WR sicher nicht

gidhap, 2022-09-09:
korrekt, deshalb hatte ich das oben schon mal angemerkt. Heist also ich bekomme die Daten aus der Cloud nur wenn der WR aktiv ist. Was ja OK ist, man muss es nur wissen.

fuxx-consulting, 2022-09-09:
[image]
Mit der Firmwareversion kann ich auf Lokal ohne Cloud wechseln?

PhileMC, 2022-09-09:
Hallo, super Arbeit und **vielen Dank schonmal.** 

Habe den WIFI-Stick mit Version 3.001.02. und bekomme zuverlässig die Inverter-Daten.

simatec, 2022-09-09:
Ja das geht

PhileMC, 2022-09-09:
@gidhap: Also ich hatte für die Cloudabfrage die SN des Wifi-Stick verwendet.

https://www.solaxcloud.com/proxyApp/proxy/api/getRealtimeInfo.do?tokenId=(AUS DER CLOUD WEBSITE)&sn=(STEHT AUF DEM WIFI_STICK)

vorher halt über die Solax-App oder dem WebUI den Stick in den Lokales W-Lan bringen und alles einrichten damit er die Daten in die Cloud bringt........

gidhap, 2022-09-09:
@PhileMC Meine Daten sind in der Cloud, das ist alles OK.
Dein Linkbeispiel bring exakt dieselbe Fehlermeldung wie der Adapter.

Welchen Wifi Stick hast Du? [Den Pocket Wifi V3.0 ?](https://greenakku.de/Zubehoer/Systemueberwachung/Fernbedienung-Ueberwachung/solaX-Pocket-WiFi-Kommunikationsmodul-V3::3990.html) Die Seriennummern dieser Sticks beginnen mit SV...

PhileMC, 2022-09-09:
@gidhap

ah Okay, ne sorry hab die V2 Variante da ist aber die die Software Version 3.001.02. drauf.

Sorry

closed 2022-09-12 by simatec