simatec/ioBroker.solax #115: add Data from DTSU666-D (Smartmeter Solax)

PhileMC, 2022-09-09:
**Daten von Netz/Smartmeter werden nicht mit in den Adapter übertragen, 
diese wären sehr wichtig um eine vernünftige Nulleinspeisung zu realisieren.**

Im Cloud request sind diese Daten enthalten:

[image]

In der Solax-App können diese Daten ebenfalls bei der Lokalen Verbindung abgerufen werden:

[image]

Sowie auf der Website:

[image]

Da ich eine hohe Autarkie anstrebe und die Täglichen Überschüsse möglichst effizient nutzen möchte wäre ich euch unglaublich dankbar das ihr diese Daten in die json integriert.

Da ich leider nicht weiß wie ihr an die Daten rankommt und ein curl bei mir negativ ausfällt bin ich auf euch angewiesen.
Ich bin leider "nur" Hardware-Entwickler und nicht so tief in der Materie wir ihr, aber falls ich euch irgendwie behilflich sein kann meldet euch einfach und ich schau was ich tun kann.

LG Philipp

fuxx-consulting, 2022-09-10:
Moin, kann ich nur unterstützen. Zur Zeit verarbeite ich das json noch zusätzlich mit einem Blockly und habe die Daten, aber nur alle 5 Minuten aus der Cloud. Was aber auch grundsätzlich reicht. Wenn das im Adapter wäre könnte ich mir wieder ein Script sparen

simatec, 2022-09-12:
@PhileMC Schalte doch den Adapter bitte auf Debug und poste mal den log.
Der Adapter erstellt alle States von denen er daten erhält automatisch.
Bei mir kommen die Daten an.
Sind die Daten aus der Cloud allerdings mit dem Wert bei 0 legt der Adapter diese bei Abfrage nicht an.

PhileMC, 2022-09-12:
@simatec

Guten Morgen, du Schreibst OBEN : "Daten aus der Cloud", mir geht es allerdings um die LOKALE VERBINDUNG. Aus der Cloud kommen die Daten.

Der 5min Intervall (aus der Cloud) ist nur leider viel zu langsam um meine Heizstäbe (3x 1,4KW über Phasenabschnitt komplett Dimmbar) zu Regeln, damit mein Hausanschluss bei "0 Watt" bleibt.

Werde nach der Arbeit die Einstellung ändern und einen Log teilen.

Frage 1: Hier werden doch die Datenpunkte gesetzt?! Ist das nicht beschränkt auf die vorgegebenen?

[image]

Frage 2: Du änderst glaub gerade den Code, nach einem "Update des Adapters" müsste der request im Log kommen, oder?

[image]

simatec, 2022-09-12:
Das du lokal abfragst, konnte ich aus deinem Issue nicht erkennen.
Bitte installiere mal die aktuelle Github-Version und schaue an welcher Stelle diese Werte in der Debugausgabe von local request kommen.

Sieht in der debugausgabe in etwa so aus:

`local request: {"type":"X1-Boost-Air-Mini","SN":"xxxxxxxx","ver":"2.033.20","Data":[4.8,0,67,0,1.3,233.4,294,32,0.4,883.4,0,323,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,50,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,2,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[0.6,4,"X1-Boost-Air-Mini","xxxxxxxxxxxxxxx",1,2.12,0,1.32,0]}`

Wenn du die Daten Daten auswertest, beginne beim auszählen mit 0
In meinen Beispiel ist z.B. die AC Spannung mit 233.4 V Wert 5

PhileMC, 2022-09-12:
@simatec 

Habe die Daten!!!!!!!!

Wert 74 -----> feedinpower (in W) (Achtung: Wert kann auch negativ sein)
Wert 76 -----> feedinenergy (in Wh) (evtl. in kWh umrechnen)
Wert 78 -----> consumeenergy (in Wh) (evtl. in kWh umrechnen) 

[image]

simatec, 2022-09-12:
Was hast du für einen Wechselrichter?

PhileMC, 2022-09-12:
Solax X3 Mic

PhileMC, 2022-09-12:
X3-MIC/PRO

simatec, 2022-09-12:
Dann stimmen deine Daten aber nicht wirklich überein. Laut dem Log von dir sind die ersten Werte die einzelnen Phasen. 
Da müsste man jetzt mal genauer schauen, ob es an Wifi v3 oder an den WR Daten liegt. 

Hab hier ein ähnliches Issue offen. 
Denke da müssen wir noch nachbessern

PhileMC, 2022-09-12:
Hi, okay da muss ich mich morgen nochmal hinsetzen und ne Gegenüberstellung machen. Schaff ich heute leider nicht mehr. Danke schonmal

simatec, 2022-09-12:
Wäre sehr interessant, ob Wifi v3 die Daten anders liefert oder ob es am 3 phasigen WR liegt

PhileMC, 2022-09-12:
Hi, 
habs mir gerade nochmal angeschaut das passt soweit.

Dein Code:
[image]
Mein Request:
[image]

**Müssten nur noch die anderen 3 Werte dazu.**

PhileMC, 2022-09-12:
@simatec 

könntest du diese Zeile noch einfügen?

74: { name: `data.feedinpower`, description: `Feed in Power M1`,type: `number`, unit: `W´, role: `value.power`}, // `Feed in Power`: (561,`W`)

simatec, 2022-09-12:
Die 3 Werte nehme ich noch mit auf

simatec, 2022-09-12:
Sind deine anderen Werte alle Plausibel und stimmen mit den Werten aus der Cloud überein?

PhileMC, 2022-09-12:
@simatec 

Ja die Daten stimmen alle und sind plausibel:

0-2: Spannung der Phasen
3-5: Strom der Phasen
.
.
.
.
Passt alles Super

PhileMC, 2022-09-13:
74: { name: 'data.feedinpower', description: 'Feed in Power M1',type: 'number', unit: 'W', role: 'value.power' }, // Feed in Power: (561,'W')

76: { name: 'data.feedinenergy', description: 'Feed in Energy',type: 'number', multiplier: 0.01, unit: 'kWh', role: 'value.power' }, // Feed in Energy: (12.2,'kWh')

78: { name: 'data.consumeenergy', description: 'Consume Energy',type: 'number', multiplier: 0.01, unit: 'kWh', role: 'value.power' }, // Consume Energy: (7.8,'kWh')

simatec, 2022-09-13:
Teste mal bitte die aktuelle Github Version ... Dort sollte es nun laufen

PhileMC, 2022-09-13:
TOP!!!!! Es läuft.

Vielen Dank falls ich dir weiterhin behilflich sein kann, melde dich einfach.

Eckdaten: 
Solax X3-MIC 9KW mit DTSU666-D Smartmeter
ab der aktuellen Version des Adapters können die Daten des SmartMeters das über Modbus am WR hängt über die Lokale Verbindung (Solax WIFI V2 (SoftwareVers. 3.00302)) abgerufen werden.

Danke an die Entwickler

closed 2022-09-13 by PhileMC

simatec, 2022-09-20:
Kannst du mir eine konkrete Zahl benennen, ab wann die 65535 abgezogen werden müssen.
Ich muss das im Adapter mit größer oder kleiner als definieren

In der nächsten Version habe ich es wie folgt drin.
Wenn der Wert kleiner 0 ist, dann werden die 65535 abgezogen

PhileMC, 2022-09-20:
Ich hab das so gelöst.
Somit kann man ca. 30Kw in beide Richtungen.

[image]

simatec, 2022-09-21:
Sollte mit der Version 0.5.5 nicht mehr nötig sein