simatec/ioBroker.solax #116: JSON aus lokaler Anbindung nicht stimmig

fuxx-consulting, 2022-09-11:
Hallo,

hier das JSON aus der lokalen Anbindung

{"inverterSN":8,
"sn":"xxxxxxxxx",
"inverterType":14,
"inverterStatus":"Wait Mode",
"uploadTime":"Sun Sep 11 2022 08:07:27 GMT+0200 (Mitteleuropäische Sommerzeit)",
"connectType":"local",
"online":true,
"firmwareVersion":"3.003.02",
"totalSize":15,
"currentdc1":0,
"currentdc2":0,
"voltagedc1":0,
"voltagedc2":0,
"outputcurrent":0,
"acvoltage":0,
"acpower":0,
"inverterTemp":0,
"yieldtoday":0,
"yieldtotal":0,
"exportedPower":5971,
"powerdc1":3739,
"powerdc2":0,
"batteryVoltage":1,
"batteryCurrent":0,
"batteryPower":42,
"batteryTemperature":0,
"batteryCapacityRemainig":18,
"totalFeed":0,
"totalconsumption":4250,
"powernow":0,
"gridfrequency":2824}

hier das fast zeitgleiche JSON aus der API

{"success":true,"exception":"Query success!","result":{
"inverterSN":"xxxxxxxxx",
"sn":"xxxxxxx",
"acpower":0.0,
"yieldtoday":0.0,
"yieldtotal":0.0,
"feedinpower":0.0,
"feedinenergy":0.0,
"consumeenergy":0.0,
"feedinpowerM2":0.0,
"soc":98.0,
"peps1":0.0,
"peps2":0.0,
"peps3":0.0,
"inverterType":"14",
"inverterStatus":"107",
"uploadTime":"2022-09-11 08:05:40",
"batPower":0.0,
"powerdc1":0.0,
"powerdc2":46.0,
"powerdc3":null,
"powerdc4":null,
"batStatus":"0"}}

Die Werte passen leider überhaupt nicht. Der WR ist zur Zeit nur im EPS Modus da es morgen final an angeschlossen wird. sprich nur Akku und Solar. 

Werde nochmal gibt es dazu eine Idee woran es liegen kann?

VG und Danke für die tolle Arbeit

Andreas

fuxx-consulting, 2022-09-11:
anbei mal 2 Screenshots wo ich denke das die Werte im JSON falsch zugeordnet sind
Hier die Daten aus meiner Cloud
[image]
Hier aus dem Adapter
[image]

simatec, 2022-09-12:
Die lokale Verbindung wird je nach Einstellung alle 10 Sekunden abgefragt.
Deine Cloud holt sich dagegen nur alle 5 Minuten die Daten.
D.h. die Unterschiede sind normal, da deine Cloud nicht zwingend die gleiche Auswertungszeit hat.

fuxx-consulting, 2022-09-12:
Das ist richtig Niederhofen r die Werte stehen in den falschen datenpunkten

fuxx-consulting, 2022-09-12:
Hier mal ein Abbild zwischen beiden sind nicht mal Sekunden Unterschied. Ich denke dann weißt du was ich meine
[image]

simatec, 2022-09-12:
Welche Version hast du installiert? Hast du die Github Version installiert?
Ich denke die lokalen Daten scheinen bei deinem Wechselrichter anders zu sein, als bei Solax ...

fuxx-consulting, 2022-09-12:
ja die Github 4.6 
Firmware ist 3.003.02, Seriennummer beginnt mit SXX
Auch der Inverterstatus kommt dort als Wait mode an, ist aber EPS-Prüfmodus
hier jetzt nochmal mit den PV Daten habe mal einen günstigen Moment abgewartet.
[image]

simatec, 2022-09-12:
OK, wir können hier sehr gut sehen, dass die Ausgaben deines WR nicht mit dem von Solax übereinstimmen.
Installiere mal bitte die aktuelle Github-Version.
Da habe ich ein paar Debugausgaben drin, welche zeigen, was vom WR an Daten kommt und in welcher Reihenfolge.

Ich denke für deinen WR müssen wir einen separaten Adapter erstellen.
Wenn wir das in dem Solax einbauen, ist das sehr verwirrend für andere User.

Dafür benötige ich dann aber deine Hilfe.
Du müsstest anhand der Ausgaben eine Tabelle erstellen, an welcher Position in der der Ausgabe welcher Wert kommt.

Wenn ich das habe, dann könnte ich einen Adapter für deinen WR bauen

fuxx-consulting, 2022-09-12:
Achso, die Daten über die Clou von der selben adapter version stimmen bis auf einen. sind aber nur DC1 dc2 und soc

fuxx-consulting, 2022-09-12:
meinst du die 5.0 beta?

simatec, 2022-09-12:
Einfach die aktuelle Github Version ... und dann mal die Debugausgabe für "local Request" anschauen und posten

fuxx-consulting, 2022-09-12:
Du meinst die Ausgabe unter Protokolle?

simatec, 2022-09-12:
Ja genau

fuxx-consulting, 2022-09-12:
da kommt gar nichts
[image]
Solax1 ist der lokale
da kommt nur das
[image]

was auch richtig ist, dort steht nur 8 und nicht die SN

fuxx-consulting, 2022-09-12:
ok man lernt ja nie aus

[iobroker.2022-09-12.log.zip](https://github.com/simatec/ioBroker.solax/files/9547181/iobroker.2022-09-12.log.zip)

fuxx-consulting, 2022-09-12:
und hier nochmal alles was ich schon zuordnen konnte. verstehe nur nicht warum meiner anders sein sollte als andere. oder liegt das an der Firmewareversion?

[image]

simatec, 2022-09-12:
Damit kann ich leider so nicht arbeiten ...
Du musst im log nach dem Punkt "local request" suchen und dort die Werte der Reihenfolge zuordnen ...
Dabei musst du mit dem ersten Wert bei 0 beginnen.

Am besten du machst eine Tabelle in der alle Werte zugeordnet werden.

Da andere auch die Wifi v3 nutzen, gehe ich davon aus, dass dein WR andere Daten sendet. Auch die Umrechnung gibt es bei Solax nicht.

Hier ein Beispiel:

`local request: {"type":"X1-Boost-Air-Mini","SN":"xxxxxxxxx","ver":"2.033.20","Data":[7.8,0,61.4,0,1.9,233.8,430,37,1.2,884.2,0,480,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,49.97,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,2,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[0.6,4,"X1-Boost-Air-Mini","xxxxxxxxxx",1,2.12,0,1.32,0]}`

simatec, 2022-09-12:
Deine Ausgabe im Log sieht wie folgt aus:

`local request: {"sn":"xxxxxxxxx","ver":"3.003.02","type":14,"Data":[2298,2315,2302,97,62,56,2228,1423,1292,4943,5195,3240,81,30,4241,996,5000,5000,5001,2,0,0,0,0,0,0,0,0,0,0,0,0,0,1,3649,0,0,0,0,42980,0,0,4275,0,0,1,49,1294,256,8984,3084,5641,100,0,54,0,0,0,0,0,0,0,0,0,0,0,0,0,71,0,71,0,0,0,0,0,123,0,0,12,278,0,92,0,0,0,597,0,12,0,597,0,11,0,0,0,0,0,0,0,0,0,1,100,1,30,122,256,4672,3200,0,300,227,214,34,33,1,2132,778,14135,14135,14135,14135,0,0,3434,3340,42869,0,20564,12339,18753,12852,18742,12356,13616,20564,12339,18754,12595,18743,12612,13619,20564,12339,18754,12595,18744,13124,14389,20564,12339,18754,12595,18744,13124,12592,20564,12339,18754,12595,18744,13124,13876,4099,4355,1284,1538,0,42980,0,0,0,8,0,0,4,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[15,14,"xxxxxxxxxxxxx",8,1.2,0,1.2,1.09,0,1]}`

fuxx-consulting, 2022-09-12:
Ok ich gehe das mal durch Danke

simatec, 2022-09-12:
Ist ne Fleißaufgabe, aber ich kann es leider aufgrund der fehlenden Hardware nicht

fuxx-consulting, 2022-09-12:
Kein Problem versuche da mal die Daten zu extrahieren

fuxx-consulting, 2022-09-13:
Wäre es evtl. Nicht einfacher die Q.HOME Geräte in einen eigenen Adapter zu packen. Die Geräte werden ja immer mehr. Extra für mich nur anpassen ist ja wenig effizient. Die Q.HOME sind zwar Solax Geräte aber scheinen die Werte in einer anderen Reihenfolge zu haben.

simatec, 2022-09-13:
Das hatte ich ja weiter oben geschrieben. 
aber dafür brauche ich die Werte in der richtigen Reihenfolge. 
dann kann ich dazu einen eigenen Adapter bauen.

fuxx-consulting, 2022-09-13:
Ok hatte ich falsch verstanden

fuxx-consulting, 2022-09-14:
Hallo,
Fleißarbeit fürs 1. beendet. Kommt sicherlich noch etwas dazu oder muss korrigiert werden. 

Zum Beispiel Eigennutzung PV müsste man sich berechnen aus Wert (70) - Wert (90)

Bei 3 Werten muss wenn bestimmte Werte größer 0 sind 65536 abgezogen werden. Geht um Einspeisung/Netzbezug sowie beim Akku um Strom und Leistung jeweils ob laden oder entladen. Hoffe ich habe das vernünftig erklärt.

[Mapping QCELL.zip](https://github.com/simatec/ioBroker.solax/files/9569460/Mapping.QCELL.zip)

VG
Andreas

svenomatt, 2022-09-15:
Hallo zusammen,

habe den aktuellen Adapter 0.51 ebenfalls am laufen und ähnliches wenn nicht sogar die selben fehlerhaften Meldung wie fuxx-consulting

[image]

[image]

nutze den Wifistick V3 an einem Solax Mini 1.1

fuxx-consulting, 2022-09-15:
Hallo,

der X1 ist doch ein 1Phasen WR oder? Sieht aber vom Bild her sehr ähnlich aus. Wenn du mir den Request schickst bzw. hier reinstellst kann ich recht einfach abgleichen ob das mi meiner gegenüberstellung schon hilft. in den Instanzen beim Sloax Adaper in den Experteneinstellungen auf local request umstellen und dann in den protokollen mir das JSON hier reinschreiben. Vergleiche ich dann mit meinem

svenomatt, 2022-09-16:
Hoffe das du das gemeint hast!?

{
  "inverterSN": 8,
  "sn": "XXXXXXXXXXXX",
  "inverterType": "X1-Boost/Air/Mini",
  "inverterStatus": "Wait Mode",
  "uploadTime": "Fri Sep 16 2022 10:24:37 GMT+0200 (Mitteleuropäische Sommerzeit)",
  "connectType": "local",
  "online": true,
  "firmwareVersion": "3.003.02",
  "totalSize": 1.1,
  "currentdc1": 2325,
  "currentdc2": 24,
  "voltagedc1": 560,
  "voltagedc2": 1016,
  "outputcurrent": 0,
  "acvoltage": 57,
  "acpower": 0,
  "inverterTemp": 587,
  "yieldtoday": 0,
  "yieldtotal": 4998,
  "exportedPower": 2,
  "powerdc1": 2107,
  "powerdc2": 0,
  "batteryVoltage": 7,
  "batteryCurrent": 0,
  "batteryPower": 0,
  "batteryTemperature": 0,
  "batteryCapacityRemainig": 0,
  "totalFeed": 593,
  "totalconsumption": 0,
  "powernow": 0,
  "gridfrequency": 0
}

fuxx-consulting, 2022-09-16:
leider nein. 
in den Instanzen so einstellen Deine wird wohl die solax.0  sein
[image]

Dann unter Protokolle auf Solax einstellen und das im Bild markierte kopieren. 
[image]

svenomatt, 2022-09-16:
OK, dann sollte dies das Richtige sein

local request: {"sn":"XXXXXX","ver":"3.003.02","type":4,"Data":[2334,32,752,1029,0,77,0,792,0,4999,2,2116,0,16,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,32,0,595,0,0,0,0,0,0,0,0,0,0,0,0,0,36,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[1.1,4,"XM3A11I4387005",8,2.22,0,1.4,0,0,1]}

fuxx-consulting, 2022-09-16:
Sieht gut aus schaue ich mir an

fuxx-consulting, 2022-09-16:
So ist doch deutlich kürzer als meiner, denke aufgrund der Thematik, dass es ein 1 Phasen WR ist.
Da müsste man manuell ran. Ich habe das so gelöst, das ich das Intervall auf 5 Sekunden gestellt habe und so oft neu gestartet bis das mit den Daten aus der Clou ziemlich gleichzeitig war.
[image]

habe dann die Daten aus dem Json mit denen der Cloud verglichen und konnte so nach und nach die Werte nachvollziehen.

um das JSON zu formatieren habe ich diese Seite genutzt http://jsonviewer.stack.hu/

@simatec habe heute die Solax Cloud Monitoring API Beschreibung bekommen, kannst du die gebrauchen? da sind jetzt auch die neuen Modelle drin bis Invertertype 21 drin

fuxx-consulting, 2022-09-18:
Hallo @simatec ,

ich habe mal an der angehängten main.js ein wenig etwas geändert. Änderungen sind dokumentiert. 
1. die neuen Invertertypen integriert
2. unter Expert Local Mode -> information_dataPoints einen Bereich 4: eingefügt. Ist zwar weitestgehend stimmig mit 3: aber dsp und arm kommen in der local request nicht vor. Ich habe DSP 2.07 und ARM 2.03 Diese Werte gibt es im local request nicht
3. unter Expert Local Mode -> data_dataPoints einen Bereich 4: eingefügt, da fast gar nicht mit 3: vereinbar. 

Da ich nicht weiß bzw. verstehe, wo du welche JSON-Ausgabe welchem Invertertyp zuordnest war ich an der Stelle raus. Aktuell zieht er sich bei Invertertyp 14 unter LocalMode auf die 1:
Was mir im local Request fehlt ist der timestamp

Gib aktuell folgende Besonderheiten unter data_data. Sind aber auch in der Datei dokumentiert.
Wert:
34: Einspeiseleistung kann auch Netzbezug sein. Dann muss vom Wert 65536 abgezogen werden. Wenn 34:>32768 dann 34-65536 (positiv ist einspeisen, negativ ist Netzbezug)
40: Strom Batterie wenn 40: >32768 dann 40: -65536 (positiv ist laden, negativ ist entladen)
41: Leistung Batterie wenn 41: >32768 dann 41: -65536 (positiv ist laden, negativ ist entladen)

Warum 32768? Die Konstante die abgezogen werden kann ist 65536 also muss die Mitte die Trennung sein zwischen positiv und negativ dachte ich mir

Es gibt noch weitere Werte, welche aber berechnet werden müssten. 
Zum Beispiel
Eigennutzung Tag 70: - 90: (Ertrag-Einspeisung)
Eigennutzung Total: 68: - 86:
Stromverbrauch Tag: 92: + Eigennutzung Tag
Stromverbrauch Total: 88 + Eigennutzung Total

In wie weit man das über den Adapter regeln kann/muss oder ich später als Blockly kann ich nicht sagen.

Ich hoffe du versteht mein Anfängergebrabbel und es hilft dir bei der Anpassung. 

Viele Grüße
Andreas
[main.zip](https://github.com/simatec/ioBroker.solax/files/9594262/main.zip)

simatec, 2022-09-20:
Teste mal bitte die Version 0.5.2.

> Es gibt noch weitere Werte, welche aber berechnet werden müssten.
Zum Beispiel
Eigennutzung Tag 70: - 90: (Ertrag-Einspeisung)
Eigennutzung Total: 68: - 86:
Stromverbrauch Tag: 92: + Eigennutzung Tag
Stromverbrauch Total: 88 + Eigennutzung Total

Hier ist sicher ein Blockly sinnvoll

fuxx-consulting, 2022-09-20:
Passt noch nicht. Wenn du mir schreibst in welcher Datei ich die Daten anpassen muss bereite ich dir das auf Gerne per Mail an andreas@grambow-online.de
Würde dich da gerne unterstützen, da es ja rel. viel Arbeit ist

simatec, 2022-09-20:
Was passt denn nicht?
Hast du die Github-Version installiert und im Anschluss einen Upload gemacht?

fuxx-consulting, 2022-09-20:
Ja, 
Erst die 5.02 dann die 5.03 was meinst du mit Upload?

fuxx-consulting, 2022-09-20:
Alles klar teste ich nachher. Danke für den Hinweis.

simatec, 2022-09-20:
Wenn nicht sag mal Bescheid .... Dann passe ich nochmal was an

simatec, 2022-09-20:
Teste mal bitte die 0.5.4

fuxx-consulting, 2022-09-20:
Hallo, so es passt soweit alles, Es sieht aber so aus, als ob die Daten vom 1: da mit reinlaufen. 

solax.1.data.batteryCurrent müsste negativ sein ist aber 648.36. Ich nehme an das dort ein Fehler in der Formel ist muss eigentlich -65536 sein

anbei eine PDF. alle grünen Werte sind die von 4: der gelbe ist der fehlerhafte und die nicht markierten gehören nicht dazu sind wie geschrieben vermutlich aus 1:

[JSON-054.pdf](https://github.com/simatec/ioBroker.solax/files/9609785/JSON-054.pdf)

VG
Andreas

simatec, 2022-09-20:
Stelle den Adapter mal auf Cloud und speichere. Dann wieder zurück auf lokal und erneut speichern. 
damit sind die doppelten Datenpunkte weg. 

Mit dem Wert 1 gast du aber in der Übermittlung deiner Daten nix geschrieben.

fuxx-consulting, 2022-09-20:
jetzt passt alles außer solax.1.data.batteryCurrent wenn >32768 dann muß -65536 gerechnet werden. denke da ist ein Zahlendreher drin. alles andere sieht soweit richtig aus

fuxx-consulting, 2022-09-20:
Vielen Dank 
Das ist eine tolle Arbeit. Da ich jetzt einen einfachen Weg der Zuordnung von Daten gefunden habe, falls du mal neue Inverter aufnehmen willst, kann ich dabei gerne unterstützen.

fuxx-consulting, 2022-09-20:
@svenomatt 
in der Weboberfläche unter Wechselrichter gibt es einen Bereich statistischer Bericht. Den exportieren und mit dem zum selben Zeitpunkt erzeugten local request vergleichen, dann kann man das einfach und schnell zuordnen. Bin auch jetzt erst auf diesen sehr einfachen Weg gestoßen.

fuxx-consulting, 2022-09-20:
@simatec habe den Fehler glaube ich gefunden. 

er rechnet mit dem multiplikator, dann ist der Wert unter 32768 und er zieht die 65536 nicht mehr ab.

Also erst wenn größer 32768 die 65536 abziehen und dann den Multiplicator. Dann stimmt es denke ich

Evtl. den Schwellenwert für alle auf 32768 oder 32767 setzen, habe in geschlossenen Issue gesehen das bei dem Beispiel mit 32767 gerechnet worden ist

fuxx-consulting, 2022-09-20:
Was ich gerade noch gesehen habe, der Inverterstatus wird als Zahl eingetragen nicht in Wortlaut 
[image]

fuxx-consulting, 2022-09-21:
Hallo,

anbei eine angepasste main.js. Änderungen in Zeile 540, 542, 543 und 609 Schau dir gerne mal den Lösungsansatz bzgl. BatteryCurrent an.

[main.zip](https://github.com/simatec/ioBroker.solax/files/9612258/main.zip)

bzgl. dem Inverterstatus, kann es sein, das was in Zeile 280-282 für den Cloudmodus im Lokalmodus fehlt?  

Im Cloudmodus scheint das definiert zu sein, im local nicht. Ich glaube ich muss auf meine alten Tage noch anfangen JS zu lernen. das macht Spaß

VG
Andreas

simatec, 2022-09-21:
Teste mal die Version 0.5.5 ... Da habe ich einige Bugfixes ... Die main.js brauchst du hier nicht schicken.
Wenn dir was auffallen sollte, einfach mal die Zeile als Link posten

fuxx-consulting, 2022-09-21:
Ok mache ich. Danke

simatec, 2022-09-21:
> OK, dann sollte dies das Richtige sein
>
> local request: [...]

@svenomatt Kannst du dir bitte mal die Mühe machen und die Daten mit deiner Cloud vergleichen.
Ich benötige die Reihenfolge für die Einbingung in den Adapter.

D.h. alles was oben in Data steht, muss abglichen und definiert werden.

Beispiel:

0: acvoltage
8: acfrequency (Faktor: 0,01

simatec, 2022-09-21:
@svenomatt Teste mal bitte die v0.5.6
Habe mal versucht aus deinen Daten etwas abzuleiten

svenomatt, 2022-09-22:
Moin zusammen,
habe gerde mal den adapter auf 0.56 gebracht, sieht es mal sehr gut aus.
 
Erstmal Vielen Dank erstmal dafür 

das andere @fuxx-consulting schau ich mir die nächsten Wochen an, bin gerade etwas in Zeitnot.

closed 2022-10-11 by simatec