simatec/ioBroker.solax #117: I can't downgrade

nykroy, 2022-09-17:
**Describe the bug**  
I tried to downgrade my pocket WiFi V2 versión 3.001.002 but It failed 
[image]

**To Reproduce**  

**Expected behavior**  

**Screenshots & Logfiles**  

**Versions:**  
 
[image]

**Additional context**  
Add any other context about the problem here.

simatec, 2022-09-17:
why a downgrade?  The version should fit.

nykroy, 2022-09-17:
I tried to upload the .usb that you add to GitHub for access to the local data from x1 booster but this is is the response. I understand that I need downgrade with this file for access the data from inversor. Is not right?

simatec, 2022-09-18:
your stick already has the v3 and should be suitable.  This is described in the documentation

simatec, 2022-09-20:
The documentation states that the firmware for Wifi V3 must not be less than 3.001.

closed 2022-10-11 by simatec