simatec/ioBroker.solax #144: Exported Power Value

Charlie-2222, 2023-01-22:
It seems to be that the "Exported Power" value is not translated correctly.
I have a 3 phase solax meter SMD630 installed directly at house connection, from wicht the solax inverter might get the value.

In the "solax.0.data.exportedpower" object I permannently get a value near 65535 W (hex FFFF) even in the late afternoon when there is very less harvesting and less power consumation.
When I switch on a 1000 W devices, the values decreases ! down to about 64500 W
However I expect the correct value for imported/exported power including localy consumed power 
It seems to me that here is just somethink incorrect with the sining for that value

The Solax Web Interface is showing all corret values

I have a X1-Boost/Air/Mini with the latest Adpater Version 0.5.7 an a most current ioBroker environment.
I have configured local access via WIFI 3.003.02 Stick, witch was working pretty well so far 

I am happy to test an update or provide more information

Thanks

simatec, 2023-03-04:
Bitte gib mir mehr Infos zu den Werten, damit ich die Umrechnung anpassen kann

simatec, 2023-03-04:
Fix in v0.6.0

closed 2023-03-04 by simatec