simatec/ioBroker.solax #149: negative Werte

Andre-0815-4711, 2023-02-17:
negative Werte werden mit z.B. 65148 A angezeigt

hinzufügen von "maxValue: 32768" zu den entsprechenden Datenpunkten

simatec, 2023-03-04:
Ein wenig mehr Infos wären super

Andre-0815-4711, 2023-03-04:
Wenn der Wechselrichter z.B. den Akku aus dem Netz lädt oder ein weiterer Wechselrichter mit in das Hausnetz einspeist, gehen die Werte der einzelnen Phasen ins negative.
In der main.js habe ich gesehen, dass z.B. bei FeedIn ein "maxValue: 32768" angegeben ist.
Dies habe ich bei allen Werten ergänzt die auch negativ sein können.

simatec, 2023-03-04:
Zeige mir bitte, um welche Datenpunkte es sich handelt

simatec, 2023-03-04:
Und bitte gib auch den Wechselrichter Typ an

Andre-0815-4711, 2023-03-04:
Moin,

ich habe bei mir lokal unter X3-Hybrid-G4 folgende Anpassung vorgenommen:

`    4: { /****************************************** X3-Hybrid-G4 *****************************************/
        isOnline: { name: 'info.online', description: 'Inverter Online', type: 'boolean', role: 'switch' },
        0: { name: 'data.acvoltage1', description: 'Grid Voltage 1', type: 'number', multiplier: 0.1, unit: 'V', role: 'value.power' }, // 'AC Voltage 1': (0, 'V'),
        1: { name: 'data.acvoltage2', description: 'Grid Voltage 2', type: 'number', multiplier: 0.1, unit: 'V', role: 'value.power' }, // 'AC Voltage 2': (1, 'V'),
        2: { name: 'data.acvoltage3', description: 'Grid Voltage 3', type: 'number', multiplier: 0.1, unit: 'V', role: 'value.power' }, // 'AC Voltage 3': (2, 'V'),
        3: { name: 'data.accurrent1', description: 'Grid Current 1', type: 'number', maxValue: 32768, multiplier: 0.1, unit: 'A', role: 'value.power' }, // 'Output Current 1': (3, 'A'),
        4: { name: 'data.accurrent2', description: 'Grid Current 2', type: 'number', maxValue: 32768, multiplier: 0.1, unit: 'A', role: 'value.power' }, // 'Output Current 2': (4, 'A'),
        5: { name: 'data.accurrent3', description: 'Grid Current 3', type: 'number', maxValue: 32768, multiplier: 0.1, unit: 'A', role: 'value.power' }, // 'Output Current 3': (5, 'A'),
        6: { name: 'data.acpower1', description: 'Grid Power 1', type: 'number', maxValue: 32768, unit: 'W', role: 'value.power' }, // 'AC Power 1': (6, 'W'),
        7: { name: 'data.acpower2', description: 'Grid Power 2', type: 'number', maxValue: 32768, unit: 'W', role: 'value.power' }, // 'AC Power 2': (7, 'W'),
        8: { name: 'data.acpower3', description: 'Grid Power 3', type: 'number', maxValue: 32768, unit: 'W', role: 'value.power' }, // 'AC Power 3': (8, 'W'),
        9: { name: 'data.acpower', description: 'Inverter AC-Power now', type: 'number', maxValue: 32768, unit: 'W', role: 'value.power' }, // 'AC Power': (80, 'W'),
        10: { name: 'data.dcvoltage1', description: 'PV1 Voltage', type: 'number', multiplier: 0.1, unit: 'V', role: 'value.power' }, // 'PV1 Voltage': (9, 'V'),
        11: { name: 'data.dcvoltage2', description: 'PV2 Voltage', type: 'number', multiplier: 0.1, unit: 'V', role: 'value.power' }, // 'PV2 Voltage': (10, 'V'),
        12: { name: 'data.dccurrent1', description: 'PV1 Current', type: 'number', multiplier: 0.1, unit: 'A', role: 'value.power' }, // 'PV1 Current': (11, 'A'),
        13: { name: 'data.dccurrent2', description: 'PV2 Current', type: 'number', multiplier: 0.1, unit: 'A', role: 'value.power' }, // 'PV2 Current': (12, 'A'),
        14: { name: 'data.dcpower1', description: 'PV1 Power', type: 'number', unit: 'W', role: 'value.power' }, // 'PV1 Power': (13, 'W'),
        15: { name: 'data.dcpower2', description: 'PV2 Power', type: 'number', unit: 'W', role: 'value.power' }, // 'PV2 Power': (14, 'W'),
        16: { name: 'data.acfrequency1', description: 'Grid Frequency 1', type: 'number', multiplier: 0.01, unit: 'Hz', role: 'value.power' }, // 'Grid Frequency 1': (15, 'Hz'),
        17: { name: 'data.acfrequency2', description: 'Grid Frequency 2', type: 'number', multiplier: 0.01, unit: 'Hz', role: 'value.power' }, // 'Grid Frequency 2': (16, 'Hz'),
        18: { name: 'data.acfrequency3', description: 'Grid Frequency 3', type: 'number', multiplier: 0.01, unit: 'Hz', role: 'value.power' }, // 'Grid Frequency 3': (17, 'Hz'),
        19: { name: 'info.inverterStatus', description: 'Inverter Mode', type: 'string', role: 'text' }, // 'Inverter Mode': (18, ''),
        34: { name: 'data.feedinpower', description: 'Feed in Power M1', type: 'number', maxValue: 32768, unit: 'W', role: 'value.power' }, // Feed in Power: (561,'W'),
        39: { name: 'data.batteryVoltage', description: 'battery voltage', type: 'number', multiplier: 0.01, unit: 'V', role: 'value.power' }, // 'Battery DC Voltage',
        40: { name: 'data.batteryCurrent', description: 'battery current', type: 'number', maxValue: 32768, multiplier: 0.01, unit: 'A', role: 'value.power' }, // 'Battery Current,
        41: { name: 'data.batteryPower', description: 'battery power', type: 'number', maxValue: 32768, unit: 'W', role: 'value.power' }, // 'Battery Power,
        47: { name: 'data.powerConsumer', description: 'Consumer power', type: 'number', maxValue: 32768, unit: 'W', role: 'value.power' }, // 'Consumer Power  ,   
        68: { name: 'data.yieldtotal', description: 'Inverter AC-Energy out total', type: 'number', multiplier: 0.1, unit: 'kWh', role: 'value.power.consumption' }, // 'Total Energy': (19, 'kWh'),
        70: { name: 'data.yieldtoday', description: 'Inverter AC-Energy out Daily', type: 'number', multiplier: 0.1, unit: 'kWh', role: 'value.power.consumption' }, // 'Today's Energy': (21, 'kWh'),
        80: { name: 'data.totalpvenergy', description: 'Total PV Energy', type: 'number', multiplier: 0.1, unit: 'kWh', role: 'value.power.consumption' }, // 'Total Consumption': (42, 'kWh'),
        82: { name: 'data.pvenergy', description: 'PV Energy', type: 'number', multiplier: 0.1, unit: 'kWh', role: 'value.power.consumption' }, // 'Total Consumption': (42, 'kWh'),
        86: { name: 'data.totalFeed', description: 'Total Feed-in Energy', type: 'number', multiplier: 0.01, unit: 'kWh', role: 'value.power.consumption' }, // 'Total Feed-in Energy': (41, 'kWh'),
[... cut ...]
        92: { name: 'data.consumeenergy', description: 'Consume Energy', type: 'number', multiplier: 0.01, unit: 'kWh', role: 'value.power' }, // Consume Energy: (7.8,'kWh'),
        103: { name: 'data.batteryCapacityRemainig', description: 'battery capacity remainig', type: 'number', unit: '%', role: 'value.power' }, // 'Battery Capacity Remainig,
        105: { name: 'data.batteryTemperature', description: 'battery temperature', type: 'number', unit: '      C', role: 'value.temperature' }, // 'Battery Temperature,`

simatec, 2023-03-04:
Fix in v0.6.0

closed 2023-03-04 by simatec