simatec/ioBroker.solax #150: Wifi 3.0 dongle

louisvangeldrop, 2023-02-27:
I am using a "blue" usb 3.0 Wifi dongle. The firmware version is 3.001.04. Th wifi accesspoint is Wifi_SRxxxxxxx  and the ip-addres is 192.168.10.10.

The Solax adapter doesn't return the inverter-data or any info values

simatec, 2023-03-04:
Please more informatiions...

louisvangeldrop, 2023-03-06:
I use a USB3.0 Wifi dongle with firmware version 3.01.04. The dongle uses 192.168.10.10 as  AP ip-address iso the 5.8.8.8 ip-address as used by the usb 2.0 dongle.
In the Solax Adapter code the last address is used in  the POST-request. 
If I change 5.8.8.8 by 192.168.10.10 , I do not get any data. unfotunately.

duke7901, 2023-03-11:
Also have a 3.0 on FW 3.001.02

Via 5.8.8.8 I'll receive nothing.
Only via local IP by curl.

`curl -d "optType=ReadRealTimeData&pwd=PASSWORD" -X POST http://192.168.x.x`

**Result:**
```
{"sn":"xxxxxxxxxx","ver":"3.008.10","type":4,
"Data":[2365,0,0,847,0,0,0,0,0,4998,0,63,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,63,0,0,0,0,0,0,0,0,0,0,0,0,0,9,
0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],
"Information":[1.100,4,"xxxxxxxxxx",8,2.29,0.00,1.45,0.00,0.00,1]}
```

I started decrypting the Data String...

I'm not using this ioBroker Module yet - but I'd like to..

simatec, 2023-04-26:
Please test v0.7.0

closed 2023-05-04 by simatec