simatec/ioBroker.solax #154: API call fails

andispammonyer, 2023-03-04:
**Describe the bug**  
I use your ioBroker adapter since one year for my two Solax inverters. One Hybrid one from 2019 and the other is a X1 3,6-T.
Since yesterday 2 pm I don't receive any updates from the solax cloud.
There are no error messages seen in ioBroker log. 

If I go manually to solax cloud or look into my solax app I see the latest values how many power is available. Only ioBroker adapter is affected.

I restarted my inverters already and restartet the iobroker Adapter multiple times. I also installed another instance of the solax adapter and entered again my credentials.

**Versions:**  
 - Adapter version: 0.5.7
 - JS-Controller version: 4.0.24
 - Node version: 16.16.0
 - Operating system: linux / docker

simatec, 2023-03-04:
Solax Api have a Problem

`{"exception":"The server is busy. Please try again later!","success":false}`

simatec, 2023-03-04:
url changed in v.0.6.0

closed 2023-03-04 by simatec

andispammonyer, 2023-03-04:
It works again. Thanks a lot!!!