simatec/ioBroker.solax #155: Bekomme keine LOKALEN Daten mehr vom WIFI Stick

Charlie-2222, 2023-03-16:
Ich betreibe den Adapter im LOCAL (expert) Mode
Der Adapter hat 5 Monate perfekt funktioniert. Seit einer Woche bekomme ich aber, im LOCAL Expoert Mode, keine Daten mehr vom WIFI Stick
Mir einem "curl -d "optType=ReadRealTimeData&pwd=xxxxxxxxx" -X POST http://192.168.12.xxx" in der Komandozeile auf einem PC bekomme ich die Daten
Nach Installation einer neuen 2. Instanz bekomme ich Daten aus der Solax Cloud, allso funktioniert der Stick offensichtlich. - leider werden die Daten aus der Cloud nur im 5 Minuten Takt aktualisiert.
Ich kann nicht mehr nachvollziehen ob und was sich geändert haben soll. 

Ich habe schon alles probiert, restore über den Backup Adapter,
Neuinstallation mit "iobroker/iobroker 7.2.0 Docker" Image auf meiner QNAP
Alle Solax Objekte gelöscht , Alles upgedated ...
Ich habe nun keine Idee mehr, da auch das Log trotz "Silly" Log Level keine Informationen liefert

Was kann ich noch tun ?

Das ist alles was ich nach einem Neustart im Log sehe:
solax.0 | 29152 | 2023-03-16 21:55:57.496 | info | starting.  Version 0.6.0 (non-npm:  simatec/ioBroker.solax#1e2f5bcbe778bbbb9f881ed1f4a094d9fb2e29f8) in  /opt/iobroker/node_modules/iobroker.solax, node: v16.19.1,  js-controller: 4.0.24
-- | -- | -- | -- | --
solax.0 | 28103 | 2023-03-16 21:55:52.803 | info | Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason
solax.0 | 28103 | 2023-03-16 21:55:52.801 | info | terminating
solax.0 | 28103 | 2023-03-16 21:55:52.799 | info | Got terminate signal TERMINATE_YOURSELF

**Versions:**  
Solax Adapter 0.5.7 und 0.6.0
Wifi Adapter Version 3.0.0.21 (Config Seiten sind im lokalen Netz erreichbar)

Plattform:    docker
Betriebssystem:    linux
Architektur:    x64
CPUs:    4
Geschwindigkeit:    3112 MHz
Modell:    AMD Embedded R-Series RX-421ND
RAM:    19.51 GB
System-Betriebszeit:    16 T. 19:56:08
Node.js:    v16.19.1
time:    1678562707284
timeOffset:    -60
Anzahl der Adapter:    470
NPM:    8.19.3
Datenträgergröße:    580.33 GB
freier Festplattenspeicher:    119.01 GB
Betriebszeit:    00:40:43
Aktive Instanzen:    20
Weg:    /opt/iobroker/
Hostname:    iobroker-2
js-Controller: 4.0.24

simatec, 2023-03-17:
Stelle den Adapter in den Instanzen auf Debug-Log und schaue dir den Log an.
Ansonsten würde ich sagen, dass dein Log nicht nach silly ausschaut. Wo hast du denn genau silly eingestellt?

simatec, 2023-03-17:
Beim Neustart des Adapters sollte im Debug-Level es wie folgt ausschauen:

`
2023-03-17 15:09:16.565 - debug: solax.0 (209231) Solax is started in local-mode
2023-03-17 15:09:16.567 - debug: solax.0 (209231) System longitude: 11.xxxxxxx System latitude: 52.xxxxxxxx
2023-03-17 15:09:16.567 - debug: solax.0 (209231) nightCalc started ...
2023-03-17 15:09:16.567 - debug: solax.0 (209231) nauticalDusk: 19:35
2023-03-17 15:09:16.567 - debug: solax.0 (209231) nauticalDawn: 05:13
2023-03-17 15:09:16.567 - debug: solax.0 (209231) current local Time: 15:09
2023-03-17 15:09:16.568 - debug: solax.0 (209231) calculate astrodata ...
2023-03-17 15:09:16.568 - debug: solax.0 (209231) Sun Altitude: 26°
2023-03-17 15:09:16.568 - debug: solax.0 (209231) Sun Azimut: 227.5°
2023-03-17 15:09:16.622 - debug: solax.0 (209231) Request Interval: 30 seconds
2023-03-17 15:09:16.622 - debug: solax.0 (209231) Local Request Interval started ...
2023-03-17 15:09:17.878 - debug: solax.0 (209231) local request: {"type":"X1-Boost-Air-Mini","SN":"xxxxxxxxx","ver":"2.033.20","Data":[1.8,0,70.4,0,0.6,234.8,119,26,1.9,1060.3,0,130,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,50.04,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,2,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[0.6,4,"X1-Boost-Air-Mini","xxxxxxxxxxxx",1,2.12,0,1.32,0]}
`

Charlie-2222, 2023-03-18:
Danke für dein Feedback, du hast natürlich recht, das "Silly" ging bei der Fehleranalyse (reinstall) verloren. - hätte ich auch merken können :-(
"Silly" Log ist nun unten angehängt.
Es scheint als ob aber keine Daten im ioBroker zurückkommen !? (Die Abfrage steht auf 5 Sec Intervall)

Ich habe CURL direkt auf einer Terminal Session des iobroker Containers ausgeführt 
Hier bekomme ich valide Daten zurück:

curl -d "optType=ReadRealTimeData&pwd=xxxxxxx" -X POST http://192.168.12.xxx
{"sn":"xxxxxxxx","ver":"3.008.10","type":4,"Data":[2370,51,1102,1882,2190,46,13,877,291,5003,2,7031,0,17,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,37,0,1658,0,0,0,0,0,0,734,0,31050,0,5936,3,0,38,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[4.200,4,"xxxxxxxxxxxx",8,2.29,0.00,1.45,0.00,0.00,1]}

Merkwürdig ist die ver":"3.008.10
Im WEB Interface ist es
[image]
Es gab aber keine FW Updates oder Versuche

Wenn ich den ioBroker Adapter gestartet habe läuft ein CURL oder der Aufruf das Web Interfaces in der Regel auf einen Timout
Sobald ich den Adapter beende, läuft der WEB Zugriff auf den WIFI Adapter und der CURL "flüssig" 

Hast du noch eine Idee ?

<html><body>

solax.0 | 2160 | 2023-03-18 09:17:28.759 | debug | is Night: false
-- | -- | -- | -- | --
solax.0 | 2160 | 2023-03-18 09:17:28.753 | debug | Sun Azimut: 123.2°
solax.0 | 2160 | 2023-03-18 09:17:28.753 | debug | Sun Altitude: 23.9°
solax.0 | 2160 | 2023-03-18 09:17:28.753 | debug | calculate astrodata ...
solax.0 | 2160 | 2023-03-18 09:17:28.753 | debug | current local Time: 09:17
solax.0 | 2160 | 2023-03-18 09:17:28.752 | debug | nauticalDawn: 05:29
solax.0 | 2160 | 2023-03-18 09:17:28.752 | debug | nauticalDusk: 19:46
solax.0 | 2160 | 2023-03-18 09:17:28.751 | debug | nightCalc started ...
solax.0 | 2160 | 2023-03-18 09:12:28.795 | debug | Local Request Interval started ...
solax.0 | 2160 | 2023-03-18 09:12:28.794 | debug | Request Interval: 5 seconds
solax.0 | 2160 | 2023-03-18 09:12:28.693 | debug | Sun Azimut: 122°
solax.0 | 2160 | 2023-03-18 09:12:28.692 | debug | Sun Altitude: 23.2°
solax.0 | 2160 | 2023-03-18 09:12:28.692 | debug | calculate astrodata ...
solax.0 | 2160 | 2023-03-18 09:12:28.691 | debug | current local Time: 09:12
solax.0 | 2160 | 2023-03-18 09:12:28.690 | debug | nauticalDawn: 05:29
solax.0 | 2160 | 2023-03-18 09:12:28.690 | debug | nauticalDusk: 19:46
solax.0 | 2160 | 2023-03-18 09:12:28.688 | debug | nightCalc started ...
solax.0 | 2160 | 2023-03-18 09:12:28.687 | debug | System longitude: 7.884093 System latitude: 49.1565412
solax.0 | 2160 | 2023-03-18 09:12:28.684 | debug | Solax is started in local-mode
solax.0 | 2160 | 2023-03-18 09:12:26.643 | silly | States  system redis pmessage  system.adapter.solax.0.logLevel/system.adapter.solax.0.logLevel:{"val":"silly","ack":true,"ts":1679127146633,"q":0,"from":"system.adapter.solax.0","lc":1679083151335}
solax.0 | 2160 | 2023-03-18 09:12:26.600 | info | starting.  Version 0.6.0 (non-npm:  simatec/ioBroker.solax#1e2f5bcbe778bbbb9f881ed1f4a094d9fb2e29f8) in  /opt/iobroker/node_modules/iobroker.solax, node: v16.19.1,  js-controller: 4.0.24
solax.0 | 2160 | 2023-03-18 09:12:26.189 | debug | Plugin sentry Initialize Plugin (enabled=true)
solax.0 | 2160 | 2023-03-18 09:12:26.110 | silly | statesDB connected
solax.0 | 2160 | 2023-03-18 09:12:26.108 | debug | States connected to redis: 127.0.0.1:9000
solax.0 | 2160 | 2023-03-18 09:12:26.090 | debug | States create User PubSub Client
solax.0 | 2160 | 2023-03-18 09:12:26.089 | debug | States create System PubSub Client
solax.0 | 2160 | 2023-03-18 09:12:26.078 | debug | Redis States: Use Redis connection: 127.0.0.1:9000
solax.0 | 2160 | 2023-03-18 09:12:26.075 | silly | objectDB connected
solax.0 | 2160 | 2023-03-18 09:12:26.067 | silly | redis psubscribe cfg.o.enum.*
solax.0 | 2160 | 2023-03-18 09:12:26.053 | silly | redis psubscribe cfg.o.system.user.*
solax.0 | 2160 | 2023-03-18 09:12:26.050 | debug | Objects connected to redis: 127.0.0.1:9001
solax.0 | 2160 | 2023-03-18 09:12:26.043 | debug | Objects client initialize lua scripts
solax.0 | 2160 | 2023-03-18 09:12:26.014 | debug | Objects create User PubSub Client
solax.0 | 2160 | 2023-03-18 09:12:26.013 | debug | Objects create System PubSub Client
solax.0 | 2160 | 2023-03-18 09:12:26.012 | debug | Objects client ready ... initialize now

</body>
</html>

simatec, 2023-03-18:
Die unterschiedlichen Firmware Versionen sind schon sehr komisch. 

Ich würde an deiner Stelle mal den kompletten Stick rebooten bzw. mal komplett trennen. 
Des Weiteren schaue mal nach, welche axios Version bei dir installiert ist. Da gab es in letzter Zeit immer mal wieder Bugs. 

Dann stelle mal zum testen den Intervall auf 30 Sekunden

MrDidiB, 2023-03-21:
Hast du eine Fritz!Box als Router? Ich hatte ein ähnliches Problem mit meiner openWB (Wallbox). Ich bekam plötzlich keine Daten mehr von meinem Solax (Qcells) Wechselrichter. Dann habe ich meine FritzBox neu gestartet und die Verbindung hat danach wieder funktioniert. Was genau das Problem war, weis ich aber leider nicht.

Charlie-2222, 2023-03-25:
Danke für die Tips und Hinweise.
Ich habe schon alles zurückgesetzt und auch einen neuen Container from scratch auf der QNAP mit dem buanet/iobroker v8.0.0 image erstellt und - ohne einen restore - einzig und allein den Solax adapter installiert und konfiguriert, leider ohne Erfolg, gleiches Fehlerbild
Da der WIFI Stick in der bash des iobrokers mit CURL Daten liefert, gehe ich erstmal nicht von einem Problem des Sticks aus

thorsten-vogt, 2023-04-25:
Hallo Zusammen,

ich habe vermutlich das gleiche Problem wie @Charlie-2222 .
Ich habe heute mit dem SolaX-Support telefoniert wegen ein "Problem" mit dem Akku. Maßnahme von SolaX -> Firmwareupdate aus der Ferne, seit dem keine lokalen Daten im iobroker mehr, nur noch über die Cloud.

Die lokale Abfrage:
`curl -d "optType=ReadRealTimeData&pwd=XXXXXXXXXX" -X POST http://192.168.0.abc`

liefert folgendes zurück:
`{"sn":"XXXXXXXXXX","ver":"3.008.10","type":14,"Data":[2311,2337,2319,10,12,10,137,192,115,444,0,6594,0,6,0,447,5006,5005,5004,2,0,0,0,0,0,0,0,0,0,0,0,0,0,1,0,0,0,0,0,47830,0,0,4779,0,0,1,37,444,256,4116,6419,5892,100,0,26,0,0,0,0,0,0,0,0,0,0,0,0,0,30681,0,182,116,0,0,7273,0,7067,0,23,115,32611,0,276,0,0,0,60782,2,41687,0,1083,0,16,0,0,0,0,0,0,0,0,0,1,100,1,25,228,256,5256,3600,0,350,182,163,33,33,32,2139,258,9766,9252,9252,8995,0,0,3335,3301,18436,22,21302,14389,18757,12609,16692,12611,14384,21302,14389,18757,12609,16692,12611,14384,21302,14389,18754,12345,16694,12867,14131,21302,14389,18754,12345,16694,12355,13872,21302,14389,18754,12849,16688,12867,12848,1281,1799,514,773,0,47830,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[15.000,14,"XXXXXXXXXXXXXX",8,1.30,0.00,1.28,1.09,0.00,1]}`

Vor meinem Anruf bei SolaX war die Version noch 3.006.04 (hatte ich als Wert im iobroker mit abgefragt und abgespeichert)

In der Firmware Version hat sich somit etwas geändert.

Im Webinterface des Stick selbst wird die gleiche Firmware angezeigt wie bei @Charlie-2222 
https://user-images.githubusercontent.com/119816880/226094386-1c132075-6630-4bcc-88ce-7ca96b2a63a7.png

Mein log im iobroker auf "alles" sieht so aus:

`<html>
<body>

solax.0 | 2023-04-25 19:21:04.511 | debug | Local Request Interval started ...
-- | -- | -- | --
solax.0 | 2023-04-25 19:21:04.510 | debug | Request Interval: 10 seconds
solax.0 | 2023-04-25 19:21:04.377 | debug | Sun Azimut: 277.1°
solax.0 | 2023-04-25 19:21:04.376 | debug | Sun Altitude: 11.1°
solax.0 | 2023-04-25 19:21:04.376 | debug | calculate astrodata ...
solax.0 | 2023-04-25 19:21:04.374 | debug | current local Time: 19:21
solax.0 | 2023-04-25 19:21:04.373 | debug | nauticalDawn: 04:39
solax.0 | 2023-04-25 19:21:04.372 | debug | nauticalDusk: 22:13
solax.0 | 2023-04-25 19:21:04.369 | debug | nightCalc started ...
solax.0 | 2023-04-25 19:21:04.367 | debug | System longitude: 8.24178765 System latitude: 53.2059718
solax.0 | 2023-04-25 19:21:04.340 | debug | Solax is started in local-mode
solax.0 | 2023-04-25 19:21:02.226 | silly | States system redis pmessage system.adapter.solax.0.logLevel/system.adapter.solax.0.logLevel:{"val":"silly","ack":true,"ts":1682443262210,"q":0,"from":"system.adapter.solax.0","lc":1682438943774}
solax.0 | 2023-04-25 19:21:02.167 | info | starting. Version 0.6.0 in /opt/iobroker/node_modules/iobroker.solax, node: v18.15.0, js-controller: 4.0.24
solax.0 | 2023-04-25 19:21:01.642 | debug | Plugin sentry Initialize Plugin (enabled=true)
solax.0 | 2023-04-25 19:21:01.496 | silly | statesDB connected
solax.0 | 2023-04-25 19:21:01.494 | debug | States connected to redis: 127.0.0.1:9000
solax.0 | 2023-04-25 19:21:01.405 | debug | States create User PubSub Client
solax.0 | 2023-04-25 19:21:01.402 | debug | States create System PubSub Client
solax.0 | 2023-04-25 19:21:01.375 | debug | Redis States: Use Redis connection: 127.0.0.1:9000
solax.0 | 2023-04-25 19:21:01.371 | silly | objectDB connected
solax.0 | 2023-04-25 19:21:01.362 | silly | redis psubscribe cfg.o.enum.*
solax.0 | 2023-04-25 19:21:01.316 | silly | redis psubscribe cfg.o.system.user.*
solax.0 | 2023-04-25 19:21:01.310 | debug | Objects connected to redis: 127.0.0.1:9001
solax.0 | 2023-04-25 19:21:01.268 | debug | Objects client initialize lua scripts
solax.0 | 2023-04-25 19:21:01.160 | debug | Objects create User PubSub Client
solax.0 | 2023-04-25 19:21:01.158 | debug | Objects create System PubSub Client
solax.0 | 2023-04-25 19:21:00.408 | debug | Objects client ready ... initialize now
solax.0 | 2023-04-25 19:20:59.162 | debug | Redis Objects: Use Redis connection: 127.0.0.1:9001

</body>
</html>`

Nach dem "Local Request Interval started ..." kommt nix mehr, bzw. es wiederholt sich dieser block:

`<html>
<body>

solax.0 | 2023-04-25 19:26:04.493 | debug | is Night: false
-- | -- | -- | --
solax.0 | 2023-04-25 19:26:04.463 | debug | Sun Azimut: 278.1°
solax.0 | 2023-04-25 19:26:04.463 | debug | Sun Altitude: 10.4°
solax.0 | 2023-04-25 19:26:04.462 | debug | calculate astrodata ...
solax.0 | 2023-04-25 19:26:04.462 | debug | current local Time: 19:26
solax.0 | 2023-04-25 19:26:04.461 | debug | nauticalDawn: 04:39
solax.0 | 2023-04-25 19:26:04.461 | debug | nauticalDusk: 22:13
solax.0 | 2023-04-25 19:26:04.459 | debug | nightCalc started ...

</body>
</html>`

Ein Neustart des Wechselrichters blieb leider erfolglos.

simatec, 2023-04-26:
Bitte die v0.7.0 testen

thorsten-vogt, 2023-04-27:
Ich habe jetzt die Version 0.7.2 installiert.

Ich habe drei Instanzen des Adapters:
solax.0 -> Lokale Anbindung an einen G4.2 Hybrid
solax.1 -> Cloud Anbindung an einen G4.2 Hybrid
solax.2 -> Cloud Anbindung an einen G3 Hybrid

Im Log erhalte ich diese beiden Info-Hinweise:
<html>
<body>

solax.2 | 2023-04-27 18:02:17.110 | info | State value to set for "solax.2.data.json" has to be type "string" but received type "number"
-- | -- | -- | --
solax.1 | 2023-04-27 18:02:17.014 | info | State value to set for "solax.1.data.json" has to be type "string" but received type "number"

</body>
</html>

Die Hinweise wiederholen sich im 5 Min Abfrageintervall des Adapters.

[image]
[image]

Die rot markierten Zeilen sind in der lokalen Anbindung hinzugekommen, liefern jedoch keine Daten.

Die Version 0.6.0 und 0.7.0 verlierten in der lokalen Anbindung sporadisch die Verbindung zum Wechselrichter.  Gehäuft immer dann, wenn ich über die Handyapp von SolaX Änderungen am Wechselrichter vorgenommen habe, aber nicht immer.

Ich habe dafür ein Blockly Skript zum Restart.

Charlie-2222, 2023-04-27:
Die Abfrage mit CURL funktionierte ja schon immer,
aber mit Version 7.2 bekomme ich nun plötzlich auch wieder Daten im iobroker zurück
Was hier auch immer das Problem war, es tut glücklierweise wieder

simatec, 2023-04-27:
Hast du die Firmware Version im Config Menü auf Version 3.xxx.xx eingestellt?

thorsten-vogt, 2023-04-27:
> Hast du die Firmware Version im Config Menü auf Version 3.xxx.xx eingestellt?

Ja, habe ich bereits in der 0.7.0 gemacht und die Einstellung wurde in 0.7.2 übernommen! Nochmals vielen Dank dafür!

simatec, 2023-04-28:
@thorsten-vogt Was willst du mir jetzt genau damit sagen? Kommen die Daten lokal nun an oder nicht?

thorsten-vogt, 2023-04-29:
> @thorsten-vogt Was willst du mir jetzt genau damit sagen? Kommen die Daten lokal nun an oder nicht?

Ja, ich erhalte die lokalen Daten nach Auswahl von Version 3.xxx.xx im Config Menü!

Charlie-2222, 2023-04-29:
Auch mit 2 Sekunden Abfrage Intervall scheint die Version 7.20 stabil. Danke
Was mich betrifft könnte das Ticket geschlossen werden.

closed 2023-05-04 by simatec