simatec/ioBroker.solax #161: Adapter blockiert Pocket WiFi 3

Sansibari, 2023-04-11:
**Describe the bug**  
Ausgangslage:
   Adapter: v0.6.0
   Pocket Wifi 3 Firmware Version: 3.001.02
   Betriebsart: Local IP Wifi Pocket Stick
   
Der Adapter arbeitet einwandfrei und liefert die Daten  während ca. 5 bis 10 Minuten.
Danach blockiert der Adapter die Kommunikation mit dem Pocket Wifi d.h. das Pocket Wifi kann mit der lokalen IP-Adresse via Browser nicht mehr angesprochen werden - es gibt keine Rückmeldung.

Sobald der Adapter gestoppt wird, kann das Pocket Wifi wieder problem via Browser wieder korrekt angesprochen werden.

Die Abfrage mittels curl funktioniert ohne Probleme.

Sansibari, 2023-04-13:
Danke für die schnelle Antwort!
Habe Standardeinstellung 10s verwendet.
Versuche jetzt mit 60s und melde mich später zurück.

Sansibari, 2023-04-13:
Auch mit 60s Invervall "blockiert" der Adapter den Pocket Wifi nach ca. 80 min.
Nach dem Stoppen des Adapters kann wieder mit dem Pocket WiFi kommuniziert werden.

simatec, 2023-04-13:
Stelle den Adapter auf debuglog und beobachte mal im Log, was passiert. 
bitte hier dann den Log posten

Sansibari, 2023-04-14:
N.B.: Vom host. werden auch falsche Timestamps ins log geschrieben (Zeitverschiebung von +7h nicht berücksichtigt, obwohl die Zeit in ioBroker richtig angezeigt wird).

Nachdem der Adapter keine Daten mehr vom Pocket WiFi gesendet hat, habe ich diesen manuell gestoppt).

Sansibari, 2023-04-14:
<html>
<body>

2023-04-14 01:39:04.888 - info: host.ioBroker "system.adapter.solax.0" enabled2023-04-14 01:39:05.086 - info: host.ioBroker instance system.adapter.solax.0 started with pid 334852023-04-14 08:39:05.755 - debug: solax.0 (33485) Redis Objects: Use Redis connection: 127.0.0.1:90012023-04-14 08:39:05.782 - debug: solax.0 (33485) Objects client ready ... initialize now2023-04-14 08:39:05.824 - debug: solax.0 (33485) Objects create System PubSub Client2023-04-14 08:39:05.825 - debug: solax.0 (33485) Objects create User PubSub Client2023-04-14 08:39:05.848 - debug: solax.0 (33485) Objects client initialize lua scripts2023-04-14 08:39:05.851 - debug: solax.0 (33485) Objects connected to redis: 127.0.0.1:90012023-04-14 08:39:05.865 - debug: solax.0 (33485) Redis States: Use Redis connection: 127.0.0.1:90002023-04-14 08:39:05.916 - debug: solax.0 (33485) States create System PubSub Client2023-04-14 08:39:05.917 - debug: solax.0 (33485) States create User PubSub Client2023-04-14 08:39:05.977 - debug: solax.0 (33485) States connected to redis: 127.0.0.1:90002023-04-14 08:39:06.012 - debug: solax.0 (33485) Plugin sentry Initialize Plugin (enabled=true)2023-04-14 08:39:06.177 - info: solax.0 (33485) starting. Version 0.6.0 in /opt/iobroker/node_modules/iobroker.solax, node: v16.19.1, js-controller: 4.0.232023-04-14 08:39:08.234 - debug: solax.0 (33485) Solax is started in local-mode2023-04-14 08:39:08.237 - debug: solax.0 (33485) System longitude: 99.95468467783758 System latitude: 12.573972974979982023-04-14 08:39:08.237 - debug: solax.0 (33485) nightCalc started ...2023-04-14 08:39:08.238 - debug: solax.0 (33485) nauticalDusk: 19:202023-04-14 08:39:08.238 - debug: solax.0 (33485) nauticalDawn: 05:232023-04-14 08:39:08.238 - debug: solax.0 (33485) current local Time: 08:392023-04-14 08:39:08.239 - debug: solax.0 (33485) calculate astrodata ...2023-04-14 08:39:08.239 - debug: solax.0 (33485) Sun Altitude: 35.5°2023-04-14 08:39:08.239 - debug: solax.0 (33485) Sun Azimut: 87.7°2023-04-14 08:39:08.305 - debug: solax.0 (33485) Request Interval: 30 seconds2023-04-14 08:39:08.305 - debug: solax.0 (33485) Local Request Interval started ...2023-04-14 08:39:40.560 - debug: solax.0 (33485) local request: {"sn":"SRxxxxxxxxxx","ver":"3.006.04","type":14,"Data":[0,0,0,0,0,0,0,0,0,0,3730,3654,48,49,1815,1798,0,0,0,7,1111,116,432,2300,2300,2300,48,5,18,1105,25,390,5000,0,0,0,0,0,0,36480,550,2006,3620,55,1991,1,55,0,256,10003,3592,5892,100,0,46,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,12814,0,15188,0,27,19,33017,0,35,25429,0,37,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1,71,1,37,121,256,3942,2700,350,350,337,314,34,33,225,1618,520,8995,9252,9252,0,0,0,3364,3335,32371,123,21302,14389,18497,12867,16688,12355,13621,21302,14389,18497,12867,16688,12355,13621,21302,14389,18754,13109,16688,12867,12337,21302,14389,18754,13109,16688,12611,13106,0,0,0,0,0,0,0,515,1538,770,1025,0,36480,0,0,0,8,0,0,4,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[5,14,"Hxxxxxxxxxxxxx",8,1.21,0,1.2,1.09,0,1]}2023-04-14 08:40:39.555 - debug: solax.0 (33485) local request: {"sn":"SRxxxxxxxxxx","ver":"3.006.04","type":14,"Data":[0,0,0,0,0,0,0,0,0,0,3708,3792,49,47,1830,1805,0,0,0,7,1112,116,431,2299,2300,2299,48,5,18,1106,26,390,5000,0,0,0,0,0,0,36490,550,2007,3621,55,1991,1,55,0,256,10258,3592,5892,100,0,46,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,12814,0,15188,0,27,19,33018,0,36,25429,0,37,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1,71,1,37,121,256,3942,2700,350,350,337,314,34,33,225,1618,520,8995,9252,9252,0,0,0,3364,3336,32404,123,21302,14389,18497,12867,16688,12355,13621,21302,14389,18497,12867,16688,12355,13621,21302,14389,18754,13109,16688,12867,12337,21302,14389,18754,13109,16688,12611,13106,0,0,0,0,0,0,0,515,2818,770,1025,0,36490,0,0,0,8,0,0,4,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[5,14,"Hxxxxxxxxxxxxx",8,1.21,0,1.2,1.09,0,1]}2023-04-14 08:41:09.537 - debug: solax.0 (33485) local request: {"sn":"SRxxxxxxxxxx","ver":"3.006.04","type":14,"Data":[0,0,0,0,0,0,0,0,0,0,3722,3655,49,50,1832,1829,0,0,0,7,1115,116,432,2299,2300,2300,48,5,18,1109,25,390,5000,0,0,0,0,0,0,36500,550,2044,3622,56,2028,1,55,0,256,10288,3592,5892,100,0,46,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,12814,0,15189,0,27,20,33018,0,36,25429,0,37,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1,71,1,37,121,256,3942,2700,350,350,337,314,34,33,225,1618,520,8995,9252,9252,0,0,0,3364,3336,32420,123,21302,14389,18497,12867,16688,12355,13621,21302,14389,18497,12867,16688,12355,13621,21302,14389,18754,13109,16688,12867,12337,21302,14389,18754,13109,16688,12611,13106,0,0,0,0,0,0,0,515,1538,770,1025,0,36500,0,0,0,8,0,0,4,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[5,14,"Hxxxxxxxxxxxxx",8,1.21,0,1.2,1.09,0,1]}2023-04-14 08:41:39.545 - debug: solax.0 (33485) local request: {"sn":"SRxxxxxxxxxx","ver":"3.006.04","type":14,"Data":[0,0,0,0,0,0,0,0,0,0,3715,3701,49,49,1840,1827,0,0,0,7,1113, [... cut ...]
2023-04-14 09:11:59.719 - info: solax.0 (33485) Got terminate signal TERMINATE_YOURSELF
2023-04-14 09:11:59.720 - info: solax.0 (33485) terminating
2023-04-14 09:11:59.720 - debug: solax.0 (33485) Plugin sentry destroyed
2023-04-14 09:11:59.720 - info: solax.0 (33485) Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason
2023-04-14 02:11:59.767 - info: host.ioBroker stopInstance system.adapter.solax.0 send kill signal
2023-04-14 02:12:00.319 - info: host.ioBroker instance system.adapter.solax.0 terminated with code 11 (ADAPTER_REQUESTED_TERMINATION)

</body>
</html>

Sansibari, 2023-04-14:
Es gibt noch einen weiteren Fehler: Vom Pocket Wifi wird eine Inverter-Grösse von 5kW zurück gemeldet obwohl die Anlage 15kW ist (ist auch so in der Cloud App eingetragen).

Sansibari, 2023-04-14:
[image]

Sansibari, 2023-04-14:
[image]

Sansibari, 2023-04-14:
[image]

Sansibari, 2023-04-14:
[image]

Sansibari, 2023-04-14:
Ich habe via Node-Red einen periodischen curl Abruf gestartet (30s Interval). Das Pocket WiFi sendet die Daten seit mehreren Stunden fehlerfrei.

simatec, 2023-04-14:
Ich baue am WE mal ein paar Debugs ein.
Kannst du mir mal bitte folgende Ausgabe posten?

`
cd /opt/iobroker/node_modules/iobroker.solax
npm list
`

Sansibari, 2023-04-14:
iobroker.solax@0.6.0 /opt/iobroker/node_modules/iobroker.solax
+-- UNMET DEPENDENCY @alcalzone/release-script-plugin-iobroker@^3.5.8
+-- UNMET DEPENDENCY @alcalzone/release-script-plugin-license@^3.5.9
+-- UNMET DEPENDENCY @alcalzone/release-script@^3.5.9
+-- UNMET DEPENDENCY @iobroker/adapter-core@^2.6.7
+-- UNMET DEPENDENCY @iobroker/testing@^4.1.0
+-- UNMET DEPENDENCY @types/chai-as-promised@^7.1.5
+-- UNMET DEPENDENCY @types/chai@^4.3.4
+-- UNMET DEPENDENCY @types/gulp@^4.0.10
+-- UNMET DEPENDENCY @types/mocha@^10.0.1
+-- UNMET DEPENDENCY @types/node@^18.14.2
+-- UNMET DEPENDENCY @types/proxyquire@^1.3.28
+-- UNMET DEPENDENCY @types/sinon-chai@^3.2.9
+-- UNMET DEPENDENCY @types/sinon@^10.0.13
+-- axios@1.3.5
+-- UNMET DEPENDENCY chai-as-promised@^7.1.1
+-- UNMET DEPENDENCY chai@^4.3.7
+-- UNMET DEPENDENCY eslint@^8.35.0
+-- UNMET DEPENDENCY gulp@^4.0.2
+-- UNMET DEPENDENCY mocha@^10.2.0
+-- UNMET DEPENDENCY node-schedule@^2.1.1
+-- UNMET DEPENDENCY proxyquire@^2.1.3
+-- UNMET DEPENDENCY sinon-chai@^3.7.0
+-- UNMET DEPENDENCY sinon@^15.0.1
`-- UNMET DEPENDENCY suncalc2@^1.8.1

npm ERR! code ELSPROBLEMS
npm ERR! missing: @alcalzone/release-script-plugin-iobroker@^3.5.8, required by iobroker.solax@0.6.0
npm ERR! missing: @alcalzone/release-script-plugin-license@^3.5.9, required by iobroker.solax@0.6.0
npm ERR! missing: @alcalzone/release-script@^3.5.9, required by iobroker.solax@0.6.0
npm ERR! missing: @iobroker/adapter-core@^2.6.7, required by iobroker.solax@0.6.0
npm ERR! missing: @iobroker/testing@^4.1.0, required by iobroker.solax@0.6.0
npm ERR! missing: @types/chai-as-promised@^7.1.5, required by iobroker.solax@0.6.0
npm ERR! missing: @types/chai@^4.3.4, required by iobroker.solax@0.6.0
npm ERR! missing: @types/gulp@^4.0.10, required by iobroker.solax@0.6.0
npm ERR! missing: @types/mocha@^10.0.1, required by iobroker.solax@0.6.0
npm ERR! missing: @types/node@^18.14.2, required by iobroker.solax@0.6.0
npm ERR! missing: @types/proxyquire@^1.3.28, required by iobroker.solax@0.6.0
npm ERR! missing: @types/sinon-chai@^3.2.9, required by iobroker.solax@0.6.0
npm ERR! missing: @types/sinon@^10.0.13, required by iobroker.solax@0.6.0
npm ERR! missing: chai-as-promised@^7.1.1, required by iobroker.solax@0.6.0
npm ERR! missing: chai@^4.3.7, required by iobroker.solax@0.6.0
npm ERR! missing: eslint@^8.35.0, required by iobroker.solax@0.6.0
npm ERR! missing: gulp@^4.0.2, required by iobroker.solax@0.6.0
npm ERR! missing: mocha@^10.2.0, required by iobroker.solax@0.6.0
npm ERR! missing: node-schedule@^2.1.1, required by iobroker.solax@0.6.0
npm ERR! missing: proxyquire@^2.1.3, required by iobroker.solax@0.6.0
npm ERR! missing: sinon-chai@^3.7.0, required by iobroker.solax@0.6.0
npm ERR! missing: sinon@^15.0.1, required by iobroker.solax@0.6.0
npm ERR! missing: suncalc2@^1.8.1, required by iobroker.solax@0.6.0

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/iobroker/.npm/_logs/2023-04-14T11_46_14_146Z-debug-0.log

Sansibari, 2023-04-25:
Kann ich noch etwas für das Debugging beitragen?

dskvml, 2023-04-26:
Ich hab das gleiche Problem, nur dass mein Pocket Wifi nicht nach 5-10 Minuten abschmiert, sondern erst nach ~24h. Ich habe 10s Abfrageintervalle eingestellt. Ein Neustart des iobroker Adapters hilft sofort. 
Wenn ich irgendwas dazu beitragen kann, dass das nicht mehr passiert, dann gerne.

simatec, 2023-04-26:
Stelle mal den Intervall etwas höher

simatec, 2023-04-26:
Bitte die v0.7.0 testen

dskvml, 2023-04-26:
mit 10s intervall oder mit höheren intervallen?

simatec, 2023-04-26:
Du kannst 10 Sekunden weiter belassen

simatec, 2023-04-26:
Wichtig ist, in der Config von Solax muss die richtige Firmware-Version des Sticks gewählt sein.

Sansibari, 2023-04-28:
Danke - ich werde den Test asap starten

Sansibari, 2023-04-28:
Test gestartet 10:00
Interval 10 sec.
Seit  2023-04-28 19:27 kommen keine Daten mehr

ist schon bedeutend besser! 
Nun stimmt auch die Angabe zur Leistung des Inverters.
Der Pocket Wifi bleibt auch nicht mehr blockiert - sehr gut.
Ich glaube es gibt auch ein Problem bei der Berechnung der Astrodaten (Nacht).

Einmal besten Dank für den Effort. Hoffentlich kann der Fehler auch lokalisiert werden so dass der Adapter stabil wird.
Kann ich mit weiteren Infos helfen?

Sansibari, 2023-04-28:
Korrektur: Die Astro Daten kommen weiterhin regelmässig - es werden nur keine Inverterdaten mehr geliefert.

simatec, 2023-04-28:
Verstehe deine Antwort nicht. Welche Version? Was für Daten kommen nicht? 
Debuglog?

Sansibari, 2023-04-29:
Ineressant: Nachdem "is Night: true" werden keine Daten vom Inverter mehr aktualisiert, dh. wärend der "Nacht"  kommen keine  "local request: {"sn":"xxxx","ver":"3.006.04","type":14,"Data":[0,0,0,0,0,0,0,0,0,0,3640,3701,42,41, ..... " mehr.

Die Astro-Daten kommen jedoch wie vorher gemässs Intervall: 
[image]

Nachdem "is Night: false" wurde, werden die Inverter Daten wieder aktualisiert.

Das ist soweit mal gut, dass der Adapter bisher (1 Tag) stabil läuft!
Die Inverter Daten sollten jedoch auch während der Nacht aktualisiert werden (nicht nur während dem Tag).

Die Berechnung von
     07:57:30.224 | debug | is Night: false
stimmt nicht für meine Location (gesetzt in den ioBroker Einstellungen)
Richtig wäre:
     Sunrise: 29.04.2023 06:00:58
     Sunset: 29.04.2023 18:34:02
     
Hilft das weiter oder benötigst Du Debuglog?

Sansibari, 2023-04-29:
Getestet mit Version v0.7.2

simatec, 2023-04-29:
Du kannst doch im Config Menü selber bestimmen, ob Nachts Daten kommen sollen oder nicht

Sansibari, 2023-04-29:
Vielen Dank für den Hinweis, diese Einstellung habe ich vollkommen übersehen!
Entschuldige bitte meine "Fehlermeldung".
Der Adapter läuft bisher stabil.

Eine Frage noch:
Der Adapter berechnet die "Nautical Dusk/Dawn
Meiner Ansicht nach wäre es sinnvoller, die Civil Dusk/Dawn zu berechnen (6° statt 12°)
Das entspricht dem Zeitpunkt wenn die Solorzellen keinen Strom mehr produzieren bzw. wieder anfangen zu liefern.

Ich finde kein Objekt z.B. "isNight". Es wäre sicher nützlich, dieses ebenfalls unter den Astro-Daten zu finden.
Vielen Dank!

Sansibari, 2023-05-03:
Bist Du interessiert den Adapter zu verbessern?
Da ich den Inverter im Inselbetrieb (bzw. offgrid) betreibe, werden für meinen Zweck drei wichtige Werte - welche im Datenstrom enthalten sind - nicht in die Objekte übernommen:

Solax Bezeichnung     Data Stream      Beschreibung
peps1                          Data[29]            "EPS 1 Power": (Units: W, signed),
peps2                          Data[30]            "EPS 2 Power": (Units: W, signed),
peps3                          Data[31]            "EPS 3 Power": (Units: W, signed),

Es wäre schön, diese Werte mit der nächsten Adapter-Version ebenfalls zu erhalten.
Wenn Du interessiert bist, kann ich Dir gern noch weiter Decodierungen nennen.
Besten Dank!

simatec, 2023-05-03:
Bitte teste v0.7.3

Sansibari, 2023-05-16:
Bitte diesen issue wieder öffnen - es gibt noch Fehler (siehe oben)