simatec/ioBroker.solax #162: Objekt "acpower" wird nicht aktualisiert

sebkau, 2023-04-16:
Hallo,
ich habe leider das Problem, dass nach einer unbestimmten Zeit das Objekt "acpower" nicht mehr aktualisiert wird. Alle anderen Objekte werden weiterhin problemlos aktualisiert.
Ich hatte das Problem jetzt bereits zum 2. Mal. Der Datenpunkt wird ca. 1 Woche lange aktualisiert, dann nicht mehr. Wenn ich den Adapter stoppe, den Datenpunkt lösche und den Adapter wieder starte, wird er sofort wieder neu angelegt, korrekt abgerufen und auch wieder aktualisiert.

simatec, 2023-04-16:
Bitte einen Debuglog posten

sebkau, 2023-04-17:
[solax_log.txt](https://github.com/simatec/ioBroker.solax/files/11251495/solax_log.txt)

Anbei der Log. So wie ich das sehe werden die korrekten Werte über die API noch geholt für acpower, der Datenpunkt wird aber nicht mehr aktualisiert.

simatec, 2023-04-17:
Schaue ich mir mal an ...

sebkau, 2023-04-17:
Danke! Wenn ich noch irgendwie unterstützen kann, einfach Bescheid geben!

sebkau, 2023-04-23:
Hallo nochmal,

heute ist dieser Bug wieder aufgetreten und eine Sache ist mir dabei aufgefallen. Immer wenn der Bug auftritt, hat das Objekt "acpower" einen negativen Wert. Es ist noch nie vorgekommen, dass dort ein positiver Wert stand und das Objekt nicht mehr aktualisiert wurde. Screenshot habe ich gerade gemacht, Objekt wurde sein 09:52 Uhr nicht mehr aktualisiert. Alle anderen nach wie vor problemlos.

[image]

simatec, 2023-04-26:
Hmmm wo kommt der negative Wert her? Hast du eine Batterie mit dran?

sebkau, 2023-04-26:
Eine Batterie hängt auch dran, ja, das sollte aber eigentlich keinen Einfluss auf diesen Wert haben. Ich vermute aber den Auslöser gefunden zu haben!
Mit an meinem Netz hängt auch noch ein Balkonkraftwerk. Das hatte ich schon vor der PV Anlage und hängt jetzt weiterhin mit am Netz. Ich könnte mir vorstellen, dass das den negativen Wert verursachen könnte.

simatec, 2023-04-26:
Bitte die v0.7.0 testen

sebkau, 2023-04-26:
Ich danke dir vielmals! Werde testen und berichten.

sebkau, 2023-04-30:
Hallo nochmal,

ich habe leider weiterhin Probleme, dass die Objekte nicht upgedatet werden... über den API request werden sie jedoch korrekt abgeholt... ich bin jetzt auf 0.7.2 und es hat sich jetzt auf weitere Werte ausgeweitet.

[image]

Hinweis, falls das eine Rolle spielt: Ich habe den Solax Adapter 2x installiert und hole die die Daten mit Instanz 0 lokal ab und mit 1 über die API. Grund ist, dass die Daten teils  voneinander abweichen... über lokal füttere ich den Energiefluss-Adapter, über die API die Statistik. Probleme habe ich nur mit der API, lokal funktioniert soweit sehr gut.

Aktueller Log:

solax.1 | 2023-04-30 19:38:49.485 | debug | solaxRequest: {"success":true,"exception":"Query success!","result":{"inverterSN":"XXXXXXXXXXXXXX","sn":"XXXXXXXXXX","acpower":492,"yieldtoday":49.2,"yieldtotal":1349.9,"feedinenergy":806.41,"consumeenergy":85.19,"soc":95,"inverterType":"14","inverterStatus":"102","uploadTime":"2023-04-30 19:34:29","batPower":96,"powerdc1":318,"powerdc2":317,"batStatus":"0"}}
-- | -- | -- | --
solax.1 | 2023-04-30 19:38:48.423 | debug | API Request successfully completed
solax.1 | 2023-04-30 19:38:48.421 | debug | request-result: {"success":true,"exception":"Query success!","result":{"inverterSN":"XXXXXXXXXXXXXX","sn":"XXXXXXXXXX","acpower":492,"yieldtoday":49.2,"yieldtotal":1349.9,"feedinpower":0,"feedinenergy":806.41,"consumeenergy":85.19,"feedinpowerM2":0,"soc":95,"peps1":0,"peps2":0,"peps3":0,"inverterType":"14","inverterStatus":"102","uploadTime":"2023-04-30 19:34:29","batPower":96,"powerdc1":318,"powerdc2":317,"powerdc3":null,"powerdc4":null,"batStatus":"0"}}
solax.1 | 2023-04-30 19:38:48.176 | debug | Request Interval: 5 minute(s)
solax.1 | 2023-04-30 19:38:47.892 | debug | Sun Azimut: 282.4°
solax.1 | 2023-04-30 19:38:47.891 | debug | Sun Altitude: 8.8°
solax.1 | 2023-04-30 19:38:47.891 | debug | calculate astrodata ...
solax.1 | 2023-04-30 19:38:47.889 | debug | current local Time: 19:38
solax.1 | 2023-04-30 19:38:47.888 | debug | nauticalDawn: 04:45
solax.1 | 2023-04-30 19:38:47.887 | debug | nauticalDusk: 22:03
solax.1 | 2023-04-30 19:38:47.885 | debug | nightCalc started ...
solax.1 | 2023-04-30 19:38:47.882 | debug | System longitude: 8.467231753787908 System latitude: 49.63669454967875
solax.1 | 2023-04-30 19:38:47.872 | debug | Solax is started in cloud-mode
solax.1 | 2023-04-30 19:38:45.593 | info | starting. Version 0.7.2 in /opt/iobroker/node_modules/iobroker.solax, node: v18.16.0, js-controller: 4.0.24
solax.1 | 2023-04-30 19:38:43.767 | debug | Plugin sentry Initialize Plugin (enabled=true)
solax.1 | 2023-04-30 19:37:28.959 | info | Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason
solax.1 | 2023-04-30 19:37:28.957 | info | terminating
solax.1 | 2023-04-30 19:37:28.952 | info | Got terminate signal TERMINATE_YOURSELF
solax.1 | 2023-04-30 19:36:46.257 | info | State value to set for "solax.1.data.json" has to be type "string" but received type "number"
solax.1 | 2023-04-30 19:36:41.640 | info | starting. Version 0.7.2 in /opt/iobroker/node_modules/iobroker.solax, node: v18.16.0, js-controller: 4.0.24

simatec, 2023-04-30:
Hast du die richtige Version eingestellt?

sebkau, 2023-04-30:
Ja, in der solax.0 wo ich die Daten lokal hole ist die richtige Firmwareversion des Sticks eingestellt, ja. In der solax.1 hole ich die Daten ja über die API, da fällt die Option ja weg.

simatec, 2023-04-30:
Welchen Wechselrichter hast du?

sebkau, 2023-04-30:
Das ist ein Q Cells Q.Volt HYB-G3-3P Soweit ich weiß sind das aber einfach umgebrandete Solax Geräte. Also ich benutze auch die Solax Cloud App damit.

Aber für mich zum Verständnis, hat das eine Auswirkung? Die Werte die ich im Debug log über den API request kommen sehe sind absolut korrekt. Oder weichen da irgendwo Bezeichnungen/Formatierungen ab?

simatec, 2023-05-01:
Das kann durchaus sein, dass es ein unter der Haube ein Solax ist. Welchem Typ entspricht dein WR bei Solax?
Hier muss man lokal unterscheiden. 
poste mal die Ausgabe mit curl

sebkau, 2023-05-01:
Das erklärt dann evtl. auch, wieso in der lokalen Instanz nicht alle Werte den Bezeichnungen entsprechen... tut mir echt Leid, das hatte ich komplett vergessen zu erwähnen... der Elektriker hat bei der Installation auch immer nur von Solax gesprochen, ich hab wirklich nicht soweit gedacht, dass das abweichen könnte...

Curl sagt (Seriennummern habe ich wieder durch XXX ersetzt):

`seb@AirvonSebastian ~ % curl -d "optType=ReadRealTimeData&pwd=XXXXXXXXXX" -X POST http://192.168.178.XX
{"sn":"XXXXXXXXXX","ver":"3.003.02","type":14,"Data":[2361,2377,2381,72,70,53,1701,1670,1243,4614,3444,3733,68,65,2350,2457,5000,5000,5001,2,0,0,0,0,0,0,0,0,0,0,0,0,0,1,4296,0,0,0,0,32140,0,0,3197,0,0,1,55,318,256,10762,270,5893,100,0,51,0,0,0,0,0,0,0,0,0,0,0,0,0,13815,0,287,5,0,0,2803,0,3239,0,26,61,14748,0,332,0,0,0,17220,1,8523,0,2115,0,2,0,0,0,0,0,0,0,0,0,1,99,1,30,91,256,3504,2400,0,300,239,209,33,33,32,1620,777,14135,14135,14135,0,0,0,3336,3321,38897,9,20564,12339,18753,12852,18740,12612,14645,20564,12339,18754,12851,18736,12868,13110,20564,12339,18754,12851,18736,13124,13873,20564,12339,18754,12851,18736,12868,13109,0,0,0,0,0,0,0,513,4097,1281,771,0,32140,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[10.000,14,"XXXXXXXXXX",8,1.20,0.00,1.20,1.09,0.00,1]}%`

Welches Solax Modell es genau ist sehe ich leider nirgends, auf dem Wechselrichter sind aber auch überall Solax Aufkleber. Aufgrund der Bezeichnung würde ich tippen, dass es der SolaX X3-10.0-G3 hybrid ist.

simatec, 2023-05-01:
Muss ich mir in Ruhe zwecks Übersetzung der Daten anschauen

sebkau, 2023-05-01:
Ok, danke dir vielmals! Wenn ich irgendwie unterstützen kann gib gerne Bescheid!

Was ich allerdings nicht verstehe... ich habe jetzt testweise mal beide Instanzen gelöscht und zurückgerollt auf die stable 0.6.0 die ja vorher funktionierte. Auch da bekomme ich jetzt nur noch die Werte wie oben im Screenshot...

simatec, 2023-05-02:
So ich habe mir das nochmal angeschaut. Du hast hier etwas für Verwirrung gesorgt, da du lokal und Cloud in einem Issue mischst.
Type 14 ist lokal integriert und sollte funktionieren.
Cloud ist bei allen WR gleich und sollte auch funktionieren.
Lösche mal die Instanz 1 und erstelle sie neu, oder stelle die Instanz auch mal auf lokal und dann nach dem speichern wieder auf Cloud.

Ich halte es aber für nicht sinnvoll 2 Instanzen zu haben. Die Werte zwischen Cloud und Lokal weichen nur ab, da die Clouddaten nicht genau sind und nur alle 5 Minuten aktualisiert werden, während die lokalen Daten Echtzeitdaten sind.

Genauer ist hier definitiv die lokale Variante.

sebkau, 2023-05-02:
Hallo nochmal und danke für deine Unterstützung! Ja ich habe gemerkt ist etwas chaotisch geworden, entschuldige....

Also, folgendes habe ich noch herausgefunden.

1. Beim Wechselrichter handelt es sich um einen X3-Hybrid-G4, nicht wie vorher vermutet um den G3.

**API**
2. Ich habe jetzt noch einmal alle Instanzen gelöscht, alle Adapter gelöscht, alle Objekte gelöscht. Adapter in Version 0.7.2 installiert und ich bekomme trotzdem nur Werte zu folgenden Objekten

[image]

Über den API request kommen ja aber alle Werte korrekt rein wie wir im debug log gesehen haben, es scheint also an der Stelle zwischen dem abholen der Daten aus dem API Request und dem schreiben in die Objekte zu klemmen...

**Lokal**
Die Werte weichen nicht nur um ein paar kwh ab, sondern erheblich. Ich kann mir auch nicht erklären wo das herkommt...
z.B. bekomme ich bei "totalFeed" einen Wert von 188 kwh zurück

[image]

Tatsächlich eingespeist habe ich allerdings schon 842 kwh, wie man Online und in der App sehen kann.
[image]

Bevor ich hier aber noch mehr Durcheinander mache... Soll ich vielleicht lieber diesen Issue schließen und diese Themen in einen neuen übernehmen?

simatec, 2023-05-03:
Bitte teste v0.7.3

sebkau, 2023-05-03:
Das schaut hervorragend aus, ich habe über die API nun wieder alle Werte :) Vielen Dank!

[image]

closed 2023-05-03 by simatec