simatec/ioBroker.solax #166: Issue + Solution for ioBroker (showing the wrong value fot batpower)

jamakr4, 2023-05-07:
The probleme that I was facing is that the solax adapter reads out the wrong value for "batpower". When the value is actually =0.0 it does not write it in the batpower datapoint. Instead it just remains on the previous value which for my specific need is an issue because as I am doing a full visualisation, it shows the wrong value and with that beeing the case it also triggers the wrong animation in my vis. 

I fixed it by writting a Blokly script in the ioBroker that triggers every 10 seconds and if the soc (state of charge) of the Battery is =100 it just changes the value of the datapoint to 0 (You also need to allow "Writable" for batpower to work).

Not really sure why this happens maybe it is because I am using it with a Qcells system instead of a Solax one but the rest just works perfectly. BTW Solax and Qcells kind of belong together so the adapter is cross compatible to qcells in case you didnt know.

[image]

simatec, 2023-05-07:
Cloud or local?

jamakr4, 2023-05-07:
Cloud I think

simatec, 2023-05-08:
You think or are you sure?
I needed this information a little more precisely...

simatec, 2023-05-28:
Fix in v0.7.5

closed 2023-05-28 by simatec