simatec/ioBroker.solax #167: AC Power for 3phases is not correctly accumulated

MagicSven81, 2023-05-08:
The power of all three phases is not correctly accumulated.
Solax X3 mic 6k G2 with USB Dongle v3 (expert mode)

[image]

simatec, 2023-05-08:
Please post the Debuglog with json data

MagicSven81, 2023-05-08:
<html>
<body>

solax.0 | 2023-05-08 14:24:27.768 | debug | local request: {"sn":"SRMxxxxx","ver":"3.006.04","type":16,"Data":[2374,2372,2349,9,10,11,261,244,222,4829,0,0,16,0,0,779,0,0,4998,4998,4999,2,50,0,42,6000,35,32,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,729,0,1,4,1,0,19487,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[6,16,"MC206TIB180111",8,1.18,0,1.15,1.01,0,1]}
-- | -- | -- | --

</body>
</html>

MagicSven81, 2023-05-08:
**There are also some values i didn´t trust.**

[image]

simatec, 2023-05-08:
The data from type 16 could not yet be determined 100% due to missing hardware.

But you can support here with pleasure and post your read out data here with the assignment of all values.
Then these can flow with into the adapter

`"Data":[2374,2372,2349,9,10,11,261,244,222,4829,0,0,16,0,0,779,0,0,4998,4998,4999,2,50,0,42,6000,35,32,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,729,0,1,4,1,0,19487,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[6,16,"MC206TIB180111",8,1.18,0,1.15,1.01,0,1]`

From the values you had to assign the data and post best in the following format here.

https://github.com/simatec/ioBroker.solax/blob/dad49a5f586c4463a710eea9cf732d76b597c266/main.js#L518

MagicSven81, 2023-05-08:
Sorry aber jetzt hab ich den Faden verloren.
Wie kann Type16 ein "missing hardware" sein?
Der USB Stick v3 ist am Gerät eingesteckt, Adapter im iobroker installiert.
Zusätzliche Hardware wird da ja nicht mehr benötigt.

simatec, 2023-05-08:
Du musst die Werte mit denen aus der Cloud vergleichen und dann zuordnen.

Es geht los mit 0 und dann halt demensprechend weiter

0: 2347, // AC-Power 1 diff10
1: 2372, // AC-Power 2 diff10
2: 2349 // AC-Power 3 diff10 

Die Auswertung und Zuordnung kannst nur du machen, da mir die Livedaten vom WR fehlen.
Andere User haben vergleiche mit der Cloud gemacht und die Daten somit ermittelt. Beachte aber, dass die Werte teilweise umgerechnet werden müssen. diff10 wäre hier 2349/10 = 234,9 V

MagicSven81, 2023-05-09:
<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/hp/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/hp/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

  |   |   |   |  
-- | -- | -- | -- | --
0 | 2388 | AC Voltage 1 | diff 10 | V
1 | 2372 | AC Voltage 2 | diff 10 | V
2 | 2367 | AC Voltage 3 | diff 10 | V
3 | 8 | Output Current 1 | diff 10 | A
4 | 9 | Output Current 2 | diff 10 | A
5 | 10 | Output Current 3 | diff 10 | A
6 | 234 | AC Power 1 |   | W
7 | 213 | AC Power 2 |   | W
8 | 195 | AC Power 3 |   | W
9 | 4768 | PV1 Voltage | diff 10 | V
10 | 0 | PV2 Voltage | diff 10 | V
11 | 0 | PV3 Voltage | diff 10 | V
12 | 14 | PV1 Current | diff 10 | A
13 | 0 | PV2 Current | diff 10 | A
14 | 0 | PV3 Current | diff 10 | A
15 | 694 | PV1 Power | diff 10 | W
16 | 0 | PV2 Power | diff 10 | W
17 | 0 | PV3 Power | diff 10 | W
18 | 5001 | Grid Frequency 1 | diff 100 | Hz
19 | 5000 | Grid Frequency 2 | diff 100 | Hz
20 | 5001 | Grid Frequency 3 | diff 100 | Hz
21 | 2 | Inverter Mode |   |  
22 | 245 | Total Energy | diff 10 | kWh
23 | 0 |   |   |  
24 | 171 | Today's Energy | diff 10 | kWh
25 | 6000 | ???? |   |  
26 | 35 | Temperature 1 |   | °C
27 | 32 | Temperature 2 |   | °C
28 | 0 |   |   |  
29 | 0 |   |   |  
30 | 0 |   |   |  
31 | 0 |   |   |  
32 | 0 |   |   |  
33 | 0 |   |   |  
34 | 0 |   |   |  
35 | 0 |   |   |  
36 | 0 |   |   |  
37 | 0 |   |   |  
38 | 0 |   |   |  
39 | 0 |   |   |  
40 | 0 |   |   |  
41 | 0 |   |   |  
42 | 0 |   |   |  
43 | 0 |   |   |  
44 | 0 |   |   |  
45 | 0 |   |   |  
46 | 0 |   |   |  
47 | 0 |   |   |  
48 | 0 |   |   |  
49 | 0 |   |   |  
50 | 0 |   |   |  
51 | 0 |   |   |  
52 | 0 |   |   |  
53 | 0 |   |   |  
54 | 0 |   |   |  
55 | 0 |   |   |  
56 | 0 |   |   |  
57 | 0 |   |   |  
58 | 0 |   |   |  
59 | 0 |   |   |  
60 | 0 |   |   |  
61 | 0 |   |   |  
62 | 0 |   |   |  
63 | 0 |   |   |  
64 | 0 |   |   |  
65 | 0 |   |   |  
66 | 0 |   |   |  
67 | 0 |   |   |  
68 | 0 |   |   |  
69 | 0 |   |   |  
70 | 0 |   |   |  
71 | 0 |   |   |  
72 | 0 |   |   |  
73 | 0 |   |   |  
74 | 0 |   |   |  
75 | 0 |   |   |  
76 | 0 |   |   |  
77 | 0 |   |   |  
78 | 644 | AC Power |   | W
79 | 0 |   |   |  
80 | 1 |   |   |  
81 | 6 |   |   |  
82 | 1 |   |   |  
83 | 0 |   |   |  
84 | 10689 bis 11997 | ???? |   |  

</body>

</html>

simatec, 2023-05-10:
Hmmm die Daten verwirren mich etwas... Laut Spezi hat der nur 2 Strings und keine 3

MagicSven81, 2023-05-10:
Das ist richtig. Aber so sieht der json log aus.Kann ich irgendwie noch eine Gegenprüfung machen?Oder aber die haben wegen der G2 Pro Version bereits das integriert. Die Pro Version hat wohl 3 String Eingänge.--Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

MagicSven81, 2023-05-10:
Auch hier wird von 3 Strings gesprochen wenn man die Daten ausliest.
Somit ist meine Liste oben verifiziert.

Kann ich die selbst in iobroker implementieren oder geht das nur über ein Update durch dich?

MagicSven81, 2023-05-13:
Hi, wäre super wenn du mir anhand der Daten weiter helfen könntest. 
Gerne würde ich auch eine beta Version testen

simatec, 2023-05-16:
Werde die nächsten Tage das ganze mal einfließen lassen.

MagicSven81, 2023-05-17:
Das klingt gut. Dann kann ich ausgiebig testen.

MagicSven81, 2023-05-26:
Hi,
konntest du es schon implementieren?

simatec, 2023-05-26:
Mache ich am WE…

simatec, 2023-05-29:
Teste mal bitte die v0.7.5

MagicSven81, 2023-05-29:
Hier fehlen jetzt die Werte wie:
- acpower
- powerdc1
- powerdc2

[image]

simatec, 2023-05-29:
Bitte den Debuglog posten und mal einen Upload machen

MagicSven81, 2023-05-29:
Habe gerade den Adapter nochmal komplett neu installiert - jetzt klappt alles.
Alle Werte sind da und sind stimmig mit dem SD72M Smartmeter der direkt in der Leitung zum Wechselrichter hängt.

Aktualisierung alle 3 Sekunden funktioniert auch sauber.

[image]

closed 2023-05-29 by MagicSven81