simatec/ioBroker.solax #168: no data on version 0.7.4

Andre-0815-4711, 2023-05-18:
Ich nutze aktuell die Version 0.6.0 und Wifi Stick´s mit der Version 2 und 3.
Stick V2.0 = 2.033.20 und Stick V3.0 = 3.003.02
Wenn ich jetzt auf die Version 0.7.4 aktualisiere bekomme ich keine Daten mehr von den Stick´s V3.0.

simatec, 2023-05-18:
Dann hat du mit sehr großer Wahrscheinlichkeit den Hinweis beim Update ignoriert, dass du bei einem Update auf > 0.7.x in den Einstellungen einmal die Firmware Version des Sticks festlegen musst. 

Gehe in die Einstellungen vom Adapter und stelle doch für deinen Stick mit Version 3 auch auf diese Version

Andre-0815-4711, 2023-05-18:
Moin,

das habe ich in der Tat nicht gelesen. jetzt läuft es.
Bei meinem X3 Hybrid G4 bleibt die InverterTemp auf 0°C, gibt der WR diesen Wert nicht aus?
Dann würde mich interessieren ob es möglich ist die Volladezyklen der Batterie auszulesen, der Solax Support kann diese in der Cloud einsehen, für Enduser ist diese Funktion leider gesperrt.

MfG

closed 2023-05-18 by Andre-0815-4711

simatec, 2023-05-18:
Im Debug Modus siehst du, was aus dem Wechselrichter ausgelesen wird.
Dort kannst du mal schauen, ob du die Werte findest und hier die Ausgabe posten.
Dafür musst du mir dann die Stelle angeben, an der die Ausgabe in dem JSON Format steht.
Die Ausgabe beginnt vom Zählen mit 0

Andre-0815-4711, 2023-05-19:
[image]

Könnten das der Wert 116 (die 0 fehlt im Notepad++) sein?

simatec, 2023-05-23:
Das könnte gut sein... Musst du mal etwas abgleichen und die Werte gegenüberstellen.
Auch die Temperatur musst du mir dann mal geben.
Ich kann mangels Hardware da nichts auslesen...

simatec, 2023-05-29:
Teste mal bitte die v0.7.5

Andre-0815-4711, 2023-05-30:
Läuft, in welcher Datei sind jetzt die Werte der Wechselrichtermodelle versteckt. 
Die tägliche Be- und Entladung der Batterie in kWh hätte ich noch, die Gesamtwerte verstecken sich da glaube auch noch.

simatec, 2023-05-31:
Die Inverter Daten findest du [hier](https://github.com/simatec/ioBroker.solax/blob/master/lib/inverterData.js).
Bitte gib mir die passenden Zuordnungen mit Namen, Einheit und Umrechnungsfaktor, damit ich sie einpflegen kann