simatec/ioBroker.solax #170: Temperatur Wechselrichter

Horstelix, 2023-06-07:
Servus,

ich nutze den Adapter 0.7.5 seit einiger Zeit, echt Klasse!
Hab im Issue #168 gelesen, dass noch die Temperaturdaten zum Wechselrichter fehlen.

Ich habe einen von Q-Cells gelabelten X-3 Hybrid G4 ohne aktive Kühlung und mir eine Kühlung gebastelt, da der WR doch recht warm wird.

Habe heute die Temperatur  an den rückseitigen Kühlrippen mit laufender Kühlung gemessen, dann die Kühlung deaktiviert und anschließend wieder aktiviert. 

[image]

Aufgrund der gemessenen Werte gehe ich davon aus, dass die Temperatur des WR im Wert #54 in °C ausgegeben wird.

Horstelix, 2023-06-13:
Ich hab noch ein wenig weiter recherchiert. #54 ist die Temperatur vom Inverter.

Und ein paar andere Werte hab ich auch noch gefunden:
#74 - Total Battery Discharge Energy (kWh)
#75 - Total Battery Charge Energy (kWh)
#78 - Todays Battery Discharge Energy (kWh / 10)
#79 - Todays Battery Charge Energy (kWh / 10)

thorsten-vogt, 2023-06-14:
> #74 - Total Battery Discharge Energy (kWh / 10)
Kann ich bestätigen
> #75 - Total Battery Charge Energy (kWh / 10)
Es ist #76 
> #78 - Todays Battery Discharge Energy (kWh / 10)
> #79 - Todays Battery Charge Energy (kWh / 10)
Die beiden Werte (78+79) kann ich bestätigen

Ergänzend dazu #75 ist der Überlaufcounter von #74 und #77 ist der Überlaufcounter von #76.

Auch dass #54 ein Temperaturwert vom Inverter ist kann ich bestätigen.
Der SolaX X3 Hybrind G4.2 in der 12/15kW Version hat aktive Lüfter. Ich habe bei mir noch den #46 als weiteren Temperaturwert identifiziert.
Ich habe den 15kW Wechselrichter, Ab einer bestimmten Temperatur geht ein interner Lüfter an (Vielleicht ein CPU Kühler?) und dann gibt es noch die beiden großen Lüfter, die ich jedoch noch nicht in Aktion gesehen habe. 

Ich habe ggf. noch folgenden Wert gefunden:
#106 - Battery remaining energy (kWh / 10)

Horstelix, 2023-06-15:
ich hab mich beim Wert #75 vertan, oben schon korrigiert. 

#46 könnte auch ein Temperaturwert sein, der schwankt bei mir aber nur in einem engen Bereich zwischen 46 und 49.
Ich habe allerdings den 8kW Wechselrichter mit passiver Kühlung. Ob der einen internen Lüfter hat, weiß ich nicht. Zumindest habe ich noch nichts gehört. Auch nicht, als mein Bastelkühler noch nicht in Aktion war.

thorsten-vogt, 2023-06-25:
Ich habe weitere Werte entdeckt:
Nr. 71 - Total Battery Input from Grid (kWh / 10)
Stimmt mit dem Wert, der direkt am Display des Wechselrichters ablesbar ist überein, dort im Menü "On-grid" als "Input Total" ablesbar
Nr. 72 - Ist der Überlauf-Counter von Wert 71
Vermutung: Nr. 73 Todays Battery input from Grid (kWh / 10)
würde zu den anderen Werten im Bereich von Nr. 71-79 passen.

Nr. 46 schwankt bei mir als Temperaturwert von 37-56 Grad Celsius. Ich vermute das ist die CPU Temperatur des Inverters.
An dem Tag im Frühjahr an dem Nr. 46 37Grad warm war, war die Leistungselektronik Nr. 54 nur 27 Grad C. An einem sonnigen Tag wie heute geht die CPU Nr. 46 auf 56 Grad Celsius (wegen dem CPU Lüfter, der bei >50 Grad hörbar ist), Die Leistungselektronik Nr. 54, geht jedoch auf 62 Grad (weiterhin ohne dass die beiden großen Lüfter zuschalten. Mit meinem Ost-West-Dach erreiche ich nur ca. 12.5kW von 15.0 möglichen kW),

thorsten-vogt, 2023-07-23:
Nr. 46 ist der interne Temperatursensor. Bei 53°C geht der interne Lüfter an und bei 52°C geht der interne Lüfter wieder aus. Ich habe leider keinen weiteren Wert gefunden, der auf einen Lüfterstatus oder Lüfterdrehzahl hinweisen könnte.

erwini, 2023-07-25:
Werden die neu gefundenen Werte irgendwann eingebaut, oder kann man das auch selber  machen?

simatec, 2023-07-27:
Ja die Daten werden eingepflegt ...

simatec, 2023-07-30:
@Horstelix  @thorsten-vogt 
Könnte bitte eine von euch noch einmal die Werte bitte zusammenfassen? Am besten direkt mit Wert, Bezeichnung, Einheit und Umrechnungsfaktor

erwini, 2023-07-30:
Adapterupdate 0.7.6 installiert und Werte werden angezeigt. Funktioniert somit bestens. 
Die "Nr. 71 - Total Battery Input from Grid (kWh / 10)" ist wohl (noch?) nicht eingebaut, aber das kommt glaub ich auch selten vor ...
Vielen besten Dank!!

thorsten-vogt, 2023-07-31:
Ich habe auch das Adapterupdate 0.7.6 installiert. Bei mir fehlt auch noch der Wert Nr. 71 und noch der interne Temperaturwert Nr. 46 (ggf. nur bei den 12+15kW Wechselrichtern vorhanden). Der Rest sieht sehr gut aus, vielen Dank!

Horstelix, 2023-08-01:
Servus,

hab 0.7.6 gestern auch installiert, funktioniert alles.
Den Temperaturwert Nr. 46 gibts auch bei meinem Wechselrichter (8 kW, ohne Lüfter)

Danke an simatec

closed 2023-10-07 by Horstelix