simatec/ioBroker.solax #171: Überlauf der Summenwerte

thorsten-vogt, 2023-06-11:
Ich habe einen SolaX X3 Hybridwechselrichter 15.0 G4.2 im lokalen modus mit dem Adapter 0.7.5

Ich habe ein Überlaufen der Summenwerte beobachtet:

[image]

Der Wert für 
Total PV Energy / data.totalpvenergy (Zeile 80)
ist bei mir 1x "Übergelaufen", daher ist jetzt in Zeile 81 eine 1
Der Überlauf ist bei 6554 in Zeile 80 passiert, lt. Grafana.
Total Feed-in Energ	/ data.totalFeed (Zeile 86)
ist bei mit 7 mal "Übergelaufen" daher ist jetzt in Zeile 87 eine 7
Der Überlauf ist bei 655 in Zeile 87 passiert, lt. Grafana.

Anbei mein aktuelles Ergebnis des curl:
`{"sn":"XXXXXX","ver":"3.008.10","type":14,"Data":[2358,2408,2365,163,156,155,3844,3762,3667,11273,4891,6235,134,84,6556,5271,5002,5002,5001,2,0,0,0,0,0,0,0,0,0,0,0,0,0,1,10816,0,0,0,0,47910,0,0,4779,0,0,1,53,457,256,4612,2829,5894,100,0,60,0,0,0,0,0,0,0,0,0,0,0,0,0,64966,0,469,116,0,0,9876,0,9763,0,29,43,3056,1,506,0,0,0,17523,7,41830,0,4010,0,0,0,0,0,0,0,0,0,0,0,1,76,1,29,133,256,5256,3600,0,350,215,193,33,33,43,2139,258,9766,9252,9252,8995,0,0,3338,3301,26234,30,21302,14389,18757,12609,16692,12611,14384,21302,14389,18757,12609,16692,12611,14384,21302,14389,18754,12345,16694,12867,14131,21302,14389,18754,12345,16694,12355,13872,21302,14389,18754,12849,16688,12867,12848,1281,1799,514,769,0,47910,0,0,0,0,16,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[15.000,14,"XXXXXXX",8,1.30,0.00,1.28,1.09,0.00,1]}`

Ich vermute das
Total Consumption	/ data.totalconsumption (Zeile 88)
und 
Inverter AC-Energy out total	/ data.yieldtotal (Zeile 68)
beide jeweils in den Zeilen darauf (89 und 69) Überlaufcounter haben. Da die Werte bei mir noch keinen "Überlauf" erreicht haben, also noch bei 0 stehen, kann ich das noch nicht bestätigen.

Ich weiß jetzt nicht was besser ist. Ich würde eine "fertige" Aufmultiplizierung der "Total" Werte bevorzugen. Evtl. zusätzlich die Ausgabe der Überlaufcounter und ggf. die "Rohdaten" für die Nutzer, die sich bereits selbst etwas gebaut haben?

Nebenbei ist mir aufgefallen dass die Werte:
23	 EPS 1 Voltage	data.voltageeps1
24	 EPS 2 Voltage	data.voltageeps2
25	 EPS 3 Voltage	data.voltageeps3
26	EPS 1 Current	data.currenteps1
27	EPS 2 Current	data.currenteps2
28	EPS 3 Current	data.currenteps3
alle W als Maßeinheit haben, dass müsste V bzw. A sein.

Vielen Dank für die viele Arbeit, die Du in dieses Projekt investierst!

simatec, 2023-07-30:
Kannst du hier bitte die genauen Werte und Umrechnungsfaktoren ermitteln?
Aufgrund der fehlenden Hardware wäre das für mich Glaskugel schauen ;-)

thorsten-vogt, 2023-07-31:
Ich habe versucht alles Überlaufwerte in einer Tabelle zusammen zu fassen:

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:\Users\Thorsten\AppData\Local\Temp\msohtmlclip1\01\clip.htm">
<link rel=File-List
href="file:///C:\Users\Thorsten\AppData\Local\Temp\msohtmlclip1\01\clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

Zeile | Beschreibung | Parameter | Umrechnung | Maßeinheit | Überlaufzeile | Hinzurechnung
-- | -- | -- | -- | -- | -- | --
68 | Inverter AC-Energy out total | data.yieldtotal | diff 10 | kWh | 69 | 6553,5
71 | 0,1kw From Grid to Battery total | (fehlt noch) | diff 10 | kWh | 72 | 6553,5
74 | 0,1 kw aus akku 74 | solax.0.data.totalBatteryChargeEnergy | diff 10 | kWh | 75 | 6553,5
76 | 0,1 kw in akku 76 | solax.0.data.totalBatteryChargeEnergy | diff 10 | kWh | 77 | 6553,5
80 | Total PV Energy | data.totalpvenergy | diff 10 | kWh | 80 | 6553,5
86 | Total Feed-in Energ | data.totalFeed | diff 100 | kWh | 87 | 655,35
88 | Total Consumption | data.totalconsumption | diff 100 | kWh | 89 | 655,35
90 | Feed in Energy | data.feedinenergy | diff 100 | kWh | 91 | 655,35
92 | Consume Energy | data.consumeenergy | diff 100 | kWh | 93 | 655,35

</body>

</html>

Ich gehe von dem maximalen Integer, also 65535 als Überlaufwert aus und dann entsprechend durch 10 oder durch 100 dividiert.

simatec, 2023-08-02:
Ehrlich gesagt stehe ich mit deinen Überlaufzeilen auf dem Schlauch... Was soll ich damit genau anfangen?

thorsten-vogt, 2023-08-04:
Hallo @simatec, 

erstmal vielen Dank für Deinen Einsatz dieses Projekt voranzutreiben!

Ich versuche mein Anliegen noch mal anders zu erläutern. 

Zuerst am Beispiel von data.totalFeed (Nr. 86). Wenn ich das mit Grafana visualisiere, dann erhalte ich folgende Abbildung:
[image]
Man kann 11 "Überläufe" von 655,35 auf 0 erkennen. Daher steht bei mir der Wert Nr. 87 auf 11.

`{"sn":"SVLPMFYCFC","ver":"3.009.03","type":14,"Data":[2341,2363,2352,39,41,34,907,939,783,2629,5020,6335,37,13,1882,828,4998,5000,4999,2,0,0,0,0,0,0,0,0,0,0,0,0,0,1,2081,0,0,0,0,48220,0,0,4800,0,0,1,44,548,256,11540,1041,5896,100,0,38,0,0,0,0,0,0,0,0,0,0,0,0,0,36302,1,369,116,0,0,12633,0,13007,0,50,196,41835,1,531,0,0,0,55794,11,42083,0,2767,0,2,0,0,0,0,0,0,0,0,0,1,99,1,23,223,256,5256,3600,0,350,254,213,33,33,55,2139,259,9766,9252,9252,8995,0,0,3338,3330,47856,1,21302,14389,19013,12850,16691,12359,14132,21302,14389,19013,12850,16691,12359,14132,21302,14389,18754,12345,16694,12867,14131,21302,14389,18754,12345,16694,12355,13872,21302,14389,18754,12849,16688,12867,12848,513,3844,1284,1029,0,48220,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[15.000,14,"H34C15I4657059",8,1.33,0.00,1.31,1.09,0.00,1]}`

Ich könnte mir eine der beiden Varianten vorstellen:

Variante A:
Berechnung innerhalb des Adapters, in der Art:
data.totalFeed = (Wert von Nr. 86 + (Wert von 87 * 65535)) / 100

Variante B:
Ausgabe der Überlaufe, damit man die Berechnung in Grafana etc. durchführen kann:
data.totalFeed = Wert von Nr. 86
data.totalFeedRollover = Wert von Nr. 87

Ich würde Variante A bevorzugen.

Zweites Beispiel data.yieldtotal (Nr. 68) mit nur einem "Überlauf" (Nr. 69):
[image]

Hier würde Variante A so aussehen:
data.yieldtotal = (Wert von Nr. 68 + (Wert von 69 * 65535)) / 10

Meine Tabelle oben noch mal anders "Formuliert":
data.yieldtotal = (Wert von Nr.68 + (Wert von Nr.69 * 65535)) / 10
(fehlt noch) = (Wert von Nr.71 + (Wert von Nr.72 * 65535)) / 10
data.totalBatteryDischargeEnergy = (Wert von Nr.74 + (Wert von Nr.75 * 65535)) / 10
data.totalBatteryChargeEnergy = (Wert von Nr.76 + (Wert von Nr.77 * 65535)) / 10
data.totalpvenergy = (Wert von Nr.80 + (Wert von Nr.80 * 65535)) / 10
data.totalFeed = (Wert von Nr.86 + (Wert von Nr.87 * 65535)) / 100
data.totalconsumption = (Wert von Nr.88 + (Wert von Nr.89 * 65535)) / 100
data.feedinenergy = (Wert von Nr.90 + (Wert von Nr.91 * 65535)) / 100
data.consumeenergy = (Wert von Nr.92 + (Wert von Nr.93 * 65535)) / 100

Ich hoffe damit konnte ich weiter helfen und mich verständlicher ausdrücken.

simatec, 2024-01-01:
In der nächsten Version werden die Überlaufwerte mit ausgegeben, damit man sie in Grafana anpassen und berechnen kann.

closed 2024-01-01 by simatec