simatec/ioBroker.solax #174: History Daten werden zu spät kopiert

erwini, 2023-07-25:
**Describe the bug**  
Die History Daten werden exakt 2 Stunde zu spät ausgelesen und geschrieben.

**Expected behavior**  
Die Daten sollten um 23:59:50 gelesen werden

Das sieht nach einem Zeitzonenproblem aus. Wie und wo kann man das beeinflussen? Einstellung am Adapter selber habe ich keine gefunden.

simatec, 2023-07-25:
Da musst du dein Betriebssystem prüfen

erwini, 2023-07-25:
date auf der linux cli gibt mir die aktuelle Zeit korrekt in UTC wieder, also -2h zur lokalen CEST ...

simatec, 2023-07-25:
Naja das ist ja nicht korrekt... es sollte dir die lokale Zeit ausgeben. Also musst du sie anpassen

erwini, 2023-07-29:
Linux auf lokale Zeitzone umgestellt und Adapter restartet. Funktioniert.

closed 2023-07-29 by erwini