simatec/ioBroker.solax #175: Adapter verliert Verbindung zur Solaranlage nach Wechsel des Clients im Heimnetz von Router auf Repeater

jamakr4, 2023-09-05:
Sobald der Client (welcher die Daten abfragt) im Heimnetz seine Verbindung vom Router auf z.B einen Repeater wechselt verliert der Adapter seine Verbindung zur Anlage. Der Adapter muss dann wieder neu gestartet werden um wieder eine Verbindung herzustellen und Daten zu liefern. Die Abfrage der Daten erfolgt lokal.

thorsten-vogt, 2023-09-05:
Ich hatte so ein Problem auch mal bei der lokalen Verbindung festgestellt. Die Cloud-Verbindung hatte nie das Problem.

Ich habe das durch dieses blockly-Skript "gelöst":
[image]

Ich muss aber auch dazu sagen, dass ich nicht weiß, ob das Problem noch existiert. Die Debug-Ausgabe von dem Skript habe ich mit der aktuellen Version nicht mehr im Log entdeckt. Ich habe aber auch meine WLAN-Abdeckung verbessert.

So sieht das in Javaskript aus:
`var stop2, timeout, timeout2;

stop2 = false;
on({id: [].concat(['solax.0.info.online']).concat(['system.adapter.solax.0.alive']), val: false}, async function (obj) {
    if (stop2 == false) {
    stop2 = true;
    setState("system.adapter.solax.0.alive"/*solax.0 alive*/, false);
    timeout = setTimeout(async function () {
      setState("system.adapter.solax.0.alive"/*solax.0 alive*/, true);
      console.warn('Der Solax.0 Adapter wurde vom Skript neu gestartet!');
      timeout2 = setTimeout(async function () {
        stop2 = false;
      }, 30000);
    }, 1000);
  }
});`

simatec, 2023-09-06:
> Sobald der Client (welcher die Daten abfragt) im Heimnetz seine Verbindung vom Router auf z.B. einen Repeater wechselt verliert der Adapter seine Verbindung zur Anlage. Der Adapter muss dann wieder [...]

Im Code ist die Info Connection Anzeige so verarbeitet, dass ein Abruf der Daten vom WR erfolgt. Ist dieser erfolgreich und es besteht somit eine Verbindung, wird die Anzeige grün. Das selber Spiel passiert auch Nachts... Der WR ist nicht erreichbar, da kein Strom produziert wird... Dementsprechend ist die Anzeige Orange mit Ausrufezeichen, da keine Verbindung besteht. Trotzdem versucht der Adapter im Intervall abzufragen und wartet auf die Wiedererreichbarkeit, was dann auch am Morgen der Fall ist.

D.h. der Adapter ist immer aktiv und versucht im eingestellten Intervall Daten abzurufen. Geliengt das, wird die Connection State Grün... andernfalls kommt das orange Ausrufezeichen.

In eurem Fall wäre lokal bei euch zu prüfen, warum es keine Verbindung gibt, bzw. sie verloren geht. 
Ich habe die Erfahrung gemacht, dass der Stick ein gutes und stabiles WLAN Signal braucht. Bei mir ging vor Austausch eines Repeaters ebenfalls die Verbindung ab und an verloren.

closed 2023-12-31 by simatec