simatec/ioBroker.solax #178: Data for Inverter X1 Hybrid G4 are not correct

trottellumme0, 2023-10-22:
Please add the Inverter Solax X1 Hybrid G4 to the adapter. At the moment, when operating this inverter, the data is displayed at incorrect data points.

`{"inverterSN":"xxxxxxxxxxxxxxx","sn":"xxxxxxxxxxxx","inverterType":"X1-Hybrid-G4","inverterStatus":"Wait Mode","uploadTime":"Thu Oct 19 2023 21:56:17 GMT+0200 (Mitteleuropäische Sommerzeit)","connectType":"local","online":true,"connection":true,"firmwareVersion":"3.006.04","totalSize":5,"acvoltage1":0,"acvoltage2":0,"acvoltage3":0,"accurrent1":0,"accurrent2":0,"accurrent3":0,"acpower1":0,"acpower2":0,"acpower3":0,"acpower":0,"dcvoltage1":0,"dcvoltage2":20.900000000000002,"dccurrent1":0,"dccurrent2":0.30000000000000004,"dcpower1":0,"dcpower2":0,"acfrequency1":0,"acfrequency2":0,"acfrequency3":0,"voltageeps1":0,"voltageeps2":10,"voltageeps3":0,"currenteps1":0,"currenteps2":4,"currenteps3":0,"powereps1":0,"powereps2":0,"powereps3":0,"feedinpower":512,"batteryVoltage":0,"batteryCurrent":0,"batteryPower":0,"cpuTemperature":0,"powerConsumer":0,"inverterTemperature":219,"yieldtotal":0,"yieldtoday":0,"totalBatteryInputFromGrid":0,"totalBatteryDischargeEnergy":300,"totalBatteryChargeEnergy":6253.6,"todaysBatteryDischargeEnergy":0,"todaysBatteryChargeEnergy":0,"totalpvenergy":0,"pvenergy":0,"totalFeed":0,"totalconsumption":0,"feedinenergy":0,"consumeenergy":0.2,"batteryCapacityRemainig":0,"batteryTemperature":0,"batteryRemainingEnergy":0,"batteryChargingCycles ":0}`

`2023-10-19 11:43:16.664 - [34mdebug[39m: solax.3 (10170) local request: {"sn":"xxxxxxxxxxxx","ver":"3.006.04","type":15,"Data":[2310,10,191,5000,2217,0,8,0,192,0,2,208,0,2,0,0,0,0,0,0,0,0,0,0,100,0,17,3598,0,0,0,0,0,0,512,0,20,0,191,28,0,0,0,0,0,0,0,0,0,0,0,1,1,0,217,0,65345,65535,65386,65535,0,0,0,0,0,0,0,0,0,0,0,0,0,0,3000,0,62536,65535,0,0,0,0,0,0,0,2,0,0,0,0,0,0,20,0,0,0,0,0,0,11008,4875,5898,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[5,15,"Hxxxxxxxxxxxxxx",8,1.29,0,1.28,1.04,0,1]}`

Many thanks

simatec, 2023-10-23:
Please test v0.9.0

trottellumme0, 2023-11-03:
Looks good, but there are still these points:
1. The Info message in log:
`2023-11-03 19:41:23.494	info	State value to set for "solax.3.info.inverterStatus" has to be type "string" but received type "number"`
I think, this is the value at adress 10 and 2 means "Normal Mode" and 0 means "Wait Mode".

2. The value at adress 2 goes from 4 to 3 to 2 to 1 to 0 and then to 65531... I think it is a negative inverter ac power...
`2023-11-03 15:55:23.394  - [34mdebug[39m: solax.3 (15892) local request: {"sn":"xxxxxxxxxxxxx","ver":"3.006.04","type":15,"Data":[2315,6,4,4997,701,0,1,0,10,0,2,353,0,13,0,0,0,0,0,0,0,0,0,0,100,0,19,3600,0,0,0,0,0,0,512,0,50,0,4,29,0,0,0,0,0,0,0,0,0,0,0,1,3,0,369,0,65531,65535,65388,65535,0,0,0,0,0,0,0,0,0,0,0,0,0,0,3000,0,62536,65535,0,0,0,0,0,0,0,12,0,0,0,0,0,0,50,0,0,0,0,0,0,14091,783,5899,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[5,15,"Hxxxxxxxxxxxxxxx",8,1.29,0,1.28,1.04,0,1]}
2023-11-03 15:56:23.422  - [34mdebug[39m: solax.3 (15892) local request: {"sn":"xxxxxxxxxxxxx","ver":"3.006.04","type":15,"Data":[2298,6,4,4997,701,0,1,0,9,0,2,353,0,13,0,0,0,0,0,0,0,0,0,0,100,0,19,3597,0,0,0,0,0,0,512,0,50,0,4,29,0,0,0,0,0,0,0,0,0,0,0,1,3,0,369,0,65531,65535,65386,65535,0,0,0,0,0,0,0,0,0,0,0,0,0,0,3000,0,62536,65535,0,0,0,0,0,0,0,12,0,0,0,0,0,0,50,0,0,0,0,0,0,14347,783,5899,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[5,15,"Hxxxxxxxxxxxxxxx",8,1.29,0,1.28,1.04,0,1]}
2023-11-03 15:57:23.427  - [34mdebug[39m: solax.3 (15892) local request: {"sn":"xxxxxxxxxxxxx","ver":"3.006.04","type":15,"Data":[2314,6,3,5001,701,0,1,0,9,0,2,353,0,13,0,0,0,0,0,0,0,0,0,0,100,0,19,3603,0,0,0,0,0,0,512,0,50,0,3,29,0,0,0,0,0,0,0,0,0,0,0,1,3,0,369,0,65533,65535,65387,65535,0,0,0,0,0,0,0,0,0,0,0,0,0,0,3000,0,62536,65535,0,0,0,0,0,0,0,12,0,0,0,0,0,0,50,0,0,0,0,0,0,14603,783,5899,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[5,15,"Hxxxxxxxxxxxxxxx",8,1.29,0,1.28,1.04,0,1]}
2023-11-03 15:58:23.432  - [34mdebug[39m: solax.3 (15892) local request: {"sn":"xxxxxxxxxxxxx","ver":"3.006.04","type":15,"Data":[2331,6,2,4998,701,0,0,0,5,0,2,353,0,13,0,0,0,0,0,0,0,0,0,0,100,0,19,3615,0,0,0,0,0,0,512,0,50,0,2,29,0,0,0,0,0,0,0,0,0,0,0,1,3,0,369,0,65533,65535,65387,65535,0,0,0,0,0,0,0,0,0,0,0,0,0,0,3000,0,62536,65535,0,0,0,0,0,0,0,12,0,0,0,0,0,0,50,0,0,0,0,0,0,14859,783,5899,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[5,15,"Hxxxxxxxxxxxxxxx",8,1.29,0,1.28,1.04,0,1]}
2023-11-03 15:59:23.459  - [34mdebug[39m: solax.3 (15892) local request: {"sn":"xxxxxxxxxxxxx","ver":"3.006.04","type":15,"Data":[2314,6,2,4999,700,0,0,0,6,0,2,353,0,13,0,0,0,0,0,0,0,0,0,0,100,0,19,3598,0,0,0,0,0,0,512,0,50,0,2,29,0,0,0,0,0,0,0,0,0,0,0,1,3,0,369,0,65534,65535,65385,65535,0,0,0,0,0,0,0,0,0,0,0,0,0,0,3000,0,62536,65535,0,0,0,0,0,0,0,12,0,0,0,0,0,0,50,0,0,0,0,0,0,15115,783,5899,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[5,15,"Hxxxxxxxxxxxxxxx",8,1.29,0,1.28,1.04,0,1]}
2023-11-03 16:00:23.388  - [34mdebug[39m: solax.3 (15892) local request: {"sn":"xxxxxxxxxxxxx","ver":"3.006.04","type":15,"Data":[2325,6,1,5002,701,0,0,0,6,0,2,353,0,13,0,0,0,0,0,0,0,0,0,0,100,0,19,3607,0,0,0,0,0,0,512,0,50,0,1,29,0,0,0,0,0,0,0,0,0,0,0,1,3,0,369,0,65530,65535,65390,65535,0,0,0,0,0,0,0,0,0,0,0,0,0,0,3000,0,62536,65535,0,0,0,0,0,0,0,12,0,0,0,0,0,0,50,0,0,0,0,0,0,11,784,5899,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[5,15,"Hxxxxxxxxxxxxxxx",8,1.29,0,1.28,1.04,0,1]}
2023-11-03 16:01:23.451  - [34mdebug[39m: solax.3 (15892) local request: {"sn":"xxxxxxxxxxxxx","ver":"3.006.04","type":15,"Data":[2324,6,1,5000,701,0,1,0,8,0,2,353,0,13,0,0,0,0,0,0,0,0,0,0,100,0,19,3600,0,0,0,0,0,0,512,0,50,0,1,29,0,0,0,0,0,0,0,0,0,0,0,1,3,0,369,0,65534,65535,65387,65535,0,0,0,0,0,0,0,0,0,0,0,0,0,0,3000,0,62536,65535,0,0,0,0,0,0,0,12,0,0,0,0,0,0,50,0,0,0,0,0,0,267,784,5899,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[5,15,"Hxxxxxxxxxxxxxxx",8,1.29,0,1.28,1.04,0,1]}
2023-11-03 16:02:23.456  - [34mdebug[39m: solax.3 (15892) local request: {"sn":"xxxxxxxxxxxxx","ver":"3.006.04","type":15,"Data":[2324,6,1,5001,700,0,0,0,6,0,2,353,0,13,0,0,0,0,0,0,0,0,0,0,100,0,19,3603,0,0,0,0,0,0,512,0,50,0,1,29,0,0,0,0,0,0,0,0,0,0,0,1,3,0,369,0,65533,65535,65387,65535,0,0,0,0,0,0,0,0,0,0,0,0,0,0,3000,0,62536,65535,0,0,0,0,0,0,0,12,0,0,0,0,0,0,50,0,0,0,0,0,0,523,784,5899,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0 [... cut ...] ,0,65387,65535,0,0,0,0,0,0,0,0,0,0,0,0,0,0,3000,0,62536,65535,0,0,0,0,0,0,0,12,0,0,0,0,0,0,50,0,0,0,0,0,0,2571,784,5899,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[5,15,"Hxxxxxxxxxxxxxxx",8,1.29,0,1.28,1.04,0,1]}`

In the  next few days I will take a screenshot from SolaxCloud at the time when the value is 655xx.

Many thanks. :-)

simatec, 2024-01-01:
Fix in the next Version

closed 2024-01-01 by simatec