simatec/ioBroker.solax #180: Wallbox Integration

Andre-0815-4711, 2023-10-30:
Moin,

anbei ein paar Werte die ich schon von der Wallbox identifizieren konnte.
`{ /****************************************** X1/X3-EVC *****************************************/
        isOnline: { name: 'info.online', description: 'Charger Online', type: 'boolean', role: 'switch' },
        2: { name: 'data.acvoltage1', description: 'Grid Voltage 1', type: 'number', multiplier: 0.1, unit: 'V', role: 'value.power' }, // 'AC Voltage 1': (0, 'V'),
        3: { name: 'data.acvoltage2', description: 'Grid Voltage 2', type: 'number', multiplier: 0.1, unit: 'V', role: 'value.power' }, // 'AC Voltage 2': (1, 'V'),
        4: { name: 'data.acvoltage3', description: 'Grid Voltage 3', type: 'number', multiplier: 0.1, unit: 'V', role: 'value.power' }, // 'AC Voltage 3': (2, 'V'),
        13: { name: 'data.accurrent1', description: 'Charge Current 1', type: 'number', maxValue: 32768, multiplier: 0.1, unit: 'A', role: 'value.power' }, // 'Output Current 1': (3, 'A'),
        14: { name: 'data.accurrent2', description: 'Charge Current 2', type: 'number', maxValue: 32768, multiplier: 0.1, unit: 'A', role: 'value.power' }, // 'Output Current 2': (4, 'A'),
        15: { name: 'data.accurrent3', description: 'Charge Current 3', type: 'number', maxValue: 32768, multiplier: 0.1, unit: 'A', role: 'value.power' }, // 'Output Current 3': (5, 'A'),
        16: { name: 'data.accurrent1', description: 'Grid Current 1', type: 'number', maxValue: 32768, multiplier: 0.1, unit: 'A', role: 'value.power' }, // 'Grid Current 1': (3, 'A'),
        17: { name: 'data.accurrent2', description: 'Grid Current 2', type: 'number', maxValue: 32768, multiplier: 0.1, unit: 'A', role: 'value.power' }, // 'Grid Current 2': (4, 'A'),
        18: { name: 'data.accurrent3', description: 'Grid Current 3', type: 'number', maxValue: 32768, multiplier: 0.1, unit: 'A', role: 'value.power' }, // 'Grid Current 3': (5, 'A'),
        23: { name: 'data.plugTemperature', description: 'Plug temperature', type: 'number', unit: '°C', role: 'value.temperature' }, // 'Ladepistolen Temperature,
        24: { name: 'data.boardTemperature', description: 'Mainboard temperature', type: 'number', unit: '°C', role: 'value.temperature' }, // 'Mainboard Temperature,
}
`
information_dataPoints:
`{ /***************************************** X1/X3-EVC ***************************************/
                    0: { name: 'info.totalSize', description: 'Total Size of Power', type: 'number', unit: 'kW', role: 'value.power' },
                    2: { name: 'info.chargerSN', description: 'Unique identifier of charger (Serial No.)', type: 'string', role: 'text' },
	            4: { name: 'info.version', description: 'Version', type: 'string', role: 'text' },`

Hier noch einmal der komplette Datensatz:
 {"SN":"SQKASDFGHJ","ver":"3.004.11","type":1,"Data":[0,0,23226,23502,23810,0,0,0,0,0,0,0,0,0,0,0,73,65494,65485,0,0,0,0,0,20,0,0,15,0,0,0,0,0,4994,5002,5000,4916,7690,5898,1,0,0,0,0,0,0,0,0,1,100,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1,1,1,0,0,119,1204,1000,64876,1],"Information":[11,1,"C3110123456789",1,1.08,1.01,0,0,0,1],"OCPPServer":"","OCPPChargerId":""}
 
Wenn mein Auto da ist gibt es ein update der Werte.

simatec, 2023-12-31:
> Wenn man Zeile 1 später noch schreiben könnte, wäre das ein Knaller!

Was meinst du damit genau?

Azrael4224, 2023-12-31:
> 
> Was meinst du damit genau?

Die Box unterstützt verschiedene Lademodi, die man entweder über die WB selbst oder über die App ändern kann. Leider finde ich keine Möglichkeit, den Modus über IOBroker zu steuern. 
Daher mein Wunschdenken, den Wert an die Wallbox zurück zu schreiben. 
Allerdings weiß ich nicht, ob der Adapter schreiben unterstützt.

Das wären die Werte:
'0= Stop 1= Fast 2=Green 3=Eco'

Danke und guten Rutsch!
Rich

simatec, 2024-01-01:
In der nächsten Version hinzugefügt...

closed 2024-01-01 by simatec