simatec/ioBroker.solax #183: Adapter verliert Cloud Verbindung

cdaerr, 2023-11-10:
Heute morgen gegen 3:30 hat der Adapter die Verbindung zur SolaX Cloud verloren.
Der direkte Aufruf der API URL, so wie sie in der SolaX Cloud beschrieben wir funktioniert einwandfrei.

Eine Umstelelung des Loglevel auf debug zeigte folgendes:
[image]

**Versions:**  
 - Adapter version: 0.7.7
 - JS-Controller version: 5.0.12
 - Node version: v16.20.2
 - Operating system: linux

simatec, 2023-11-10:
Dann war sicher die Cloud nicht erreichbar .. Da kann der Adapter dann leider nichts machen

cdaerr, 2023-11-10:
Nein, Cloud ist sowohl über App als auch API URL erreichbar.

Problem hat nur der Adapter.

simatec, 2023-11-10:
Welche URL verwendest du für den API Abruf?

cdaerr, 2023-11-10:
Steht in der Fehlerbeschreibung:

Die API, wie in der SolaX Cloud beschrieben mit Key und Seriennummer.

cdaerr, 2023-11-10:
[image]

You see it works. It is not a problem of SolaX Cloud

simatec, 2023-11-10:
Teste mal bitte folgende URL

`https://www.eu.solaxcloud.com:9443/proxy/api/getRealtimeInfo.do?tokenId=xxxxxxxx&sn=xxxxxxxx`

cdaerr, 2023-11-10:
Timeout:
[image]

simatec, 2023-11-10:
Dann hast du die Lösung ... Der Adapter verwendet diese URL ...

simatec, 2023-11-10:
Mein Vorschlag ... teste mal die lokale Verbindung ... Die bringt am ende wesentlich mehr

simatec, 2023-11-10:
> steht eigentlich alles in der Fehlerbeschreibung

da stand keine URL

cdaerr, 2023-11-10:
Habe einen LAN Donlge, keinen WLAN Dongle, der unterstützt keine Lokale Verbindung.

cdaerr, 2023-11-10:
> > steht eigentlich alles in der Fehlerbeschreibung
> 
> da stand keine URL

Es ist die URL, die du bekommst, wenn du dich in deinem Cloud Konto einloggst. Also die "offizielle"

Matt77CHE, 2023-11-10:
Also bei mir ist das Selbe Problem, verwende den WLAN-Dongle.

cdaerr, 2023-11-10:
Ja, bei mir auch. Der Zugriff über Port 9443 funktioniert auch wieder, auch wenn es nicht die "offizielle" URL für den API Zugriff ist.

simatec, 2023-11-10:
> Vielleicht wäre aber eine Umstellung auf die in der Cloud angegebene URL aber trotzdem sinnvoll?

Der Wechsel fand stand, da die offizielle URL oft Timeouts hatte

closed 2023-11-10 by simatec

Matt77CHE, 2023-11-10:
Ok - bin soweit gekommen, dass ich auf das Web-Interface Zugriff habe - meiner Version ist 3.001.02 und wenn ich versuche das Downgrade zu machen, bekomme ich ein fail und er bleibt auf der 3.001.02