simatec/ioBroker.solax #185: Solax Pocket Wifi 2 mit Firmware 3.x.x.x

winampdevil, 2023-12-11:
Hallo, ich möchte meinen Solax X3 Hybrid G2 über eine lokale Verbindung mit iobroker verbinden und habe dazu einen Pocket Wifi V2 (grau).
Als Firmware ist hier v3.001.02 installiert. In der Beschreibung taucht diese HW/SW-Kombination nicht auf, ein Downgrade auf v2.x schlägt fehl.
Die Verbindung iobroker - solax wifi wird hergestellt, aber die Inverterwerte sind allesamt 0.
Ist dieses Verhalten schon einmal vorgekommen?

Danke & Grüße
Frank

Nachtrag:
Über den LAN-Port des WR wird eine Cloudverbindung hergestellt, das funktioniert.

kommando828, 2023-12-12:
Can't help but can confirm some Wifi V2's do come with V3 firmware as I also have one.

simatec, 2023-12-12:
https://github.com/simatec/ioBroker.solax/blob/master/docs/de/solax.md#experteneinstellungen

Zitat:
Achtung, wer in den Experteneinstellungen den lokalen Modus aktiviert sollte im Vorfeld zwingend die aktuelle Firmwareversion seines Pocket Wifi Sticks prüfen. **Eine Firmware Version größer 2.30.20 (Wifi-Pocket V1/V2) und kleiner als 3.001 (Wifi-Pocket V3) darf der Stick nicht installiert haben**, da Solax in höheren Versionen den lokalen Zugriff blockiert und es zu einem Absturz des Wifi Sticks führt.

winampdevil, 2023-12-12:
@simatec 
Die Firmwarevoraussetzungen sind mir bekannt.
Leider hat mein Wifi-Pocket **V2** bereits eine Firmware **3.001**

Der beschriebene Downgradepfad (s.u.) funktioniert leider nicht.

Zitat:
"Sollte die Version größer 2.033.20 (Wifi-Pocket V1/V2) und kleiner 3.001 (Wifi-Pocket V3) sein, könnt ihr im gleichen Tab über den Menüpunkt "Update Firmware (.usb)" die korrekte Version flashen."

Sollte ein V2-Stick mit v3-Firmware (auch wenn sie >= 3.001 ist) nicht funktionieren, dann muss ich schauen, ob ein einen anderen Stick bekomme.

simatec, 2023-12-12:
Ich hatte damals auch Probleme beim flashen der Firmware. Bei mir lag es an der Stromversorgung des Sticks. Nachdem ich ein USB Netzteil zur Stromversorgung genommen habe, lief der Downgrade problemlos...

winampdevil, 2023-12-15:
Ich habe nun mit einer externen 3,3V-Spannungsquelle gearbeitet und versucht, den Stick mit der v2-Firmware zu flashen -> Kein Erfolg. (Fehler: Firmware Check fail, system reboot)

Werde es nun mit Freepocket versuchen.
https://blog.chrisoft.io/blog/2023/10/01/freepocket-creacion-de-un-pincho-alternativo-para-inversores-solax/
Aktuell liefert der daran angeschlossenen Solax X3-Hybrid G2 (oder G3) sekündlich Werte...

winampdevil, 2023-12-21:
Bevor ich diesen Fall schließe, hier meine Erkenntnisse:
- Solax Pocket Wifi v2 (weißer Stick) mit Firmware v3.x lässt sich nicht auf Firmware v2.x downgraden.
- Solax Pocket Wifi v2 (weißer Stick) mit Firmware v2.x, die größrer v2.033 ist, lässt sich auf Firmware v2.033 downgraden.
- Freepocket (siehe obiger Link) funktioniert stabil, liefert aber teilweise falsche Werte (z.B. Spannung des Akkus ist um Faktor 10 zu hoch)

closed 2023-12-21 by winampdevil

kommando828, 2023-12-21:
No battery here so Freepocket will work for me then.