simatec/ioBroker.solax #189: X3 G4 Overflow Values

thorsten-vogt, 2024-01-04:
Hi simatec,

thanks for adding the Overflow Values for the X3 G4 Model.

Could you change the Value from kw to "Count" or "Number" for these Values. There should also be no multiplier.
Perhaps you could also add an "l" to "Overfow" at 77/87/93 to have "Overflow".

69: { name: 'data.yieldtotalOverflow', description: 'Inverter AC-Energy out total Overflow', type: 'number', multiplier: 0.1, unit: 'kWh', role: 'value.power.consumption' }, // 'Total Energy': (19, 'kWh'),
75: { name: 'data.totalBatteryDischargeEnergyOverflow', description: 'Total Battery Discharge Energy Overflow', type: 'number', multiplier: 0.1, unit: 'kWh', role: 'value.power.consumption' }, // Total Battery Discharge Energy
77: { name: 'data.totalBatteryChargeEnergyOverfow', description: 'Total Battery Charge Energy Overflow', type: 'number', multiplier: 0.1, unit: 'kWh', role: 'value.power.consumption' }, // Total Battery Charge Energy
81: { name: 'data.totalpvenergyOverflow', description: 'Total PV Energy Overflow', type: 'number', multiplier: 0.1, unit: 'kWh', role: 'value.power.consumption' }, // 'Total Consumption': (42, 'kWh'),
87: { name: 'data.totalFeedOverfow', description: 'Total Feed-in Energy Overfow', type: 'number', multiplier: 0.01, unit: 'kWh', role: 'value.power.consumption' }, // 'Total Feed-in Energy': (41, 'kWh'),
 91: { name: 'data.feedinenergyOverflow', description: 'Feed in Energy Overflow', type: 'number', multiplier: 0.01, unit: 'kWh', role: 'value.power' }, // Feed in Energy: (12.2,'kWh'),
93: { name: 'data.consumeenergyOverfow', description: 'Consume Energy Overfow', type: 'number', multiplier: 0.01, unit: 'kWh', role: 'value.power' }, // Consume Energy: (7.8,'kWh'),

The "Line" 88 + 89 are the same except the number. There should be the "Overflow" additive in Line 89, but without the multiplier and Count or Number as unit.

Perhaps you can also add line 72 as "totalBatteryInputFromGridOverflow".

Thanks for your effort!

simatec, 2024-01-04:
> Could you change the Value from kw to "Count" or "Number" for these Values. There should also be no multiplier.
> Perhaps you could also add an "l" to "Overfow" at 77/87/93 to have "Overflow".

Where does it not need a multiplier? That's not clear from your list.
Which units need to be changed? This is not clear from your list.

Since I don't have this hardware and can't test it with it, I need very specific information...

simatec, 2024-01-13:
Can you please provide more precise data on this?

thorsten-vogt, 2024-01-14:
I'm not a developer, so please check the syntax, but I would write it this way:

69: { name: 'data.yieldtotalOverflow', description: 'Inverter AC-Energy out total Overflow', type: 'number', role: 'indicator.state' }, // Counter of 'data.yieldtotal' overflows - Calculation example: 'data.realyieldtotal = 'data.yieldtotalOverflow'*6553,5+'data.yieldtotal'
72: { name: 'data.totalBatteryInputFromGridOverflow', description: 'Total Battery Input from Grid Overflow', type: 'number', role: 'indicator.state' }, // Counter of 'data.totalBatteryInputFromGrid' overflows - Calculation example: 'data.realYieldTotal = 'data.totalBatteryInputFromGridOverflow'*6553,5+'data.totalBatteryInputFromGrid'
75: { name: 'data.totalBatteryDischargeEnergyOverflow', description: 'Total Battery Discharge Energy Overflow', type: 'number', role: 'indicator.state' }, // Counter of 'data.totalBatteryDischargeEnergy' overflows - Calculation example: data.realTotalBatteryInputFromGrid = 'data.totalBatteryDischargeEnergyOverflow'*6553,5+'data.totalBatteryDischargeEnergy'
77: { name: 'data.totalBatteryChargeEnergyOverflow', description: 'Total Battery Charge Energy Overflow', type: 'number', role: 'indicator.state' }, // Counter of 'data.totalBatteryChargeEnergy' overflows - Calculation example: data.realTotalBatteryChargeEnergy = 'data.totalBatteryChargeEnergyOverflow'*6553,5+'data.totalBatteryChargeEnergy'
81: { name: 'data.totalpvenergyOverflow', description: 'Total PV Energy Overflow', type: 'number', role: 'indicator.state' }, // Counter of 'data.totalpvenergy' overflows - Calculation example: data.realTotalpvenergy = 'data.totalpvenergyOverflow'*6553,5+'data.totalpvenergy'
87: { name: 'data.totalFeedOverflow', description: 'Total Feed-in Energy Overflow', type: 'number', role: 'indicator.state' }, // Counter of 'data.totalFeed' overflows: Calculation example - data.realTotalFeed = 'data.totalFeedOverflow'*655,35+'data.totalFeed'
89: { name: 'data.totalConsumptionOverflow', description: 'Total Consumption Overflow', type: 'number', role: 'indicator.state' }, // Counter of 'data.totalconsumption' overflows - Calculation example: data.realTotalConsumption = 'data.totalConsumptionOverflow'*655,35+'data.totalconsumption'
91: { name: 'data.feedinenergyOverflow', description: 'Feed in Energy Overflow', type: 'number', role: 'indicator.state' }, // Counter of 'data.feedinenergy' overflows - Calculation example: data.realFeedinenergy = 'data.feedinenergyOverflow'*655,35+'data.feedinenergy'
93: { name: 'data.consumeenergyOverfow', description: 'Consume Energy Overflow', type: 'number', role: 'indicator.state' }, // Counter of 'data.consumeenergy' overflows - Calculation example: data.realConsumeenergy = 'data.consumeenergyOverflow'*655,35+'data.consumeenergy'

simatec, 2024-01-14:
Fix in v09.3

closed 2024-01-14 by simatec

Charlie-2222, 2024-06-10:
The Adapter is working well also for a "X1-Boost/Air/Mini" for several month.
But now I do have the same issue with Overflow of the "yieldtotal" value.
However i see no overflow value in my object datastructure.
 Could it be that the enhancement in v09.3 does not include the "X1-Boost/Air/Mini" ?