simatec/ioBroker.solax #191: keine Wallbox Daten mehr

Andre-0815-4711, 2024-01-14:
seit dem letzten Update kommen kein Wallbox Daten mehr

simatec, 2024-01-14:
Bitte poste den Debuglog, die Einstellungen des Adapters und die Version des Adapters.

Andre-0815-4711, 2024-01-14:
Version ist 0.9.2.
`<html>
<body>

2024-01-14 13:33:30.332 - debug: solax.3 (57326) Start Local Request
--
2024-01-14 13:33:31.479 - debug: solax.3 (57326) local request: {"SN":"SQKBYF1ABC","ver":"3.004.11","type":1,"Data":[3,0,22891,23694,23028,0,0,0,0,0,0,0,0,0,3941,0,65461,65493,54,4,65516,12,65088,0,13,0,1,15,0,0,0,0,0,4992,5000,5002,8237,3597,6145,7,13123,12850,12592,16712,12336,12337,12592,0,1,100,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,7978,3344,6145,0,1,1,1,0,0,119,898,266,64876,1],"Information":[11,1,"C31103J2123456",1,1.08,1.01,0,0,0,1],"OCPPServer":"","OCPPChargerId":""}
2024-01-14 13:33:31.506 - debug: solax.3 (57326) End Local Request

</body>
</html>`

Einstellungen:
[image]

Es wird nur der Plugstatus erkannt?
[image]

die letzten Daten wurden am 13.01.24 zwischen 23:00 Uhr und 0:00 Uhr ausgelesen

Andre-0815-4711, 2024-01-14:
Ich glaube ich habe den Fehler schon gefunden.
Wenn das Auto mit laden fertig ist, geht der Plugstatus auf den Wert 3. also Fertig/Aufgeladen/complete

Andre-0815-4711, 2024-01-14:
Main.js korrigiert.

closed 2024-01-14 by Andre-0815-4711