simatec/ioBroker.solax #193: Wallbox Werte nicht vorhanden / nicht aktualisiert.

Azrael4224, 2024-01-15:
Grüßt euch!

Leider sind bei der Wallbox abfrage noch ein paar Werte unstimmig:

[image]

acpower0: keine Übermittlung der Gesamtleistung
acpower3: Textfeld ohne Leistung
capacity: keine Übermittlung
chargeduration: evtl falsches Datenformat (Datum <> Sekunden)
totalcharged: keine Aktualisierung,

Die Temperaturwerte stimmen auch nicht, die habe ich im JSON String aber schon nicht gefunden.

Bin mir jetzt nicht sicher ob es an meiner Installation oder an dem Adapter liegt. Habe deshalb noch eine neue Instanz installiert und mit der Wallbox verbunden. Da werden die Werte mit Fehler (außer acpower3) dann gar nicht mehr erzeugt.

[image]

Debug message:

`local request: {"SN":"SQWGQWHPAS","ver":"3.004.11","type":1,"Data":[2,2,23225,23196,23118,599,604,601,1390,1400,1388,4179,20,0,5243,0,47,47,65488,64940,65184,64888,63998,0,12,0,2,15,0,0,0,0,0,4992,4992,4994,2105,3852,6145,1,0,0,0,0,0,0,0,0,1,100,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1701,0,10256,3851,6145,0,1,1,1,0,0,120,607,100,64936,2],"Information":[11,1,"C31103J1644075",1,1.08,1.01,0,0,0,1],"OCPPServer":"","OCPPChargerId":""}
`

Version 0.9.3

[image]

Grüße und Danke
Richard

simatec, 2024-01-16:
Ich kann aufgrund fehlender Hardware nur die Sachen verarbeiten, die mir als Issue übermittelt werden.
Bitte den String auswerten und hier posten. Die json beginnt mit Wert 0

Azrael4224, 2024-01-16:
{"sn":"SQWGQWHPAS","inverterType":"X1-LX","uploadTime":"Tue Jan 16 2024 14:12:25 GMT+0100 (Central European Standard Time)","connectType":"local","online":true,"connection":true,"firmwareVersion":"3.004.11","totalSize":11,"chargerSN":"C31103J1644075","version":1.08,"acpower":0,"plugstatus":"Unplugged","chargemode":"Stop","acvoltage1":229.19,"acvoltage2":228.68,"acvoltage3":231.12,"accurrent1":0,"accurrent2":0,"accurrent3":0,"acpower1":0,"acpower2":0,"capacity":0,"totalcharged":580.7,"plugTemperature":0,"boardTemperature":12,"plugged":0,"chargeduration":0,"acpower3":"Wait Mode"}

simatec, 2024-01-16:
Nein aus deinem String im ersten Post brauche ich die Zuordnung

Azrael4224, 2024-01-17:
Okay, da hast du ja meine Vorlage schon in die inverterData.js übernommen. Das sollte bis auf Kleinigkeiten (Eco und Green vertauscht in Zeile 1) soweit noch stimmen.

simatec, 2024-01-17:
Naja die habe ich dann übernommen und dort ist es dann wohl vertauscht. Darum die Bitte das zu prüfen. Ich kann es aufgrund fehlender Hardware nicht nachstellen. Und die Werte habe ich 1 zu 1 übernommen

Azrael4224, 2024-01-18:
Die Habe ich gerade penibel Geprüft und einen Parser über das Log laufen lassen. Das schaut alles gut aus.
Kann denn jemand mit der Wallbox meinen Fehler bestätigen oder widerlegen? Hab ja immer noch meine IOB Instanz in verdacht.

Azrael4224, 2024-01-18:
Getan, danach wurden sie wieder mit den gleichen Werten wie vorher angelegt. Habe auch eine neue Instanz installiert, gleiches Verhalten. Downdate auf 0.9.0, alles erfolglos.

Azrael4224, 2024-01-22:
Und heute habe ich auf dem Reserve PI IOBroker Server installiert und die Solax instanz eingerichtet. Exakt das gleiche Verhalten im ersten Post.

simatec, 2024-01-22:
Du musst hieraus die Werte ermitteln:

`local request: {"SN":"SQWGQWHxxx","ver":"3.004.11","type":1,"Data":[2,2,23225,23196,23118,599,604,601,1390,1400,1388,4179,20,0,5243,0,47,47,65488,64940,65184,64888,63998,0,12,0,2,15,0,0,0,0,0,4992,4992,4994,2105,3852,6145,1,0,0,0,0,0,0,0,0,1,100,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1701,0,10256,3851,6145,0,1,1,1,0,0,120,607,100,64936,2],"Information":[11,1,"C31103J1xxxxx",1,1.08,1.01,0,0,0,1],"OCPPServer":"","OCPPChargerId":""} `

Die Zuordnung unter Data beginnt mit 0

Azrael4224, 2024-01-22:
Habe ich. Die Zuordnung, die du verwendet hast, stammt von mir. Ist auch in Ordnung.

Azrael4224, 2024-01-22:
[Andre-0815-4711](https://github.com/Andre-0815-4711)
Kannst du das so nachvollziehen?

simatec, 2024-01-22:
@Andre-0815-4711  Kannst du das nachstellen?

Andre-0815-4711, 2024-01-22:
Moin,

der Fehler bei mir war, dass der 4. PlugStatus (ID 3 Complete) nicht vorhanden war. Das hat doch tatsächlich dazu geführt, dass alle Werte nicht aktualisiert wurden.
Den hab ich mir dann schnell selber in die Main.js geschrieben und schon kamen wieder alle Daten. Ich hatte auch meine Instanz im Verdacht, als ich aber gesehen habe, das der Datensatz im Log vorhanden war bin ich über die 3 gestolpert.

MfG
André

Andre-0815-4711, 2024-01-22:
An der Stelle könnte man den Wert 0 auch eigentlich anders benennen, anstatt PlugStatus wäre ChargeStatus treffender.
Ich habe auch noch 3x Stromstärke gefunden, bin mir aber noch nicht sicher was das sein soll. Am Donnerstag soll ja die Sonne scheinen, dann schau ich mir das noch mal an...

simatec, 2024-01-23:
Stelle die Instanz auf Debug... Ich verstehe dein Issue leider noch immer nicht so richtig... Wenn du die selbe Wallbox hast wie @Andre-0815-4711 und bei ihm alle Werte passen, liegt eventuell ein Problem mit der Verbindung zu deiner Wallbox vor.

Grundsätzlich könnt ihr mir hier nur die Inputs liefern und die Werte dem String zuordnen. Ich kann aufgrund fehlender Hardware das absolut nicht nachstellen.

Andre-0815-4711, 2024-01-23:
Hallo,

also das mit dem ACPower 3 kann ich bestätigen. Ich habe die 11kW Wallbox aber mein Auto kann nur mit 2 Phasen also 7,4kW laden, deswegen habe ich das wohl bisher ignoriert.
[image]

simatec, 2024-01-23:
Aber Wait Mode ist ja nicht korrekt ... Da stimmen deine Werte, die du mir übermittelt hast nicht. Dieser Wert muss immer eine Zahl sein.

Bitte überprüfe das noch einmal... Ich möchte die Version ins Stable bringen und kann das mit falschen Werten für die Wallbox nicht machen

Andre-0815-4711, 2024-01-23:
Der Wert stimmt, zu 100% kann ich das aber nur beweisen, wenn ich ein Auto hätte das mit allen 3 Phasen also 11kW laden kann.
Ich habe mir letztens mal die Arbeit gemacht und die Werte aus dem Log in eine Excel zu übertragen und zu sehen was sich wann ändert.
[image]
Dabei ist mir gerade aufgefallen, dass der Wert "gridacpower" den ich am Sonntag identifiziert habe nicht richtig ist, das ist die Load vom Haus. Konnte ich am Sonntag nicht genau sehen da der Akku vom Haus leer war.
[image]

Azrael4224, 2024-01-23:
Das mit dem Wert für acpower3 kann ich anbringen:
[image]

Ich parse hier das IOB Log, speichere es in einer SQL DB und zerlege den String dann in die einzelnen Spalten.

[Andre-0815-4711](https://github.com/Andre-0815-4711) Hast du die Werte ACPower, Capacity, TotalCharged und ChargeDuration? Und sind die auch gefüllt?

Andre-0815-4711, 2024-01-23:
ACPower, Capacity und TotalCharged habe ich und die Werte sind auch aktualisiert.
ChargeDuration habe ich auch schon gesehen, müsste ich aber mal beim nächsten Ladevorgang noch mal kontrollieren.

Vorhin ist mir noch aufgefallen, dass der Wert 0 PlugStatus in ChargerStatus umbenannt werden müsste und die Stati dazu auch. 
Aktuell habe ich das Auto nicht angesteckt und in der APP wird der Verbindungsstatus mit verfügbar angezeigt. Wenn man die APP auf Englisch umstellt steht dann auch Charger Status und available. Dann macht der Wert 26 auch mehr Sinn, nur werden da dann 0, 1 und 2 ausgegeben was dann ja auch wieder einer zu viel ist.

simatec, 2024-01-24:
Ich denke ermittelt erstmal die Werte genau, bevor es im Adapter dann auch vollumfänglich unterstützt werden kann

Azrael4224, 2024-01-25:
`0: { name: 'data.chargerstatus', description: 'Plug Status', type: 'number', multiplier: 1, unit: '', role: 'indicator.state' }, // '0=Unplugged 1=Waiting 2=Charging 3=Finished 4=Error': (0, ''), 
1: { name: 'data.chargemode', description: 'Charge Mode', type: 'number', multiplier: 1, unit: '', role: 'indicator.state' }, // '0=Stop 1=Fast 2=Eco 3=Green': (1, ''), 
2: { name: 'data.acvoltage1', description: 'Grid Voltage 1', type: 'number', multiplier: 0,01, unit: 'V', role: 'value.power' }, // '': (2, 'V'), 
3: { name: 'data.acvoltage2', description: 'Grid Voltage 2', type: 'number', multiplier: 0,01, unit: 'V', role: 'value.power' }, // '': (3, 'V'), 
4: { name: 'data.acvoltage3', description: 'Grid Voltage 3', type: 'number', multiplier: 0,01, unit: 'V', role: 'value.power' }, // '': (4, 'V'), 
5: { name: 'data.accurrent 1', description: 'AC Current 1', type: 'number', multiplier: 0,01, unit: 'A', role: 'value.power' }, // '': (5, 'A'), 
6: { name: 'data.accurrent 2', description: 'AC Current 2', type: 'number', multiplier: 0,01, unit: 'A', role: 'value.power' }, // '': (6, 'A'), 
7: { name: 'data.accurrent 3', description: 'AC Current 3', type: 'number', multiplier: 0,01, unit: 'A', role: 'value.power' }, // '': (7, 'A'), 
8: { name: 'data.acpower1', description: 'AC Power 1', type: 'number', multiplier: 0,01, unit: 'W', role: 'value.power' }, // '': (8, 'W'), 
9: { name: 'data.acpower2', description: 'AC Power 2', type: 'number', multiplier: 1, unit: 'W', role: 'value.power' }, // '': (9, 'W'), 
10: { name: 'data.acpower3', description: 'AC Power 3', type: 'number', multiplier: 1, unit: 'W', role: 'value.power' }, // '': (10, 'W'), 
11: { name: 'data.acpower', description: 'AC Power', type: 'number', multiplier: 1, unit: 'W', role: 'value.power' }, // '': (11, 'W'), 
12: { name: 'data.charged', description: 'Carged Capacity', type: 'number', multiplier: 0,1, unit: 'KW/h', role: 'value.power' }, // '': (12, 'KW/h'), 
14: { name: 'data.totalcharged', description: 'Total Charged', type: 'number', multiplier: 0,1, unit: 'KW/h', role: 'value.power' }, // '': (14, 'KW/h'), 
26: { name: 'data.plugged', description: 'Car Plugged', type: 'number', multiplier: 1, unit: '', role: 'indicator.state' }, // '0=No 1=Yes 2=Yes': (26, ''), 
80: { name: 'data.chargeduration', description: 'Charge Duration', type: 'number', multiplier: 1, unit: 's', role: 'value.time' }, // '': (80, 's')`

Das ist mein Vorschlag.

Änderungen:
 - weiterer Status 4=Error
 - Rename Zeile 0
 - Rename Zeile 12
 - Remove Zeile 23
 - Remove Zeile 24
 - Lookups zu Zeile 26 hinzugefügt

Azrael4224, 2024-01-25:
`0: { name: 'data.chargerstatus', description: 'Plug Status', type: 'number', multiplier: 1, unit: '', role: 'indicator.state' }, // '0=Unplugged 1=Waiting 2=Charging 3=Finished 4=Error': (0, ''), 
1: { name: 'data.chargemode', description: 'Charge Mode', type: 'number', multiplier: 1, unit: '', role: 'indicator.state' }, // '0=Stop 1=Fast 2=Eco 3=Green': (1, ''), 
2: { name: 'data.acvoltage1', description: 'Grid Voltage 1', type: 'number', multiplier: 0,01, unit: 'V', role: 'value.power' }, // '': (2, 'V'), 
3: { name: 'data.acvoltage2', description: 'Grid Voltage 2', type: 'number', multiplier: 0,01, unit: 'V', role: 'value.power' }, // '': (3, 'V'), 
4: { name: 'data.acvoltage3', description: 'Grid Voltage 3', type: 'number', multiplier: 0,01, unit: 'V', role: 'value.power' }, // '': (4, 'V'), 
5: { name: 'data.accurrent 1', description: 'AC Current 1', type: 'number', multiplier: 0,01, unit: 'A', role: 'value.power' }, // '': (5, 'A'), 
6: { name: 'data.accurrent 2', description: 'AC Current 2', type: 'number', multiplier: 0,01, unit: 'A', role: 'value.power' }, // '': (6, 'A'), 
7: { name: 'data.accurrent 3', description: 'AC Current 3', type: 'number', multiplier: 0,01, unit: 'A', role: 'value.power' }, // '': (7, 'A'), 
8: { name: 'data.acpower1', description: 'AC Power 1', type: 'number', multiplier: 0,01, unit: 'W', role: 'value.power' }, // '': (8, 'W'), 
9: { name: 'data.acpower2', description: 'AC Power 2', type: 'number', multiplier: 1, unit: 'W', role: 'value.power' }, // '': (9, 'W'), 
10: { name: 'data.acpower3', description: 'AC Power 3', type: 'number', multiplier: 1, unit: 'W', role: 'value.power' }, // '': (10, 'W'), 
11: { name: 'data.acpower', description: 'AC Power', type: 'number', multiplier: 1, unit: 'W', role: 'value.power' }, // '': (11, 'W'), 
12: { name: 'data.charged', description: 'Carged Capacity', type: 'number', multiplier: 0,1, unit: 'KW/h', role: 'value.power' }, // '': (12, 'KW/h'), 
14: { name: 'data.totalcharged', description: 'Total Charged', type: 'number', multiplier: 0,1, unit: 'KW/h', role: 'value.power' }, // '': (14, 'KW/h'), 
26: { name: 'data.plugged', description: 'Car Plugged', type: 'number', multiplier: 1, unit: '', role: 'indicator.state' }, // '0=No 1=Yes 2=Yes': (26, ''), 
80: { name: 'data.chargeduration', description: 'Charge Duration', type: 'number', multiplier: 1, unit: 's', role: 'value.time' }, // '': (80, 's')`

Hier mal mein Vorschlag.

Änderungen:
- Neuer Wert 4=Error für Zeile 0
- Rename Zeile 0
- Rename Zeile 12
- Remove Zeile 23
- Remove Zeile 24
- Lookup für Zeile 23

Andre-0815-4711, 2024-01-25:
Moin,
ich war auch fleißig.... sind schon als Pullrequest eingestellt.
Aber erst mal zu Deinen Änderungen:

- Neuer Wert 4=Error für Zeile 0  - stimme ich zu, Error hatte ich noch nicht
- Rename Zeile 0 - stimme ich zu, steht schon im PullRequest aber Statusbeschreibung 1:1 aus der Solax APP
- Rename Zeile 12 - ja kann man machen, für mich würde die aktuelle ID passen, als Name ist ja Carged Capacity angegeben
- Remove Zeile 23 - stimme ich zu
- Remove Zeile 24 - stimme ich nicht zu, die gleiche Temperatur wird in der Solax App auch angezeigt
- Lookup für Zeile 23 oder doch die 26? - stimme ich zu, steht schon im PullRequest

Was den Fehler mit acpower und acpower3 angeht, der ist auch bei mir noch vorhanden.
[image]
[image]

Warum auch immer, scheint sich hier der InverterStateLocal einzumischen.
[image]

MfG
André

Azrael4224, 2024-01-25:
Super, Vielen Dank!

Da ich leider nicht in der Lage bin, die Änderungen einzupflegen, darf ich dich bitten, das zu tun?
Die Namen der Objekte sind mir egal.

Werte aus dem WR brauche ich nicht, stören aber auch nicht. Meine hiermit Zeile 19 bis 24. So wie ich das sehe, werden die nur über den Bus weiter gereicht.

Andre-0815-4711, 2024-01-25:
Den Fehler mit ACpower3 habe ich auch gefunden und im Pullrequest behoben.
[image]

Andre-0815-4711, 2024-01-25:
So, @Azrael4224 hat die Einstellungen aus den PullRequest mal local vorgenommen und ist jetzt alle Fehler los.

Lediglich den Status Preparing könnte man noch treffender umbenennen. 
Ready to Charge kommt dem Status am nächsten.

`const _wallboxStateLocal = {
    0: 'Available',
    1: 'Preparing',
    2: 'Charging',
    3: 'Complete',
    4: 'Error'
};`

Azrael4224, 2024-01-26:
Ja, kann ich bestätigen. Mit Andres Hilfe sind alle Fehler erledigt.
Danke!

closed 2024-02-04 by simatec