simatec/ioBroker.solax #201: X3-EVC Wallbox Control?

erwini, 2024-02-18:
Hi,

Small question: Is it possible to control the  X3-EVC Wallbox loacaly. At the moment it is contolled via vendor app, but what happens, if the cloud conenction is not available any more or lost or if you want to control it via iobroker? (talking about the charging modes)
I believe, it is not include in this adapter?

simatec, 2024-02-25:
yes... It is possible to control the X3-EVC Wallbox loacaly

erwini, 2024-02-25:
how? Any hint?

erwini, 2024-02-25:
Maybe there is a missunderstanding. I am able to read out the values from the wallbox. But i have not found any object, where it is possible to control and change the setting/mode of the wallbox, like the mode STOP, GREEN, ECO and so on.

simatec, 2024-02-25:
The adapter only evaluates the data and writes the data to objects