simatec/ioBroker.solax #203: Fehlerprotokoll Solax-Adapter über Token-ID

6smile9, 2024-03-07:
Hallo zusammen,

im Protokoll von ioBroker taucht permanent folgende Fehlermeldung auf bei Verwendung des Solax-Adapters über Token-ID:
 
  State value to set for "solax.3.data.json" has to be stringified but received type "number"

Diese Fehlermeldung erscheint im 5-Minuten-Takt und ist unabhängig vom Wechselrichtertyp (erscheint beim X3 Hybrid und beim X1 Mini).

Neustart und Neuinstallation des Adapters brachten keine Besserung.

Version Solax-Adapter: 0.9.6
ioBroker läuft auf Proxmox, Node.js v18.19.0, NPM 10.2.3

Danke und Gruß

thorsten-vogt, 2024-03-11:
Ich habe die Meldung "State value to set for "solax.2.data.json" has to be type "string" but received type "number"" bei meinem X3 Hybrid auch, aber nur bei der Abfrage der Cloud, nicht bei der lokalen Abfrage. Scheint jedoch keinen Einfluss auf die Funktion des Adapters zu haben.

simatec, 2024-03-11:
Kannst du bitte mal die Daten vom Objekt posten.... Die findest du in den Objekten (Stift auf der rechten Seite)

Sollte in etwa so ausschauen...

````
{
  "type": "state",
  "common": {
    "name": "json data",
    "type": "json",
    "role": "state",
    "read": true,
    "write": false
  },
  "native": {},
  "from": "system.adapter.solax.0",
  "user": "system.user.admin",
  "ts": 1704155363325,
  "_id": "solax.0.data.json",
  "acl": {
    "object": 1636,
    "state": 1636,
    "owner": "system.user.admin",
    "ownerGroup": "system.group.administrator"
  }
}
````

6smile9, 2024-03-11:
Hallo,
hier sind, wie gewünscht, die Daten vom Objekt:
{
  "type": "state",
  "common": {
    "name": "json data",
    "type": "json",
    "role": "state",
    "read": true,
    "write": false
  },
  "native": {},
  "from": "system.adapter.solax.3",
  "user": "system.user.admin",
  "ts": 1709713954443,
  "_id": "solax.3.data.json"
}

thorsten-vogt, 2024-03-12:
Hallo @simatec ,

ich habe das Objekt gelöscht, den Adapter neu gestartet und es wurde neu angelegt. die Fehlermeldung bleibt jedoch.

Hier meine Version:
`
Version 0.9.6 (non-npm: simatec/ioBroker.solax) in /opt/iobroker/node_modules/iobroker.solax, node: v18.17.1, js-controller: 4.0.24
`

Hier die Fehlermeldung:
`
State value to set for "solax.2.data.json" has to be stringified but received type "number"
`

Hier die Objektdaten nach Neuerstellung:
`
{
  "type": "state",
  "common": {
    "name": "json data",
    "type": "json",
    "role": "state",
    "read": true,
    "write": false
  },
  "native": {},
  "from": "system.adapter.solax.2",
  "user": "system.user.admin",
  "ts": 1710238243851,
  "_id": "solax.2.data.json"
}
`

Evtl. liegt es an meiner veralteten js-controller Version? Ist aber, wie gesagt, für den Betrieb nicht kritisch und alles funktioniert.

6smile9, 2024-03-12:
Hallo @simatec,
ich  habe jetzt alle Objekte gelöscht und alle Instanzen neugestartet. Leider kommt die Fehlermeldung für die Instanzen, die die API nutzen immer noch:
State value to set for "solax.3.data.json" has to be stringified but received type "number"
State value to set for "solax.2.data.json" has to be stringified but received type "number"

@thorsten-vogt
ich habe die js-controller Version 5.0.19

whaensch, 2024-06-23:
Leider kann ich anscheinend keinen eigenen Punkt aufmachen?

Ich habe IoBroker mit dem Solax--Adapter über direkten lokalen Zugriff installiert. Leider werden die Werte falsch angezeigt. 
In der Objektbeschreibung steht X3-FTH statt irgendetwas mit X1.
Die Fehlermeldung mit dem String und Number habe ich auch - dabei mir ist aufgefallen, dass bei "InverterSN" eine 8 steht und nicht die originale SN des Inverters

**Versions:**  
 - Adapter version: 0.9.6
 - JS-Controller version: 5.0.19
 - Node version: 20.15.0
 - Operating system: Windows 10

[image]
[image]

closed 2024-09-17 by simatec