simatec/ioBroker.solax #211: SOLAX TOU not works with Version 0.9.12

MicroWulf76, 2024-10-12:
Hallo,

der Adapter 0.9.12 und die vorhergehenden liefern falsche bzw. keine Werte, wenn das TOU im Solax Wechselrichter aktiviert wird.
(Protokol "wrong value")
Sobald zurückgeschaltet wird auf "self use" funktionieren die Abfragen wieder korrekt.

Ansonsten finde ich den Adapter super
mfG Mirko Pabst

simatec, 2024-10-12:
Wie fragst du die Daten ab? Lokal oder Cloud?
Was ist TOU? Das sagt mir erstmal nix. 
was steht im Debuglog?

MicroWulf76, 2024-10-12:
Hallo,

 

vielen Dank für das Feedback.

 

Die Daten werden im IOBroker über SOLAX Cloud abgefragt.

 

TOU ist ein Betriebsmodus (relativ neu) Time of Use.

Da können die Betriebszustände des Wechselrichters per Kalender und Uhrzeit gewechselt werden.

Die Konfiguration geschieht  über die APP bzw die Webseite von Solax.

 

Wie beschrieben, sobald dieser Modus gewählt wird, kann er mit den meisten Daten aus der Cloud nichts mehr anfangen.

 

 

mfG

simatec, 2024-10-12:
Bitte poste hier den Debuglog, damit ich sehen kann, was an Daten kommt

MicroWulf76, 2024-10-12:
Hallo, 

 

ich hoffe es ist das was sie wollen ;-)

 

2024-10-12 16:09:30.919 - info: host.IOBroker stopInstance system.adapter.solax.1 (force=false, process=true)

2024-10-12 16:09:30.926 - info: solax.1 (1892) Got terminate signal TERMINATE_YOURSELF

2024-10-12 16:09:30.929 - info: solax.1 (1892) terminating

2024-10-12 16:09:30.930 - debug: solax.1 (1892) Plugin sentry destroyed

2024-10-12 16:09:30.931 - info: solax.1 (1892) Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason

2024-10-12 16:09:31.061 - info: host.IOBroker stopInstance system.adapter.solax.1 send kill signal

2024-10-12 16:09:31.433 - info: solax.1 (1892) terminating

2024-10-12 16:09:31.548 - info: host.IOBroker instance system.adapter.solax.1 terminated with code 11 (ADAPTER_REQUESTED_TERMINATION)

2024-10-12 16:09:34.893 - info: host.IOBroker instance system.adapter.solax.1 in version "0.9.12" (non-npm: simatec/ioBroker.solax#9e787e4ba0cf3b6db02d78043e5aaab555fb6b4a) started with pid 2004

2024-10-12 16:09:36.778 - debug: solax.1 (2004) Redis Objects: Use Redis connection: 127.0.0.1:9001

2024-10-12 16:09:36.835 - debug: solax.1 (2004) Objects client ready ... initialize now

2024-10-12 16:09:36.837 - debug: solax.1 (2004) Objects create System PubSub Client

2024-10-12 16:09:36.839 - debug: solax.1 (2004) Objects create User PubSub Client

2024-10-12 16:09:36.922 - debug: solax.1 (2004) Objects client initialize lua scripts

2024-10-12 16:09:36.929 - debug: solax.1 (2004) Objects connected to redis: 127.0.0.1:9001

2024-10-12 16:09:36.994 - debug: solax.1 (2004) Redis States: Use Redis connection: 127.0.0.1:9000

2024-10-12 16:09:37.023 - debug: solax.1 (2004) States create System PubSub Client

2024-10-12 16:09:37.025 - debug: solax.1 (2004) States create User PubSub Client

2024-10-12 16:09:37.114 - debug: solax.1 (2004) States connected to redis: 127.0.0.1:9000

2024-10-12 16:09:37.169 - debug: solax.1 (2004) Plugin sentry Initialize Plugin (enabled=true)

2024-10-12 16:09:37.652 - info: solax.1 (2004) starting. Version 0.9.12 (non-npm: simatec/ioBroker.solax#9e787e4ba0cf3b6db02d78043e5aaab555fb6b4a) in /opt/iobroker/node_modules/iobroker.solax, node: v19.9.0, js-controller: 6.0.11

2024-10-12 16:09:39.829 - debug: solax.1 (2004) Solax is started in cloud-mode

2024-10-12 16:09:39.833 - debug: solax.1 (2004) System longitude: 11.462134 | System latitude: 51.7559549

2024-10-12 16:09:39.834 - debug: solax.1 (2004) nightCalc started ...

2024-10-12 16:09:39.836 - debug: solax.1 (2004) dusk: 19:02

2024-10-12 16:09:39.837 - debug: solax.1 (2004) dawn: 07:01

2024-10-12 16:09:39.837 - debug: solax.1 (2004) current local Time: 16:09

2024-10-12 16:09:39.839 - debug: solax.1 (2004) calculate astrodata ...

2024-10-12 16:09:39.839 - debug: solax.1 (2004) Sun Altitude: 18.3°

2024-10-12 16:09:39.840 - debug: solax.1 (2004) Sun Azimut: 230°

2024-10-12 16:09:40.051 - debug: solax.1 (2004) Request Interval: 5 minute(s)

2024-10-12 16:09:40.252 - debug: solax.1 (2004) Solax Cloud is available ... Status: 200

2024-10-12 16:09:40.293 - debug: solax.1 (2004) request-result: {"success":true,"exception":"Query success!","result":{"inverterSN":"H34A12J2872313","sn":"SNWFHVYGGM","acpower":-31,"yieldtoday":1.2,"yieldtotal":8,"feedinpower":0,"feedinenergy":2.3,"consumeenergy":28.31,"feedinpowerM2":0,"soc":95,"peps1":0,"peps2":0,"peps3":0,"inverterType":"14","inverterStatus":"102","uploadTime":"2024-10-12 15:00:02","batPower":31,"powerdc1":0,"powerdc2":0,"powerdc3":null,"powerdc4":null,"batStatus":"0"},"code":0}

2024-10-12 16:09:40.300 - debug: solax.1 (2004) API Request successfully completed

2024-10-12 16:09:41.308 - debug: solax.1 (2004) solaxRequest: {"success":true,"exception":"Query success!","result":{"inverterSN":"H34A12J2872313","sn":"SNWFHVYGGM","acpower":-31,"yieldtoday":1.2,"yieldtotal":8,"feedinenergy":2.3,"consumeenergy":28.31,"soc":95,"inverterType":"14","inverterStatus":"102","uploadTime":"2024-10-12 15:00:02","batPower":31,"batStatus":"0"},"code":0}

2024-10-12 16:09:41.452 - info: solax.1 (2004) State value to set for "solax.1.data.json" has to be type "string" but received type "number"

MicroWulf76, 2024-10-12:
Ich muss nochmal etwas nachreichen:

 

Nach kurzer Zeit häufen sich die Fehlermeldungen und es werden für immer mehr Werte null geliefert,

 

2024-10-12 16:19:42.555 - warn: solax.1 (2004) request error: Error: undefined is not a valid state value

2024-10-12 16:19:43.851 - info: web.0 (5969) <== Disconnect system.user.admin from 192.168.10.2 flot

2024-10-12 16:19:43.856 - info: web.0 (5969) <== Disconnect system.user.admin from 192.168.10.2 vis.0

2024-10-12 16:19:44.220 - info: web.0 (5969) ==> Connected system.user.admin from 192.168.10.2

2024-10-12 16:19:44.661 - info: web.0 (5969) ==> Connected system.user.admin from 192.168.10.2

2024-10-12 16:20:03.726 - info: web.0 (5969) <== Disconnect system.user.admin from 192.168.10.2 flot

2024-10-12 16:20:03.729 - info: web.0 (5969) <== Disconnect system.user.admin from 192.168.10.2 vis.0

2024-10-12 16:20:04.175 - info: web.0 (5969) ==> Connected system.user.admin from 192.168.10.2

2024-10-12 16:20:04.615 - info: web.0 (5969) ==> Connected system.user.admin from 192.168.10.2

2024-10-12 16:20:26.244 - info: web.0 (5969) <== Disconnect system.user.admin from 192.168.10.2 flot

2024-10-12 16:20:26.247 - info: web.0 (5969) <== Disconnect system.user.admin from 192.168.10.2 vis.0

2024-10-12 16:20:26.784 - info: web.0 (5969) ==> Connected system.user.admin from 192.168.10.2

2024-10-12 16:20:27.247 - info: web.0 (5969) ==> Connected system.user.admin from 192.168.10.2

2024-10-12 16:20:34.900 - info: web.0 (5969) ==> Connected system.user.admin from 192.168.10.70

2024-10-12 16:20:35.421 - info: web.0 (5969) ==> Connected system.user.admin from 192.168.10.70

2024-10-12 16:20:39.775 - info: web.0 (5969) <== Disconnect system.user.admin from 192.168.10.70 vis.0

2024-10-12 16:20:39.776 - info: web.0 (5969) <== Disconnect system.user.admin from 192.168.10.70 flot

2024-10-12 16:20:40.092 - info: web.0 (5969) ==> Connected system.user.admin from 192.168.10.70

2024-10-12 16:20:40.596 - info: web.0 (5969) ==> Connected system.user.admin from 192.168.10.70

2024-10-12 16:21:01.598 - info: web.0 (5969) <== Disconnect system.user.admin from 192.168.10.70 vis.0

2024-10-12 16:21:01.598 - info: web.0 (5969) <== Disconnect system.user.admin from 192.168.10.70 flot

2024-10-12 16:21:01.676 - info: web.0 (5969) ==> Connected system.user.admin from 192.168.10.70

2024-10-12 16:21:02.227 - info: web.0 (5969) ==> Connected system.user.admin from 192.168.10.70

2024-10-12 16:21:15.161 - info: solax.0 (26008) State value to set for "solax.0.data.json" has to be type "string" but received type "number"

2024-10-12 16:21:27.550 - info: web.0 (5969) <== Disconnect system.user.admin from 192.168.10.2 flot

2024-10-12 16:21:27.814 - info: web.0 (5969) ==> Connected system.user.admin from 192.168.10.2

2024-10-12 16:22:03.622 - info: web.0 (5969) <== Disconnect system.user.admin from 192.168.10.70 flot

2024-10-12 16:22:03.624 - info: web.0 (5969) <== Disconnect system.user.admin from 192.168.10.70 vis.0

2024-10-12 16:22:09.515 - info: web.0 (5969) <== Disconnect system.user.admin from 192.168.10.2 flot

2024-10-12 16:22:09.518 - info: web.0 (5969) <== Disconnect system.user.admin from 192.168.10.2 vis.0

2024-10-12 16:22:10.227 - info: web.0 (5969) ==> Connected system.user.admin from 192.168.10.2

2024-10-12 16:22:10.673 - info: web.0 (5969) ==> Connected system.user.admin from 192.168.10.2

2024-10-12 16:24:39.900 - debug: solax.1 (2004) nightCalc started ...

2024-10-12 16:24:39.902 - debug: solax.1 (2004) dusk: 19:02

2024-10-12 16:24:39.902 - debug: solax.1 (2004) dawn: 07:01

2024-10-12 16:24:39.904 - debug: solax.1 (2004) current local Time: 16:24

2024-10-12 16:24:39.904 - debug: solax.1 (2004) calculate astrodata ...

2024-10-12 16:24:39.905 - debug: solax.1 (2004) Sun Altitude: 16.5°

2024-10-12 16:24:39.905 - debug: solax.1 (2004) Sun Azimut: 233.4°

2024-10-12 16:24:39.966 - debug: solax.1 (2004) is Night: false

2024-10-12 16:24:40.053 - debug: solax.1 (2004) API Request started ...

2024-10-12 16:24:40.118 - debug: solax.1 (2004) Solax Cloud is available ... Status: 200

2024-10-12 16:24:40.154 - debug: solax.1 (2004) request-result: {"success":true,"exception":"Query success!","result":{"inverterSN":"H34A12J2872313","sn":"SNWFHVYGGM","acpower":2,"yieldtoday":1.2,"yieldtotal":8,"feedinpower":0,"feedinenergy":2.3,"consumeenergy":28.31,"feedinpowerM2":0,"soc":95,"peps1":0,"peps2":0,"peps3":0,"inverterType":"14","inverterStatus":"-1","uploadTime":"2024-10-12 15:20:02","batPower":-1,"powerdc1":0,"powerdc2":0,"powerdc3":null,"powerdc4":null,"batStatus":"0"},"code":0}

2024-10-12 16:24:40.159 - debug: solax.1 (2004) API Request successfully completed

2024-10-12 16:24:41.165 - debug: solax.1 (2004) solaxRequest: {"success":true,"exception":"Query success!","result":{"inverterSN":"H34A12J2872313","sn":"SNWFHVYGGM","acpower":2,"yieldtoday":1.2,"yieldtotal":8,"feedinenergy":2.3,"consumeenergy":28.31,"soc":95,"inverterType":"14","inverterStatus":"-1","uploadTime":"2024-10-12 15:20:02","batPower":-1,"batStatus":"0"},"code":0}

2024-10-12 16:24:41.201 - warn: solax.1 (2004) request error: Error: undefined is not a valid state value

2024-10-12 16:26:14.949 - info: solax.0 (26008) State value to set for "solax.0.data.json" has to be type "string" but received type "number"

2024-10-12 16:26:49.913 - info: web.0 (5969) ==> Connected system.user.admin from 192.168.10.70

2024-10-12 16:26:49.994 - info: web.0 (5969) <== Disconnect system.user.admin from 192.168.10.70

2024-10-12 16:26:50.612 - info: web.0 (5969) ==> Connected system.user.admin from 192.168.10.70

2024-10-12 16:26:51.127 - info: web.0 (5969) ==> Connected system.user.admin from 192.168.10.70

2024-10-12 16:27:03.382 - info: web.0 (5969) <== Disconnect system.user.admin from 192.168.10.70 vis.0

2024-10-12 16:27:03.383 - info: web.0 (5969) <== Disconnect system.user.admin from 192.168.10.70 flot

MicroWulf76, 2024-10-14:
Guten Morgen,

ich habe mir nochmals angeschaut wo eventuell der Fehler liegt:
2024-10-14 06:41:17.749 - debug: solax.1 (21114) request-result: {"success":true,"exception":"Query success!","result":{"inverterSN":"H34A12J2872313","sn":"SNWFHVYGGM","acpower":170,"yieldtoday":0.9,"yieldtotal":15.3,"feedinpower":0,"feedinenergy":5.3,"consumeenergy":28.41,"feedinpowerM2":0,"soc":65,"peps1":0,"peps2":0,"peps3":0,"inverterType":"14","inverterStatus":"-1","uploadTime":"2024-10-14 05:40:01","batPower":-170,"powerdc1":0,"powerdc2":0,"powerdc3":null,"powerdc4":null,"batStatus":"0"},"code":0}
--

Die Daten werden offensichtlich korrekt angelifert.

Allerdings werden leglich, wie im Bild zu sehen nur 3 Daten (batPower, yielttoday und yieldtotal) übernommen (siehe zeitstempel).

Ich hoffe diese Feststellung hilft irgendwie bei der Eingrenzung weiter.

[image]

simatec, 2024-10-14:
Ich sehe da erstmal keinen Unterschied bei der Cloudabfrage... Kannst du mir mal noch den Debuglog senden, wenn `TOU` deaktiviert ist. Grundsätzlich würde ich aber erstmal sagen, dass es für die Cloudabfrage keinen Unterschied machen sollte.

MicroWulf76, 2024-10-15:
Hier das Ganze nochmal ohne TOU.
Sobald ich auf "Self Use" umschalte werden alle Werte wieder korrekt zugewiesen.

solax.1 | 2024-10-15 21:14:03.639 | info | State value to set for "solax.1.data.json" has to be type "string" but received type "number"
-- | -- | -- | --
solax.1 | 2024-10-15 21:14:03.468 | debug | solaxRequest: {"success":true,"exception":"Query success!","result":{"inverterSN":"H34A12J2872313","sn":"SNWFHVYGGM","acpower":141,"yieldtoday":1.9,"yieldtotal":18.7,"feedinenergy":5.4,"consumeenergy":28.41,"soc":69,"inverterType":"14","inverterStatus":"102","uploadTime":"2024-10-15 20:09:59","batPower":-141,"batStatus":"0"},"code":0}
solax.1 | 2024-10-15 21:14:02.462 | debug | API Request successfully completed
solax.1 | 2024-10-15 21:14:02.458 | debug | request-result: {"success":true,"exception":"Query success!","result":{"inverterSN":"H34A12J2872313","sn":"SNWFHVYGGM","acpower":141,"yieldtoday":1.9,"yieldtotal":18.7,"feedinpower":0,"feedinenergy":5.4,"consumeenergy":28.41,"feedinpowerM2":0,"soc":69,"peps1":0,"peps2":0,"peps3":0,"inverterType":"14","inverterStatus":"102","uploadTime":"2024-10-15 20:09:59","batPower":-141,"powerdc1":0,"powerdc2":0,"powerdc3":null,"powerdc4":null,"batStatus":"0"},"code":0}
solax.1 | 2024-10-15 21:14:02.411 | debug | Solax Cloud is available ... Status: 200
solax.1 | 2024-10-15 21:14:02.316 | debug | API Request started ...
solax.1 | 2024-10-15 21:14:02.206 | debug | is Night: false
solax.1 | 2024-10-15 21:14:02.192 | debug | Sun Azimut: 292°
solax.1 | 2024-10-15 21:14:02.192 | debug | Sun Altitude: -27.2°
solax.1 | 2024-10-15 21:14:02.192 | debug | calculate astrodata ...
solax.1 | 2024-10-15 21:14:02.191 | debug | current local Time: 21:14
solax.1 | 2024-10-15 21:14:02.191 | debug | dawn: 07:06
solax.1 | 2024-10-15 21:14:02.190 | debug | dusk: 18:56
solax.1 | 2024-10-15 21:14:02.190 | debug | nightCalc started ...

simatec, 2024-10-17:
Dieser Fehler tritt auf, wenn nicht alle Daten kommen bzw. halt in der Abfrage Daten fehlen.
Dies ist in deinem vorletzten Log der Fall.

Warum laufen da zwei Instanzen? Fragst du den WR bzw. die Cloud 2x ab?
Da könnte das Problem liegen. Die Solax Cloud ist da etwas nervig und nicht gerade gut. Du solltest, dann eher überlegen, das ganze lokal abzufragen.

MicroWulf76, 2024-10-17:
Hallo,

 

nein ich habe zwei Wechselrichter von Solax und daher die zwei Instanzen.

So lange wie beide in „self use“ laufen ist auch alles prima. 

 

Ich musste halt nur das Zeitlimit auf 5 Minuten setzen, da die sonst die API, für einige Zeit, sperren.

 

mfG

simatec, 2024-10-17:
Kannst du mir mal bitte die API Adresse, die du auf der Webseite von Solax unter Dienst-API findest posten?
Bitte tokenId und sn unkenntlich machen

simatec, 2024-10-17:
Um die komplette Adresse sehen zu können, musst du die tokenId aktualisieren

MicroWulf76, 2024-10-17:
Na, ich hoffe mal das ist was du wolltest:

 

https://global.solaxcloud.com/proxyApp/proxy/api/getRealtimeInfo.do?tokenId=20241017235849726777715&sn=

simatec, 2024-10-17:
Ja das passt… also Cloud passt auch. Ich baue morgen mal ein paar Debugs ein, mit denen du testen kannst