simatec/ioBroker.solax #217: Keine Daten nach Setup: Nicht verbunden mit Gerät oder Dienst

ph1ppm, 2024-11-29:
**Describe the bug**  
Credentials werden im Adapter hinzugefügt: API-Key (in App und Cloud identisch) + Seriennummer SN des Pocket-WLAN, aber testweise auch aller anderen Komponenten (WR und Datahub1000) erfolglos ausprobiert.
Verbindung laut obejct connected = false, somit keine Daten verfügbar.
Logfile anbei.

**Screenshots & Logfiles**  
[log.txt](https://github.com/user-attachments/files/17957796/log.txt)
[image]
[image]

**Versions:**  
 - Adapter version: <v0.9.12>
 - JS-Controller version: <6.0.11> 
 - Node version: <v18.19.1>

simatec, 2024-12-11:
Auch hier bitte die Config als JSON File und den debuglog von Solax posten

ph1ppm, 2024-12-11:
[debug.txt](https://github.com/user-attachments/files/18097516/debug.txt)
[solax.admin_jsonConfig.json](https://github.com/user-attachments/files/18097837/solax.admin_jsonConfig.json)

simatec, 2024-12-11:
Bitte deine Config als JSON File posten und nicht die jsonConfig.json

ph1ppm, 2024-12-11:
Wo genau finde ich die?

simatec, 2024-12-11:
Du kannst die Konfiguration im Menü vom Solax Adapter über die Pfeile oben rechts runterladen

ph1ppm, 2024-12-12:
Danke für den Hinweis, hier die Config. Ich hab meine SN und den Token mal unkenntlich gemacht...

[solax.0.json](https://github.com/user-attachments/files/18109774/solax.0.json)

simatec, 2024-12-12:
Bitte teste folgende URL über den Browser:

https://www.eu.solaxcloud.com/proxyApp/proxy/api/getRealtimeInfo.do?tokenId=xxxxxx&sn=xxxxxx

ph1ppm, 2024-12-12:
Fehlermeldung beim Aufruf der URL:
{"exception":"token invalid!","code":103,"tokenId":"xxxxxx","success":false}

Token und SN natürlich vorher eingesetzt.
Ich hab auch nochmal den Token im Webinterface der Solax Cloud abgeglichen und er ist definitiv korrekt.

**//Update**: Neugenerierung des Tokens bringt die Instanz jetzt zum fliegen... keine Ahnung, warum es mit dem Vorherigen nicht geklappt hat.

closed 2024-12-12 by ph1ppm