simatec/ioBroker.solax #218: Falscher Type für json

peterpan-user, 2024-12-20:
**Describe the bug**  
Jede 5 Minuten erhalte ich die folgende Info im Log:
  State value to set for "solax.0.data.json" has to be type "string" but received type "number"

vgl. bitte auch #181 , scheint bei mir aber keine Auswirkung gehabt haben. 

**Versions:**  
 - Adapter version: 0.9.12
 - JS-Controller version: 7.0.3
 - Node version: v20.18.1
 - Operating system: ubuntu

**Additional context**  
Meldung erhalte ich schon länger

htrecksler, 2024-12-29:
Gleiches Verhalten bei Controller 7.1.5 und Adapterversion 9.1.12

simatec, 2024-12-29:
Zeige bitte die Config von `solax.0.data.json`

htrecksler, 2024-12-29:
{
  "type": "state",
  "common": {
    "name": "json data",
    "type": "string",
    "role": "state",
    "read": true,
    "write": false
  },
  "native": {},
  "from": "system.adapter.solax.0",
  "user": "system.user.admin",
  "ts": 1723466510107,
  "_id": "solax.0.data.json",
  "acl": {
    "object": 1638,
    "state": 1638,
    "owner": "system.user.admin",
    "ownerGroup": "system.group.administrator"
  }
}

simatec, 2024-12-29:
Es sollte so aussehen...

````
{
  "type": "state",
  "common": {
    "name": "json data",
    "type": "string",
    "role": "json",
    "read": true,
    "write": false
  },
  "native": {},
  "from": "system.adapter.solax.0",
  "user": "system.user.admin",
  "ts": 1644221196513,
  "_id": "solax.0.data.json",
  "acl": {
    "object": 1636,
    "state": 1636,
    "owner": "system.user.admin",
    "ownerGroup": "system.group.administrator"
  }
}
````

peterpan-user, 2024-12-29:
So sieht es bei mir aus:
`{
  "type": "state",
  "common": {
    "name": "json data",
    "type": "string",
    "role": "json",
    "read": true,
    "write": false
  },
  "native": {},
  "from": "system.adapter.solax.0",
  "user": "system.user.admin",
  "ts": 1706544406556,
  "_id": "solax.0.data.json",
  "acl": {
    "object": 1636,
    "state": 1636,
    "owner": "system.user.admin",
    "ownerGroup": "system.group.administrator"
  }
}`

simatec, 2024-12-29:
@peterpan-user Und was hast du für eine Fehlermeldung?

peterpan-user, 2024-12-29:
Keine Fehlermeldung, aber alle 5 Minuten die Info "State value to set for "solax.0.data.json" has to be type "string" but received type "number"" im Log.

htrecksler, 2024-12-29:
ich habe gerade "role" auf "json" umgestellt, ansonsten konnte ich aber keinen Unterschied feststellen.
Ich hab danach den Adapter neu gestartet, aber ohne Auswirkung.

Die Meldung lautet immer noch: 
State value to set for "solax.0.data.json" has to be type "string" but received type "number"

simatec, 2024-12-29:
Was hast du denn für Werte in dem State?

htrecksler, 2024-12-29:
{
  "success": true,
  "exception": "Query success!",
  "inverterSN": "H34A08I6377636",
  "sn": "SN8GN9GTSV",
  "inverterType": "X3-Hybrid-G4",
  "inverterStatus": "Idle Mode",
  "uploadTime": "2024-12-29 16:32:52",
  "connectType": "cloud",
  "connection": true,
  "acpower": 0,
  "yieldtoday": 0.8,
  "yieldtotal": 2309.4,
  "feedinpower": -11171,
  "feedinenergy": 1703.85,
  "consumeenergy": 1265.16,
  "soc": 8,
  "powerdc1": 0,
  "powerdc2": 0,
  "batPower": 0
}

peterpan-user, 2024-12-29:
{
  "connectType": "cloud",
  "connection": true,
  "exception": "Query success!",
  "inverterSN": "H34A10J3778051",
  "inverterStatus": "Normal Mode",
  "inverterType": "X3-Hybrid-G4",
  "sn": "SYP3THD365",
  "success": true,
  "uploadTime": "2024-12-29 16:40:37",
  "acpower": -18,
  "batPower": 0,
  "consumeenergy": 2603.59,
  "feedinenergy": 2189.7,
  "feedinpower": -3424,
  "powerdc1": 0,
  "powerdc2": 0,
  "soc": 8,
  "yieldtoday": 0.9,
  "yieldtotal": 5468.2
}

htrecksler, 2024-12-29:
es ist ja jetzt nicht tragisch, ist ja "nur" eine Info, aber eigentlich gehört sie da nicht hin, und ich versuche möglichst mein System aktuell und mein log sauber zu halten.

simatec, 2024-12-29:
Die Meldung ist Verwirrend, da euer Daten ein String sind und keine Number.
Lauit offiziller Doku ist die Role und der Type auch korekkt. 
https://github.com/ioBroker/ioBroker.docs/blob/master/docs/en/dev/stateroles.md#common

Könnt ihr die Daten aus dem State mal loggen, damit man sieht, ob da eventuell irgendwann statt des json Strings eine Nummer geschrieben wird?

htrecksler, 2024-12-29:
ich hab mir mal ein kleines Script erstellt. 

on({id:'solax.0.data.json',change:'any'}, function (obj) {
    log('Wert JSON: ' + JSON.stringify(obj), 'warn');
});
mal sehen was kommt...

peterpan-user, 2024-12-29:
Mit dem obigen Script:

29.12.2024, 17:02:36.121	[info ]: javascript.0 (3440) Stopping script script.js.Homedingskacke.Solax
29.12.2024, 17:02:38.817	[info ]: javascript.0 (3440) Start JavaScript script.js.Homedingskacke.Solax (Javascript/js)
29.12.2024, 17:02:38.835	[info ]: javascript.0 (3440) script.js.Homedingskacke.Solax: registered 1 subscription, 0 schedules, 0 messages, 0 logs and 0 file subscriptions

solax.0
2024-12-29 17:05:56.613	info	State value to set for "solax.0.data.json" has to be type "string" but received type "number"

29.12.2024, 17:05:56.617	[warn ]: javascript.0 (3440) script.js.Homedingskacke.Solax: Wert JSON: {"id":"solax.0.data.json","newState":{"val":0,"ts":1735488356614,"ack":true,"lc":1735488356614,"from":"system.adapter.solax.0","q":0,"user":"system.user.admin"},"oldState":{"val":"{\"connectType\":\"cloud\",\"connection\":true,\"exception\":\"Query success!\",\"inverterSN\":\"H34A12J3779051\",\"inverterStatus\":\"Idle Mode\",\"inverterType\":\"X3-Hybrid-G4\",\"sn\":\"SYP4THD465\",\"success\":true,\"uploadTime\":\"2024-12-29 17:00:37\",\"acpower\":0,\"batPower\":0,\"consumeenergy\":2604.29,\"feedinenergy\":2189.7,\"feedinpower\":-1176,\"powerdc1\":42,\"powerdc2\":0,\"soc\":8,\"yieldtoday\":0.9,\"yieldtotal\":5468.2}","ts":1735488057904,"ack":true,"lc":1735488057904,"from":"system.adapter.solax.0","q":0,"user":"system.user.admin"},"state":{"val":0,"ts":1735488356614,"ack":true,"lc":1735488356614,"from":"system.adapter.solax.0","q":0,"user":"system.user.admin"}}
29.12.2024, 17:05:57.918	[warn ]: javascript.0 (3440) script.js.Homedingskacke.Solax: Wert JSON: {"id":"solax.0.data.json","newState":{"val":"{\"connectType\":\"cloud\",\"connection\":true,\"exception\":\"Query success!\",\"inverterSN\":\"H34A12J3779051\",\"inverterStatus\":\"Idle Mode\",\"inverterType\":\"X3-Hybrid-G4\",\"sn\":\"SYP4THD465\",\"success\":true,\"uploadTime\":\"2024-12-29 17:05:37\",\"acpower\":0,\"batPower\":0,\"consumeenergy\":2604.29,\"feedinenergy\":2189.7,\"feedinpower\":-1231,\"powerdc1\":0,\"powerdc2\":0,\"soc\":8,\"yieldtoday\":0.9,\"yieldtotal\":5468.2}","ts":1735488357916,"ack":true,"lc":1735488357916,"from":"system.adapter.solax.0","q":0,"user":"system.user.admin"},"oldState":{"val":0,"ts":1735488356614,"ack":true,"lc":1735488356614,"from":"system.adapter.solax.0","q":0,"user":"system.user.admin"},"state":{"val":"{\"connectType\":\"cloud\",\"connection\":true,\"exception\":\"Query success!\",\"inverterSN\":\"H34A12J3779051\",\"inverterStatus\":\"Idle Mode\",\"inverterType\":\"X3-Hybrid-G4\",\"sn\":\"SYP4THD465\",\"success\":true,\"uploadTime\":\"2024-12-29 17:05:37\",\"acpower\":0,\"batPower\":0,\"consumeenergy\":2604.29,\"feedinenergy\":2189.7,\"feedinpower\":-1231,\"powerdc1\":0,\"powerdc2\":0,\"soc\":8,\"yieldtoday\":0.9,\"yieldtotal\":5468.2}","ts":1735488357916,"ack":true,"lc":1735488357916,"from":"system.adapter.solax.0","q":0,"user":"system.user.admin"}}

htrecksler, 2024-12-29:
[image]

simatec, 2024-12-31:
Es sieht eigentlich gut aus... Es kommt nur der String und kein anderer numerischer Wert. Ich werde das nochmal checken, warum diese Meldung kommt

simatec, 2025-11-29:
done