simatec/ioBroker.solax #219: Microwechselrichter auslesen

Andre-0815-4711, 2024-12-22:
Moin,

Solax hat seit kürzerem neue Microwechselrichter im Programm welche ein integriertes WLAN haben.
Diese lassen sich noch nicht mit dem Adapter auslesen. Sehr wahrscheinlich müsste der Adapter an die FW angepasst werden.
Wie kann Ich Dich hier unterstützen, Daten sammel und bereitstellen?

MfG 
André

simatec, 2024-12-22:
Gib mal bitte den genauen Typ an... dann kann ich mal schauen, ob es schon etwas dazu gibt

Andre-0815-4711, 2024-12-22:
Das ist der X1 Micro 2-1 in der 1000W version. FW des WiFi Modul kann ich Dir morgen schreiben wenn  der Kleine wieder aufgewacht ist...

Andre-0815-4711, 2024-12-23:
Moin,

also das WIFI FW vom X1 Micro ist 008.03 .

MfG

simatec, 2024-12-29:
Da gibt es leider noch keine Daten für. Das heißt, du musst versuchen den Adapter auf Debug zu stellen und schauen, ob du Daten bekommst. Dann den Data JSON String auswerten und die einzelnen Werte zuordnen. Das ganze kannst du dann hier posten und ich kann es einpflegen.

Sollte dann in etwa so ausschauen:
https://github.com/simatec/ioBroker.solax/blob/4a5dc721f8225419aec4615ea7008d1bc065a5bb/lib/inverterData.js#L28
https://github.com/simatec/ioBroker.solax/blob/4a5dc721f8225419aec4615ea7008d1bc065a5bb/lib/inverterData.js#L71

simatec, 2024-12-29:
Dann wird dieser Wechselrichter aktuell keine lokale Verbindung unterstützen

Andre-0815-4711, 2024-12-29:
Also mit der App ist eine lokale Verbindung möglich.
Das Webinterface ähnelt dem der WLAN Sticks mit der FW Version 3.
Kannst Du mir sagen wie ich das testen kann um Dir dann die Daten zur Verfügung zu stellen?

simatec, 2024-12-29:
Du kannst folgendes versuchen...

http://IP-Adresse:80/?optType=ReadRealTimeData&pwd=WR-Passwort

Andre-0815-4711, 2024-12-29:
Klappt leider nicht, wird sofort auf http://IP-Adresse/system umgeleitet.

simatec, 2024-12-29:
Dann kann ich leider nicht weiterhelfen. Da müsstest du mal testen und recherchieren, was da geht und was nicht.

closed 2025-02-02 by simatec