simatec/ioBroker.solax #223: Zusätzliche Werte in den Adapter aufnehmen

thkn1973, 2025-01-30:
Hallo 

Wäre es möglich noch weiter Werte aus dem X3 in den Adapter aufzunehmen?
Mir fehlt zb momentan Grid Import und Export, sowie auch der komplette momentane PV Ertrag.
Sprich nicht die Totalwerte sonden Momentanzustände
Wenn ich helfen kann bitte mir fragen oder tests anher.

Danke
LG Thomas

simatec, 2025-02-02:
Stelle mal die Instanz auf Debug. Dort wird dann der json String vom Abruf ausgegeben.
In diesem String musst du die passenden Werte raussuchen und mir die Position im String durchgeben.
Beim zählen der Position bitte mit 0 beginnen. Heißt, der erste Wert ist die Position 0

thkn1973, 2025-02-03:
Danke für die rasch Antwort.
Alle aufscheinenden Werte sind eh schon im Adapter. Dürfte nicht mehr geben die man damit auslesen kann.
Nach ein wenig einlesen hab ich rasugefunden das ich die momentane Gesamterzeugung mir selbst zusammenrechnen muss.
Und der NetzImport/export ist vorhanden, nur ist der Netzbezug in negativ angegeben, muss man halt auch umrechnen.
Danke LG

closed 2025-04-07 by simatec