simatec/ioBroker.solax #224: [feature request] die Sofar Solar HYD-KTL Wechselrichter in den Adapter aufnehmen

erichsch, 2025-02-01:
Im Homeassistant gibt es eine ähnliche Solax Anbindung. 
Diese beinhaltet auch die Sofar-Solar Wechselrichter.
Könnte man diese WR in den Adapter ebenfalls aufnehmen?
Anbei die Modbus Reference von den Sofar Solar HYD-KTL Wechselrichtern.

[SOFAR_Modbus_Protocol_English_G3_V1.22(only 3PH) Dezimal.xlsx](https://github.com/user-attachments/files/18627703/SOFAR_Modbus_Protocol_English_G3_V1.22.only.3PH.Dezimal.xlsx)

erichsch, 2025-02-01:
Its no Bug, its a feature Request!

simatec, 2025-02-02:
Gibt denn der Adapter im lokalen Modus etwas aus?
Stelle mal die Instanz auf debug und schaue mal, was im Log steht

simatec, 2025-04-07:
Aus dem Log sehe ich nur, dass deine Konfiguration nicht vollständig zu sein scheint

> solax.0 2025-02-04 06:53:22.508 warn system settings cannot be called up. Please check configuration!

erichsch, 2025-04-08:
Ich habe auch nur den LSE-3 (LAN). Vielleicht liegt es an dem?

Liebe Grüße Erich

closed 2025-11-29 by simatec