simatec/ioBroker.solax #226: Tagesertrag wird mehrfach in Datenbank eingetragen

Dirk-Ma, 2025-02-21:
**Describe the bug**  
Ich protokolliere den Datenpunkt "solax.0.data.yieldtoday" sowohl in einer Influx als auch in einer SQL-Datenbank. Leider kommt es immer wieder mal vor, dass der Tagesertrag vom Vortag mit in dem nächsten Tag eingetragen wird.

**To Reproduce**  
Steps to reproduce the behavior:
Reproduzieren kann ich das Verhalten nicht, aber es liegt glaube ich daran, dass der Datenpunkt kurz vor Tageswechsel auf 0 geht.

**Expected behavior**  
Das der Tageswert vom Vortag nicht als erster Wert des Folgetages eingetragen wird.

**Screenshots & Logfiles**  
Sind beigefügt

**Versions:**  
 - Adapter version: 0.9.14
 - JS-Controller version: 7.0.6
 - Node version: v18.20.6
 - Operating system: Raspbian GNU/Linux 10 (buster)

[image]
[image]
[image]
[image]

**Additional context**  
Add any other context about the problem here.

simatec, 2025-02-23:
Du könntest eventuell 0 Werte ignorieren. Aber ich schaue mir das nochmal an

Dirk-Ma, 2025-02-23:
Hi,

danke, gute Idee. Werde ich so einstellen. Vielleicht findest du ja noch etwas, was dieses Verhalten erklärt.

Gruß

Dirk

closed 2025-04-07 by simatec